arendst/Tasmota #5503: serial port hangout

tartesso26, 2019-03-20:
### BUG DESCRIPTION
i have a device conected to an arduino to control many motors and lights, i send commands from app to tasmota and tasmota comunicate with arduino by serial port to know state of motors and move them, i send commands to arduino by serial and get response by serial bridge, and everything works fine, except when app user click buttons very quick, on that situation tasmota hangout, after go offline and after some seconds backs online and works fine again.

Is there some known problem when send multiple mqtt serial commands very quick? like out of memory serial buffer full .... something? serial port is config to 115200

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Nodemcu D1mini
- [X] Tasmota binary firmware version number used : 6.0.3 / (self-compiled)

- [X] Provide the output of command ``status 0`` :
```
{"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"TS_00000006","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

ascillato, 2019-03-20:
Hi, please update to latest Tasmota 6.5.0
Your version is too old and a lots of fixes related to serial bridge, serial port were applied. 
Thanks

Jason2866, 2019-03-21:
For the usage with serial bridge i would recommend to use Core 2.4.2
Try latest stable Version from here http://thehackbox.org/tasmota/release/020402/sonoff.bin

ascillato2, 2019-03-21:
Hi, any news on this?

ascillato2, 2019-03-22:
Closing issue as there is no feedback. Please, when you have time try the solutions proposed and tell us if those solve your issue. Thanks

closed 2019-03-22 by ascillato2