arendst/Tasmota #5504: WS2812: Scheme12 not working with setoption37 set to 25

helmar74, 2019-03-20:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : _____
- [X] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [X] Development IDE - Compiler / Upload tools used : ____ / ____
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS = {"Status":{"Module":18,"FriendlyName":["sonoffWemosMini-7-GRBW"],"Topic":"sonoffWemosMini_7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffWemosMiniWS2812GRBW","OtaUrl":"http://10.0.1.2:5555/data/firmwares/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:33","StartupUTC":"2019-03-20T11:58:04","Sleep":50,"CfgHolder":4617,"BootCount":23,"SaveCount":170,"SaveAddress":"F4000"}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019.03.20 12:32:13","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSD","sonoffap"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008019","280500000119000000000000000000000000","00000000"]}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FAAE19C","00078FA0","23BA1FCE","002033C0"]}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS5 = {"StatusNET":{"Hostname":"sonoffWemosMini-7-GRBW","IPAddress":"10.0.1.181","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.2","Mac":"84:F3:EB:0C:32:A1","Webserver":2,"WifiConfig":5}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.50","MqttPort":1883,"MqttClientMask":"sonoffWemosMini-7-GRBW","MqttClient":"sonoffWemosMini-7-GRBW","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 20 12:11:37 2019","Local":"Wed Mar 20 13:11:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:12","Sunset":"18:21"}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS10 = {"StatusSNS":{"Time":"2019-03-20T13:11:37"}}
13:11:37 MQT: stat/sonoffWemosMini_7/STATUS11 = {"StatusSTS":{"Time":"2019-03-20T13:11:37","Uptime":"0T00:13:33","Vcc":2.989,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":100,"Color":"FF920E00","HSBColor":"33,95,100","Channel":[100,57,5,0],"Scheme":12,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"SSD","BSSId":"F0:B0:14:6F:B5:4F","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```

### TO REPRODUCE
I've connected a WS2812 (with GRBW colors) to a Wemos Mini D. With Version 6.4.1 
I configured 
#define USE_WS2812_CTYPE     NEO_GRBW
in my_user_config.h

All single colors an all schemes were working with this setting.

Now I tried setoption37 25.
Everything works, but setting scheme 12 only shows green color an not the "Fire scheme"

Is it a bug in Tasmota or do I have to set setoption37 to an other value? 

### EXPECTED BEHAVIOUR
scheme 12 command should show fire scheme

ascillato, 2019-03-20:
Hi,

Your status 0 shows Tasmota version 6.5.0 but then you talk about Tasmota 6.4.1

Which one are you using? 

Setoption37 was introduced in 6.4.1.19

helmar74, 2019-03-20:
I'm using 6.5.0

With 6.4.1 (before updating to 6.5.0) it was fine...

ascillato, 2019-03-20:
In the template is missing the information, so I have to ask you again. Did you use the precompiled bins from development or release, or you compile by youself the 6.5.0?

helmar74, 2019-03-20:
I compiled by myself...

ascillato, 2019-03-20:
Ok. Can you try the precompiled bin?

helmar74, 2019-03-20:
Yes I recompiled it twice.
With Version 6.4.1 I was using
#define USE_WS2812_CTYPE NEO_GRBW
in my_user_config.h

With Version 6.5.0 I left this entry to default:
#define USE_WS2812_CTYPE NEO_GRB
and I set setoption37 to 25 in Webinterface.
All single colors and schemes are working fine, except of scheme 12

When I change to
#define USE_WS2812_CTYPE NEO_GRBW
in Version 6.5.0 I have the same behaviour... All single colors and schemes are working, instead of scheme 12 (only shows green color)....

arendst, 2019-03-20:
I tested on core 2.4.2. Will test on core 2.5.0 now...

arendst, 2019-03-20:
Again, works fine on core 2.5.0 too.

helmar74, 2019-03-20:
@arendst 
Just to getting it right....

If I compile with:
#define USE_WS2812_CTYPE NEO_GRB
and setting setoption37 to 25 

is this same as compiling with:
#define USE_WS2812_CTYPE NEO_GRBW
and setting setoption37 to 0
?

ascillato2, 2019-03-30:
@helmar74 

Hi, any news on this?

ascillato2, 2019-04-10:
Closing this issue as there is no feedback.

closed 2019-04-10 by ascillato2

james-lan, 2024-09-27:
Reopening as I'm having the exact same issue with SetOption37, value 24. 

Tasmota release 14.2.0 (Version 14.2.0(release-tasmota)-2_7_7(2024-08-14T12:36:35) on ESP8285H16 (for the GRB string) 

Scheme X and ledX ignore the SetOption37 values completely, and outputs as if SetOption37 is 0. Color DOES work.

I'm not sure the expected behavior of LedX, but this appears to be a bug with Scheme per the documentation on SetOption37, noting the Schemes mentioned specifically: "SetOption37 is used for re-mapping the RGBWcWw channels. This transformation happens right before the hardware-specific output, so all the advanced features (schemes, hsb setting, etc.) works with the original, un-mapped values." 

I have multiple WS2812 chips, but some are wired differently, I have one that's the normal RGB, and a GRB string. 

When SetOption 37 is set to 0:
Behaves as expected, ie: 
Red on RGB is Red  
Red on GRB is Green
Green on RGB is Green
Green on GRB is Red

When Scheme 12 is set (as well as SetOption37 to 0), on the 
RGB: Yellow to Red chasing (R+G to R-only)
GRB: Green and Yellow chasing (G+R to G-only) 

When SetOption 37 is set to 24:
Behaves as expected, ie: 
Red on RGB is Green
Red on GRB is Red
Green on RGB is Red
Green on GRB is Green

When Scheme 12 is set (as well as SetOption37 to 24), there is no change in Scheme behavior.  
RGB: Yellow to Red chasing (R+G to R-only)
GRB: Green and Yellow chasing (G+R to G-only) 

Additionally, Scheme 10 , which is usually (on the 0/RGB):
Blank/Green/Red

is even with SetOption37 24 on the GRB: 
Blank/Red/Green

Scheme 7 is on 0/RGB:
RGB

with SetOption37 24 on the GRB: 
GRB

Scheme 6 is Red on 0/RGB, but Green on 24/GRB

Scheme 8 I can't tell, and the rest either don't use R/G or are too difficult to see, but my conclusion is that Schemes ignore SetOption37. 

I also tested ledX (ie individual LEDs) and they work as if SetOption37 were 0. 
#FF0000
RGB: Red
GRB: Green

As noted above, I'm not sure the expected behavior for ledX, as I find nothing saying if it should (but it seems like it should) 

Note that Color works as expected. 
Using the same #FF0000 as above:
RGB/0: Red
GRB/24: Green

I would venture to guess, but haven't delved deep into the code, that Scheme is relying on ledX to set pixels, but that's coming out wrong. 

Apologies if you'd prefer a new bug report, (I can rewrite it if needed) but this seems 100% the same and was closed due to lack of feedback.