arendst/Tasmota #5506: AiLight / EleLight doesn't work properly and crashes

Th0maz, 2019-03-21:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I recently bought an EleLight LED light bulb (https://www.amazon.de/dp/B0774MS7H3) which looks exactly like the AiLight from the inside.
After soldering some jumper cables to the solder points, I was able to erase flash and re-flash several versions of Tasmota firmware. The bulb comes up successfully and is accessible via wifi.
But as soon as I power on some LEDs, the bulb looses wifi connection most times.
Sometimes it stays unresponsive and I have to re-flash the firmware to make it come up again.
I tried different commands like dimmer, power, color. The behavior is always the same.
When I tried some settings for sleep, setoption60, savedata or poweronstate, the device crashed occasionally. In most cases I am only able to run one or two commands, no matter what time has passed before I do.
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : EleLight aka AiLight
- [x] Tasmota binary firmware version number used : 6.3.0, 6.4.1, 6.4.1.19, 6.5.0 / pre-and self-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
11:51:32 CMD: status 0
11:51:32 RSL: stat/sonoff/STATUS = {"Status":{"Module":27,"FriendlyName":["AiLight-01"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:51:32 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:20","StartupUTC":"2019-03-21T10:51:12","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":18,"SaveAddress":"FA000"}}
11:51:32 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:51:32 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyWifiSSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
11:51:32 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
11:51:32 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"AiLight-01","IPAddress":"192.168.111.247","Gateway":"192.168.111.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.111.28","Mac":"80:7D:3A:76:9F:14","Webserver":2,"WifiConfig":4}}
11:51:32 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_769F14","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:51:32 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 21 10:51:32 2019","Local":"Thu Mar 21 11:51:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:51","Sunset":"19:03"}}
11:51:32 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-21T11:51:32"}}
11:51:32 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-21T11:51:32","Uptime":"0T00:00:20","Vcc":3.153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":68,"POWER":"OFF","Dimmer":33,"Color":"00540000","HSBColor":"120,100,33","Channel":[0,32,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"MyWifiSSID","BSSId":"38:10:D5:E8:FC:5F","Channel":6,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
11:51:59 CMD: dimmer 100
11:51:59 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"00FF0000","HSBColor":"120,100,100","Channel":[0,100,0,0]}
11:52:01 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
11:52:01 RSL: stat/sonoff/POWER = OFF

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-21:
Probably this is the reason https://github.com/arendst/Sonoff-Tasmota/pull/4991#issuecomment-459831215
Maybe running out of free RAM. Have yout tried release with core 2.4.2 already?
http://thehackbox.org/tasmota/release/020402/sonoff.bin

Th0maz, 2019-03-21:
I tried several releases with 2.3.0 as well as 2.4.2. Doesn't make a difference so far.

ascillato, 2019-03-21:
The light was working with the original firmware?

Are you erasing with esptool.py as explained in the wiki before flashing?

Are you flashing with esptool.py?

Th0maz, 2019-03-21:
Unfortunately I didn't test the light with its original firmware. Flashing tasmota was the first thing I did after I got it...
I use the windows port of esptool (esptool.exe) and erased the flash as described in the wiki. Do you think it's worth a try to use esptool.py?

Jason2866, 2019-03-21:
Mhh, i followed the link. My bulb looks exactly the same as this. I had exact the same package.
Just tested with latest version. It does work without any issue. Probably device is defect.
Last try could be a erasing and flashing with esptool.py  I dont think it is a flashing thing....

ascillato, 2019-03-21:
If you did the erase and you use esptool.exe seems not to be a flashing issue. Sorry, but seems to be a hardware defect.

ascillato, 2019-03-21:
Before throwing that away, you can try an older version of Tasmota or another firmware just to be sure that is not software related.

Jason2866, 2019-03-21:
@ascillato  Already tried 
[image]

ascillato, 2019-03-21:
I meant that @Th0maz can try another firmware different than Tasmota.

As all those versions of Tasmota has been tested and didn't work, another firmware is likely to fail as well, but before throwing that hardware away as faulty, it is something to try.

Th0maz, 2019-03-21:
I just flashed the latest ESPURNA firmware via Tasmota OTA firmware update.
Works out of the box...
So it seems this is no hardware issue. Any ideas how I can continue troubleshooting with tasmota firmware? Don't really want to use espurna in my tasmota environment ;)

For the record: I tried using esptool.py in the meantime to erase the flash and re-flashed tasmota. Did not make a difference.

ascillato, 2019-03-21:
Can you try the precompiled bin called sonoff_basic.bin?

arendst, 2019-03-21:
Did some tests on an ailight which works just fine. Unless wemo emulation is enabled then the web gui response is bad (#5505 ). From your settings I see you are not using emulation.

In this case I cannot reproduce your issue.

Th0maz, 2019-03-21:
> 
> 
> Can you try the precompiled bin called sonoff_basic.bin?

I used the latest precompiled sonoff.bin earlier (6.5.0 with core 2.3.0).

ascillato, 2019-03-21:
Please, try the basic firmare. The one called sonoff_basic.bin

Th0maz, 2019-03-23:
As you said, I tried the Tasmota basic firmware (again flashed OTA via the espurna web interface) and it seems to be working fine so far.
Although I don't really understand why it's working all of a sudden, this issue can be closed :)

closed 2019-03-23 by Th0maz

Jason2866, 2019-03-25:
Maybe this (solved) bug https://github.com/arendst/Sonoff-Tasmota/issues/5521

Th0maz, 2019-09-03:
After an update to Tasmota version 6.6.0 this issue reoccurred. I downloaded the official sonoff-basic.bin and sonoff.bin in both versions, 2.3.0 and 2.5.2.
Tried a "reset 5", however this did not help.
The device seems to be working fine until I light up some of the LEDs (any color, any dimmer value).
Web UI seems to be responding kind of slow though.

For troubleshooting I now ordered another AiLight / EleLight bulb yesterday to make sure it's no hardware issue.

Th0maz, 2019-10-11:
The other AiLight I ordered shows the exact same behaviour. No hardware issue.

Th0maz, 2019-10-15:
Tried the workaround from #6608
Seems to work so far:
"I tryed to remove the // from #define PWM_LIGHTSCHEME0_IGNORE_SLEEP line in the sonoff.h file"