arendst/Tasmota #5507: Tasmota reseting himself

FraatRamski, 2019-03-21:
Hallo,

I have a simple setup. Wemos d1 2 button and 2 relais. 

I have created the following rule:
rule 
on Power1#State=1 do websend [10.xx.xx.xx:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon 
on Power1#State=0 do websend [10.xx.xx.xx:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%280%29​ endon

it seem to be working but after after some seconds tasmota reseting himself. Tried on 3 different wemos d1.

20:11:09 MQT: stat/sonoff/POWER1 = OFF
20:11:09 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
20:11:09 MQT: stat/sonoff/POWER2 = OFF
20:11:10 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
20:11:10 MQT: stat/sonoff/POWER2 = ON
20:11:12 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
20:11:12 MQT: stat/sonoff/POWER2 = OFF
20:11:13 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
20:11:13 MQT: stat/sonoff/POWER1 = ON
20:11:14 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
20:11:14 MQT: stat/sonoff/POWER1 = OFF
20:11:17 MQT: tele/sonoff/STATE = {"Time":"2019-03-21T20:11:17","Uptime":"0T00:00:19","Vcc":2.766,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"POWER1
":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"76:xx:E7:59:93:31","Channel":11,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:09"}}

20:11:30 CMD: rule  on Power1#State=1 do websend [10.12.14.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon  on Pow
er1#State=0 do websend [10.12.14.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%280%29​ endon
20:11:30 MQT: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":259,"Rules":"on Power1#State=1 do websend [10.xx.xx.xx:8181]/
cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon  on Power1#State=0 do websend [10.xx.xx.xx:8181]/cuxd.exe%3Fret=dom.GetObje
ct%28%22Steckdose%20152%22%29.State%280%29​ endon"}
20:11:35 CMD: rule1 on
20:11:35 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":259,"Rules":"on Power1#State=1 do websend [10.xx.xx.xx:8181]/c
uxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon  on Power1#State=0 do websend [10.xx.xx.xx:8181]/cuxd.exe%3Fret=dom.GetObjec
t%28%22Steckdose%20152%22%29.State%280%29​ endon"}
20:11:46 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
20:11:46 MQT: stat/sonoff/POWER1 = ON
20:11:46 RUL: POWER1#STATE=1 performs "websend [10.12.14.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​"
20:11:46 MQT: stat/sonoff/RESULT = {"WebSend":"Done"}
**_20:11:49 MQT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}_**
20:11:49 CFG: Standard verwenden
20:11:50 APP: starte neu

ascillato2, 2019-03-22:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-03-23:
Hi,

Please, when you have time, please, complete the troubleshooting template in order to have more information so as to properly help you. Thanks.

FraatRamski, 2019-03-24:
### BUG DESCRIPTION
Hallo,

I have a simple setup. Wemos d1 2 button and 2 relais. Tasmota 6.5

I have created the following rule:
rule
on Power1#State=1 do websend [10.12.34.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon
on Power1#State=0 do websend [10.12.34.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%280%29​ endon

it seem to be working but after after some seconds tasmota reseting himself. Tried on 3 different wemos d1.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Wemos d1 3x different devices, can reproduce issuse on each one
- [x] Tasmota binary firmware version number used : 6.5.0 / (self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : ATOM / ____
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

09:17:54 MQT: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:17:54 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:16","StartupUTC":"2019-03-24T08:17:38","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":7,"SaveAddress":"FA000"}}
09:17:54 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019.03.21 19:56:58","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:17:54 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
09:17:54 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":510,"Free":492,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000407","0FDAE394","000783A1","23B617CE","00003BC0"]}}
09:17:54 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2127","IPAddress":"xx.xx.xx89","Gateway":"xx.xx.xx1","Subnetmask":"255.255.255.0","DNSServer":"10.12.14.1","Mac":"EC:FA:BC:2F:E8:4F","Webserver":2,"WifiConfig":4}}
09:17:54 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"10.12.14.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2FE84F","MqttUser":"MqttUsr","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:17:54 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 24 08:17:54 2019","Local":"Sun Mar 24 09:17:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:45","Sunset":"19:08"}}
09:17:54 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-24T09:17:54"}}
09:17:54 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-24T09:17:54","Uptime":"0T00:00:16","Vcc":2.766,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"76:xx:E7:59:93:31","Channel":11,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:06"}}}
09:17:55 MQT: tele/sonoff/STATE = {"Time":"2019-03-24T09:17:55","Uptime":"0T00:00:17","Vcc":2.766,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"xx:D9:E7:59:93:31","Channel":11,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
09:51:41 MQT: stat/sonoff/RESULT = {"WebSend":"Done"}
09:51:48 CMD: websend [10.12.14.36:8181]/cuxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%280%29​
09:51:48 MQT: stat/sonoff/RESULT = {"WebSend":"Done"}
09:51:50 CMD: rule1
09:51:51 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":260,"Rules":"on Power1#State=1 do websend [xx.xx.xx.36:8181]/c
uxd.exe%3Fret=dom.GetObject%28%22Steckdose%20152%22%29.State%281%29​ endon on Power1#State=0 do websend [xx.xx.xx.36:8181]/cuxd.exe%3Fret=dom.GetObject
%28%22Steckdose%20152%22%29.State%280%29​ endon"}

**09:52:11 MQT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}**

09:52:11 CFG: Standard verwenden
09:52:12 APP: starte neu

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8

chksum 0x2d
csum 0x2d
v3fff3bd0
~ld

00:00:00 Projekt sonoff Sonoff Version 6.5.0(sonoff)-2_3_0
[... 41 lines cut ...]
### ADDITIONAL CONTEXT
09:51:48 SRC: WebConsole from 10.12.14.90
[... cut ...]
09:52:11 RSL: Gruppe 0, Index 1, Befehl RESET, Daten 1
09:52:11 MQT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
09:52:11 CFG: Standard verwenden
00:00:00 Projekt sonoff Sonoff Version 6.5.0(sonoff)-2_3_0
00:00:00 WIF: verbinden mit AP1 PayPerSer in Modus 11N wie sonoff-2127...
00:00:09 WIF: verbunden
00:00:09 HTP: Web-Server aktiv bei sonoff-2127 mit IP-Adresse xx.xx.xx.89
00:00:10 MQT: Verbindungsversuch...
00:00:10 MQT: verbunden

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-03-24:
Hi, Thanks for completing the template. :+1: 

In the gpio where you have set the button1, you have connected a pushbutton or a switch? 

Tasmota has a feature that if you keep pushed the button1 for more than 40 seconds, tasmota will reset all configs. If you have connected a switch, you should be using a switch as a gpio config and also you should select the switchmode that suits better your use case. 

Also, you can disable the **reset feature when pushing the button for more than 40 seconds** using the following commands in the Tasmota console:

`setoption1 1`

And

`setoption13 1`

Please, tell us if this solves your issue. Thanks

ascillato, 2019-03-25:
@FraatRamski

Hi,

Any news on this?

FraatRamski, 2019-03-26:
Is working, thanks for the help!

closed 2019-03-26 by FraatRamski

tanus10, 2020-07-29:
I know this is old but, just to note that this also solves Xiaomi-Philips Bulb continuous resetting problem. (tasmota 8.3.1)

marcoscv-work, 2024-02-11:
It also solves issues related to https://templates.blakadder.com/ESP32_Relay_X4.html