arendst/Tasmota #5508: Feature request: Option for default DispalyMode in my_user_config.h

wongnam, 2019-03-22:
(As i understand display feature did not have technical supported officially.)

Currently the default DisplayMode is always 1(date/time) after device reset to default.

I want to self-compile the fw dedicate use for my energy monitor(Wemos+PZEM004T+LCD1602 display). i usually have to type DisplayMode 2 in console to use for local sensor info, or displaymode 0 to use with my MQTT sent info. 

Please let us have an option to chose a default displaymode. or any workaround is ok. Thanks.

BTW, DisplayRows, DisplayCols, DisplayModel. Thanks.

curzon01, 2019-03-22:
workaround using [rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules):

```
Rule1 on System#Boot do DisplayMode 2 endon
Rule1 1
Rule1 4
```

wongnam, 2019-03-22:
Thanks @curzon01 . if this will be deleted after reset configuration by pressing button 40s?
I want to burn it in to ROM so that it will not be deleted after reset configuration.

curzon01, 2019-03-23:
Rules will not outlast reset device settings.

An alternative would be patch the function SettingsDefaultSet_5_10_1() in settings.ino line 840...

wongnam, 2019-03-23:
@curzon01 It was great, to me it was satisfying what I expected. Thanks so much for the solution.

closed 2019-03-23 by wongnam

ArjitJangid, 2022-12-22:
currently i have designed a display pattern of data in arduino how can i display similar data in tasmota like reading status of gpio and current time with ip to login