arendst/Tasmota #5509: Include (Wiki) the necessary config.yaml code for Home Assistant

QuantiumDev, 2019-03-22:
Wiki pages for supported devices such as [this one](https://github.com/arendst/Sonoff-Tasmota/wiki/LeFun-SK2-Smart-Power-Strip-(Model-ZLD-44USA-W)) detail how to configure the hardware device, but once configured, it can be difficult to figure out how to add it to Home Assistant. Even if you can find details on how to do so (based on the manufacturer and model), they probably won't apply since the firmware has changed.

If there could be a section on the Wiki pages for these devices that gives the yaml code for adding the device to Home Assistant's config.yaml, that would be immensely helpful.

I've asked for help on the HA community forum. so far, no one has answered.

Jason2866, 2019-03-22:
Since this is Tasmota Github and we are just a few people, we have no resources for supporting
all kind of devices for any Home Automation system.
Tasmota is a Open Source project. Your help would be highly appreciated. You are invited to add your findings to wiki. Please ask @blakadder (wiki mantainer) in Tasmota [Discord](https://discord.gg/Ks2Kzd4) how to be done.

blakadder, 2019-03-22:
We have a home assistant configuration guide [Home Assistant](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant).  
Also, the **Other devices** section is being phased out in favor of the template repository

ascillato2, 2019-03-22:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-22 by ascillato2

ascillato, 2019-03-22:
You can use Setoption19 1 for automatic discovery and you will not need to worry about configuration.yaml