arendst/Tasmota #5510: Can not upgrade from 6.4.1 to 6.5.0

mhutchy, 2019-03-22:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
When trying to upgrade from 6.4.1 to 6.5.0 on a sonoff basic I receive the following error: Upload Failed - Upload buffer miscompare

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [X] Tasmota binary firmware version number used : 6.4.1 / (pre-compiled or self-compiled ?)
- [X] Development IDE - Compiler / Upload tools used : python esptool.py / ____
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

{"Status":{"Module":1,"FriendlyName":["Blue lamp"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T12:32:55","StartupUTC":"2019-03-21T19:44:53","Sleep":50,"BootCount":4,"SaveCount":361,"SaveAddress":"FB000"},"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SFR_5380",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]},"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]},"StatusNET":{"Hostname":"sonoff-7860","IPAddress":"192.168.1.217","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:57:5E:B4","Webserver":2,"WifiConfig":4},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_575EB4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Fri Mar 22 08:17:48 2019","Local":"Fri Mar 22 09:17:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:49","Sunset":"19:05"},"StatusSNS":{"Time":"2019-03-22T09:17:48"},"StatusSTS":{"Time":"2019-03-22T09:17:48","Uptime":"0T12:32:55","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SFR_5380","BSSId":"44:CE:7D:B0:53:84","Channel":2,"RSSI":36}}}

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
Sonoff Basic Module
Blue lamp
Upload Failed

Upload buffer miscompare

Main Menu

Sonoff-Tasmota 6.4.1 by Theo Arends

```

### TO REPRODUCE
navigate to sonoff ip address
select firmware upgrade
select choose file
select new 6.5.0 firmware file sonoff.bin donwloaded from github 

### EXPECTED BEHAVIOUR
device should download firmware and reboot on new firmware

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

rqsnst, 2019-03-22:
have you tried to flash the minimal Version first and after this the wanted version?

> sonoff-minimal.bin = The Minimal version allows intermediate OTA uploads to support larger versions and does NOT change any persistent parameter. This version should NOT be used for initial installation.

refer to this [https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade](url)

mhutchy, 2019-03-22:
Hi,

Thanks for the reply, the minimal version fixed my issue.

Many thanks

closed 2019-03-22 by mhutchy

Ag3ntSn4k3, 2019-03-24:
> have you tried to flash the minimal Version first and after this the wanted version?
>
>
> refer to this [https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade](url)

Hi,
after that i could install the minimal update. but then i couldn't install the DE.bin. when uploading the file the sonoff got out and lost reboot/the WLAN connection and was reachable after knap 2-3 min again.
I tested it with Google Chrome version 73.0.3683.86 only.
After that I tried my own .bin which I usually use via serial. The message was not enough memory:-(.

Then I flashed it over Serial again without any problems.
But that can't be the solution, can it?