arendst/Tasmota #5511: change power will reboot device when MQTT enabled

apneef, 2019-03-22:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I'am using a SHP-6 with latest version (6.5.0, default version)

When i have MQTT enabled and want power On or Off via the gui off the device,
it is restarting?

If i turn of the MQTT function off, and then power On or Off via the device gui,
everyting is fine.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Blitwolf SHP6 with sonoff.bin
- [x] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

14:38:36 MQT: stat/SONOFF_01/STATUS = {"Status":{"Module":45,"FriendlyName":["SONOFF_01"],"Topic":"SONOFF_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:38:36 MQT: stat/SONOFF_01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:51","StartupUTC":"2019-03-22T13:37:45","Sleep":50,"CfgHolder":4617,"BootCount":29,"SaveCount":84,"SaveAddress":"F8000"}}
14:38:36 MQT: stat/SONOFF_01/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
14:38:36 MQT: stat/SONOFF_01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":4,"LogHost":"192.168.2.15","LogPort":514,"SSId":["WIIWIIWII",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
14:38:36 MQT: stat/SONOFF_01/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
14:38:36 MQT: stat/SONOFF_01/STATUS5 = {"StatusNET":{"Hostname":"SONOFF_01-4561","IPAddress":"192.168.2.28","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"CC:50:E3:2A:31:D1","Webserver":2,"WifiConfig":4}}
14:38:36 MQT: stat/SONOFF_01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.15","MqttPort":1883,"MqttClientMask":"SONOFF_01","MqttClient":"SONOFF_01","MqttUser":"SONOFF_01","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:38:36 MQT: stat/SONOFF_01/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 22 13:38:36 2019","Local":"Fri Mar 22 14:38:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:49","Sunset":"19:05"}}
14:38:36 MQT: stat/SONOFF_01/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:38:36 MQT: stat/SONOFF_01/STATUS10 = {"StatusSNS":{"Time":"2019-03-22T14:38:36","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Power":4,"ApparentPower":52,"ReactivePower":52,"Factor":0.08,"Voltage":245,"Current":0.213}}}
14:38:36 MQT: stat/SONOFF_01/STATUS11 = {"StatusSTS":{"Time":"2019-03-22T14:38:36","Uptime":"0T00:00:51","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
Mar 22 13:35:09 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T14:37:21","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Power":5,"ApparentPower":46,"ReactivePower":46,"Factor":0.10,"Voltage":245,"Current":0.187}}
Mar 22 13:35:11 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/STATE = {"Time":"2019-03-22T14:37:27","Uptime":"0T01:15:18","Vcc":3.174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:06"}}
Mar 22 13:35:11 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T14:37:27","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Period":0,"Power":3,"ApparentPower":48,"ReactivePower":48,"Factor":0.05,"Voltage":245,"Current":0.197}}
Mar 22 13:35:12 SONOFF_01-4561 ESP-WIF: Checking connection...
Mar 22 13:35:12 SONOFF_01-4561 ESP-WIF: Connected
Mar 22 13:35:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/STATE = {"Time":"2019-03-22T14:37:37","Uptime":"0T01:15:28","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:06"}}
Mar 22 13:35:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T14:37:38","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Period":0,"Power":4,"ApparentPower":46,"ReactivePower":46,"Factor":0.08,"Voltage":245,"Current":0.189}}

Mar 22 13:35:21 SONOFF_01-4561 ESP-SRC: WebGui from 192.168.2.249
Mar 22 13:35:25 SONOFF_01-4561 ESP-WIF: Connected
Mar 22 13:35:25 SONOFF_01-4561 ESP-HTP: Web server active on SONOFF_01-4561 with IP address 192.168.2.28
Mar 22 13:35:25 SONOFF_01-4561 ESP-APP: (UTC) Fri Mar 22 13:37:51 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: Attempting connection...
Mar 22 13:35:26 sonoff-loghost mosquitto: 1553258126: New connection from 192.168.2.28 on port 1883.
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: Connected
Mar 22 13:35:26 sonoff-loghost mosquitto: 1553258126: Client SONOFF_01 already connected, closing old connection.
Mar 22 13:35:26 sonoff-loghost mosquitto: 1553258126: Socket error on client SONOFF_01, disconnecting.
Mar 22 13:35:26 sonoff-loghost mosquitto: 1553258126: New client connected from 192.168.2.28 as SONOFF_01 (c1, k10, u'SONOFF_01').
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/LWT = Online (retained)
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: cmnd/SONOFF_01/POWER =
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/SONOFF_01/#
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/sonoffs/#
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/SONOFF_01_fb/#
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO1 = {"Module":"BlitzWolf SHP","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/SONOFF_01_fb/","GroupTopic":"sonoffs"}
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO2 = {"WebServerMode":"Admin","Hostname":"SONOFF_01-4561","IPAddress":"192.168.2.28"}
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO3 = {"RestartReason":"External System"}
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/RESULT = {"POWER":"ON"}
Mar 22 13:35:26 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/POWER = ON
Mar 22 13:35:27 SONOFF_01-4561 ESP-APP: Boot Count 29
Mar 22 13:35:27 SONOFF_01-4561 ESP-CFG: Saved to flash at F8, Count 84, Bytes 3584
[... 4 lines cut ...]
Mar 22 13:35:37 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T14:38:11","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Period":0,"Power":4,"ApparentPower":52,"ReactivePower":52,"Factor":0.07,"Voltage":245,"Current":0.213}}
Mar 22 13:35:37 SONOFF_01-4561 ESP-WIF: Checking connection...
Mar 22 13:35:37 SONOFF_01-4561 ESP-WIF: Connected
Mar 22 13:35:42 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/STATE = {"Time":"2019-03-22T14:38:21","Uptime":"0T00:00:36","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:06"}}
[... cut ...]
Mar 22 13:35:50 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/STATUS10 = {"StatusSNS":{"Time":"2019-03-22T14:38:36","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.102,"Yesterday":0.073,"Today":0.029,"Power":4,"ApparentPower":52,"ReactivePower":52,"Factor":0.08,"Voltage":245,"Current":0.213}}}
Mar 22 13:35:50 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/STATUS11 = {"StatusSTS":{"Time":"2019-03-22T14:38:36","Uptime":"0T00:00:51","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:06"}}}

Jason2866, 2019-03-22:
Could you try this version? http://thehackbox.org/tasmota/release/020402/sonoff.bin
It is the same just with core 2.4.2

Jason2866, 2019-03-22:
Do you have Belkin Emulation enabled?
See https://github.com/arendst/Sonoff-Tasmota/issues/5505#issuecomment-475468442

apneef, 2019-03-22:
@Jason2866 ,
from the gui version it works beter,
but now from MQTT point of view:

Mar 22 17:34:11 SONOFF_01-4561 ESP-WIF: Checking connection...
Mar 22 17:34:11 SONOFF_01-4561 ESP-WIF: Connected
Mar 22 17:34:13 SONOFF_01-4561 ESP-SRC: MQTT
Mar 22 17:34:13 SONOFF_01-4561 ESP-RSL: Received Topic cmnd/SONOFF_01/POWER, Data Size 2, Data ON
Mar 22 17:34:15 SONOFF_01-4561 ESP-WIF: Connected
Mar 22 17:34:15 SONOFF_01-4561 ESP-HTP: Web server active on SONOFF_01-4561 with IP address 192.168.2.28
Mar 22 17:34:16 SONOFF_01-4561 ESP-APP: (UTC) Fri Mar 22 20:00:23 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: Attempting connection...
Mar 22 17:34:16 sonoff-loghost mosquitto: 1553272456: Socket error on client SONOFF_01, disconnecting.
Mar 22 17:34:16 sonoff-loghost mosquitto: 1553272456: New connection from 192.168.2.28 on port 1883.
Mar 22 17:34:16 sonoff-loghost mosquitto: 1553272456: New client connected from 192.168.2.28 as SONOFF_01 (c1, k10, u'SONOFF_01').
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: Connected
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/LWT = Online (retained)
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: cmnd/SONOFF_01/POWER =
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/SONOFF_01/#
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/sonoffs/#
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: Subscribe to cmnd/SONOFF_01_fb/#
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO1 = {"Module":"BlitzWolf SHP","Version":"6.5.0(sonoff)","FallbackTopic":"cmnd/SONOFF_01_fb/","GroupTopic":"sonoffs"}
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO2 = {"WebServerMode":"Admin","Hostname":"SONOFF_01-4561","IPAddress":"192.168.2.28"}
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/INFO3 = {"RestartReason":"External System"}
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/RESULT = {"POWER":"OFF"}
Mar 22 17:34:16 SONOFF_01-4561 ESP-MQT: stat/SONOFF_01/POWER = OFF
Mar 22 17:34:17 SONOFF_01-4561 ESP-APP: Boot Count 10
Mar 22 17:34:17 SONOFF_01-4561 ESP-CFG: Saved to flash at FA, Count 34, Bytes 3584
Mar 22 17:34:19 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/STATE = {"Time":"2019-03-22T21:00:33","Uptime":"0T00:00:15","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":1,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:05"}}
Mar 22 17:34:19 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T21:00:33","ENERGY":{"TotalStartTime":"2019-03-22T20:51:56","Total":0.001,"Yesterday":0.000,"Today":0.001,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

I will try another MQTT broker.
Current is Mosquitto.

apneef, 2019-03-22:
> Do you have Belkin Emulation enabled?
> See [#5505 (comment)](https://github.com/arendst/Sonoff-Tasmota/issues/5505#issuecomment-475468442)

No,
Emulation is "none"

arendst, 2019-03-23:
Considering the fact that your restart reason is external system I suspect a bad power supply.

Jason2866, 2019-03-23:
If power supply is weak (not normal for this device!) You can try as last hope sonoff-basic.bin

apneef, 2019-03-25:
I did some test this weekend.
When i connect the device via the serial connection and no 230v, everything is working fine.
Wifi / MQTT is then no problem.

I have only 1 device, the others are coming. So for know i think it is in the device's power or something.
When i have the others, i will continue the test.

closed 2019-03-25 by apneef

R0Wi, 2019-11-03:
@apneef i have a quite similar problem with the GOSUND SP111 and Sonoff-Tasmota 6.6.0.7: if i toggle the power switch via the web gui without any device connected, everything is fine. But if a device is connected and i turn power off, the switch usually reboots and power is going back to "ON" after a few milliseconds. I also see the same log entry
```
00:00:05 MQT: /tele/myhome/mysocket/INFO3 = {"RestartReason":"External System"}
```
that you mentioned.

Did you manage to fix this issue somehow?

Jason2866, 2019-11-03:
Update to latest release version. 6.7.1
https://github.com/arendst/Tasmota/releases/tag/v6.7.1

R0Wi, 2019-11-03:
Worked for me. Just updated from the sonoff-basic 6.6.0 (core 2.3.0) via webgui OTA to the latest sonoff (full) 6.7.1 (core pre 2.6) and it works like charm. Thanks a lot :-)

joconcepts, 2019-11-03:
I've got these power sockets:
https://www.amazon.de/gp/product/B07H77M7G8/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1
They are the same as Gosund SP111 (seems so at least) and I've updated them to the latest tasmota version. I've always have the issue that after I press a couple of times the toggle button in the web UI I can hear the relay clicking but not completely as there is no power to the socket and the thing reboots completely. I can reproduce that on all of my sockets (I have 7 of them).
Any one a idea? I already thought that the load of the LED bulb I'm switching is to low or strange but even with a 25w classic light bulb it happens.

R0Wi, 2019-11-03:
@joconcepts which version of sonoff-tasmota do you use? Do you use the basic or full image?

joconcepts, 2019-11-03:
basic

R0Wi, 2019-11-03:
And which version number?

joconcepts, 2019-11-03:
Program Version	6.7.1(basic)
Build Date & Time	2019-10-26T13:20:09
Core/SDK Version	STAGE/2.2.2-dev(38a443e)

joconcepts, 2019-11-03:
Just to mention: I also get the restart reason "external system" when the issue happens.

R0Wi, 2019-11-03:
In my case the update to the current full version (http://thehackbox.org/tasmota/release/sonoff.bin) fixed the issue. I've done this by going to Web GUI "Firmware upgrade", filling in the mentioned url  at OTA url and pressed on Start upgrade.

The version above uses a newer core by the way.

joconcepts, 2019-11-03:
I've tried that and it doesn't work in my case, same issue...
```
Program Version	6.7.1(sonoff)
Build Date & Time	2019-10-26T13:19:13
Core/SDK Version	STAGE/2.2.2-dev(38a443e)
```

ascillato, 2019-11-04:
external system means that your device was left without energy and re enegized again. So, if you are doing nothing to your device, this means that the power regulator ia faulty in your device. In that case, as is a hardware issue, there is nothing that can be done from Tasmota Software side. Sorry.

R0Wi, 2019-11-08:
@joconcepts i have to revoke my statement. Even with the latest firmware flashed the error still exists sometimes on my Gosund SP111. If there is no device connected to the socket everything works well. But if there is one connected the chance is quite high that the device still is left without energy and therefore does a restart when it's powered again.

I don't know if this could be a general issue with the SP111? Does anyone also have this issue?

R0Wi, 2019-11-19:
In the meantime i managed it that the SP111 works properly. Therefore i used a Blitzwolf BW-SHP6 Template, which is based on module no. 45 instead of the generic 18 (https://blakadder.github.io/templates/blitzwolf_SHP6.html). The bad thing: after a few hours the socket completely stopped working (wasn't reachable, didn't connect to wifi, no led lights...). So i ended up in sending the items back and bought some Gosund SP112 which are working perfectly for many days now. I use this template: https://blakadder.github.io/templates/gosund_SP112.html .

dpjrodrigues, 2019-12-10:
Hi all,
I have the same issue and this is not MQTT related. You may think that when disabeling MQTT is not happennig because you don't see the MQTT restart  messages but if you put log to debug or pay atention to the console you'll see that it also restarts (at least in my case).

I've test two SHP6, different tasmota versions and they both did the same. The reallity is that I never seen this impacting me until I've tried to used the cheap christmas lights with sound  and I think it is related. 
with a normal light bulb I need to turn on and off probably 5 ou 6 times in a row to happen (which is not a realistic use case) while with the cheap christmas lights with sound it is like 40-50% chance of happening so I guess it is related with some power sensitivity