arendst/Tasmota #5512: syslog log not complete where MQTT off

apneef, 2019-03-22:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I am using a SHP6.
I have enable syslog to remote host.

When via console getting status of all, then in the console the status lines are displayed until status11.
But on the syslog host, there is only until status3

If i turn MQTT on, the status is being displayed correct in the syslog server.
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Blitwolz SHP6 - sonoff basic
- [x] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :

STATUS 0 OUTPUT HERE:

15:15:21 CMD: status 0
15:15:21 SRC: WebConsole from 192.168.2.249
15:15:21 RSL: Received Topic /status, Data Size 1, Data 0
15:15:21 RSL: Group 0, Index 1, Command STATUS, Data 0
15:15:21 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["SONOFF_01"],"Topic":"SONOFF_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:15:21 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:06:30","StartupUTC":"2019-03-22T14:08:51","Sleep":50,"CfgHolder":4617,"BootCount":31,"SaveCount":90,"SaveAddress":"FA000"}}
15:15:21 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:15:21 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"192.168.2.15","LogPort":514,"SSId":["WIIWIIWII",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
15:15:21 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
15:15:21 RSL: STATUS5 = {"StatusNET":{"Hostname":"SONOFF_01-4561","IPAddress":"192.168.2.28","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"CC:50:E3:2A:31:D1","Webserver":2,"WifiConfig":4}}
15:15:21 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 22 14:15:21 2019","Local":"Fri Mar 22 15:15:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:49","Sunset":"19:05"}}
15:15:21 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:15:21 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-03-22T15:15:21","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":1,"ApparentPower":49,"ReactivePower":49,"Factor":0.02,"Voltage":245,"Current":0.201}}}
15:15:21 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-03-22T15:15:21","Uptime":"0T00:06:30","Vcc":3.168,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":42,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

15:14:58 WIF: Checking connection...
15:14:58 WIF: Connected
15:15:05 RSL: SENSOR = {"Time":"2019-03-22T15:15:05","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":4,"ApparentPower":51,"ReactivePower":51,"Factor":0.07,"Voltage":245,"Current":0.210}}
15:15:05 HTP: Console
15:15:07 RSL: SENSOR = {"Time":"2019-03-22T15:15:07","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":2,"ApparentPower":56,"ReactivePower":56,"Factor":0.03,"Voltage":245,"Current":0.230}}
15:15:07 RSL: STATE = {"Time":"2019-03-22T15:15:07","Uptime":"0T00:06:16","Vcc":3.170,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":42,"LinkCount":1,"Downtime":"0T00:00:04"}}
15:15:11 RSL: SENSOR = {"Time":"2019-03-22T15:15:11","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":2,"ApparentPower":50,"ReactivePower":50,"Factor":0.03,"Voltage":245,"Current":0.205}}
15:15:16 CMD: status -
15:15:16 SRC: WebConsole from 192.168.2.249
15:15:16 RSL: Received Topic /status, Data Size 1, Data -
15:15:16 RSL: Group 0, Index 1, Command STATUS, Data -
15:15:16 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["SONOFF_01"],"Topic":"SONOFF_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:15:16 RSL: SENSOR = {"Time":"2019-03-22T15:15:16","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":3,"ApparentPower":55,"ReactivePower":55,"Factor":0.05,"Voltage":245,"Current":0.223}}
15:15:18 RSL: SENSOR = {"Time":"2019-03-22T15:15:18","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":1,"ApparentPower":52,"ReactivePower":52,"Factor":0.02,"Voltage":245,"Current":0.214}}
15:15:19 WIF: Checking connection...
15:15:19 WIF: Connected
15:15:19 RSL: SENSOR = {"Time":"2019-03-22T15:15:19","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":4,"ApparentPower":51,"ReactivePower":51,"Factor":0.08,"Voltage":245,"Current":0.209}}
15:15:21 CMD: status 0
15:15:21 SRC: WebConsole from 192.168.2.249
15:15:21 RSL: Received Topic /status, Data Size 1, Data 0
15:15:21 RSL: Group 0, Index 1, Command STATUS, Data 0
15:15:21 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["SONOFF_01"],"Topic":"SONOFF_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[... 18 lines cut ...]
Mar 22 15:15:03 SONOFF_01-4561 ESP-RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"192.168.2.15","LogPort":514,"SSId":["WIIWIIWII",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
Mar 22 15:15:06 SONOFF_01-4561 ESP-RSL: SENSOR = {"Time":"2019-03-22T15:15:25","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":2,"ApparentPower":52,"ReactivePower":52,"Factor":0.03,"Voltage":245,"Current":0.212}}
Mar 22 15:15:07 SONOFF_01-4561 ESP-RSL: SENSOR = {"Time":"2019-03-22T15:15:27","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.030,"Power":4,"ApparentPower":51,"ReactivePower":51,"Factor":0.08,"Voltage":244,"Current":0.210}}

Rsyslog with MQTT On:
Mar 22 15:20:47 SONOFF_01-4561 ESP-HTP: Console
Mar 22 15:20:49 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T15:24:32","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.031,"Power":1,"ApparentPower":48,"ReactivePower":48,"Factor":0.03,"Voltage":245,"Current":0.196}}
Mar 22 15:20:50 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T15:24:33","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.031,"Power":3,"ApparentPower":48,"ReactivePower":48,"Factor":0.07,"Voltage":245,"Current":0.196}}
Mar 22 15:20:51 SONOFF_01-4561 ESP-CMD: status 0
[... cut ...]
Mar 22 15:20:51 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/STATE = {"Time":"2019-03-22T15:24:35","Uptime":"0T00:00:16","Vcc":3.166,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"00:26:F2:15:08:46","Channel":6,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:06"}}
Mar 22 15:20:51 SONOFF_01-4561 ESP-MQT: tele/SONOFF_01/SENSOR = {"Time":"2019-03-22T15:24:35","ENERGY":{"TotalStartTime":"2019-03-21T20:12:32","Total":0.104,"Yesterday":0.073,"Today":0.031,"Period":0,"Power":4,"ApparentPower":39,"ReactivePower":39,"Factor":0.11,"Voltage":244,"Current":0.161}}

arendst, 2019-03-23:
I could reproduce. By adding a delay all syslog messages were received as expected.

Pls have a go at the latest dev release and report back your result.

apneef, 2019-03-23:
syslog is now as expected:

Mar 23 15:44:31 SONOFF_01-4561 ESP-RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:00:23","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":19,"SaveAddress":"F9000"}}
Mar 23 15:44:31 SONOFF_01-4561 ESP-RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.1(76b7d77-sonoff)","BuildDateTime":"2019-03-23T14:02:32","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
Mar 23 15:44:31 SONOFF_01-4561 ESP-RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"192.168.2.37","LogPort":514,"SSId":["WIIWIIWII",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS5 = {"StatusNET":{"Hostname":"SONOFF_01-4561","IPAddress":"192.168.2.28","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"CC:50:E3:2A:31:D1","Webserver":2,"WifiConfig":4}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:23 1970","Local":"Thu Jan 01 00:00:23 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:23","ENERGY":{"TotalStartTime":"2019-03-23T16:41:18","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
Mar 23 15:44:32 SONOFF_01-4561 ESP-RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:23","Uptime":"0T00:00:23","Vcc":3.430,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WIIWIIWII","BSSId":"B4:FB:E4:F4:04:DF","Channel":11,"RSSI":44,"LinkCount":1,"Downtime":"0T00:00:08"}}}

Thanks for the quick fix.

closed 2019-03-23 by apneef