arendst/Tasmota #5516: Reverse blind direction/toggle interlocked relays

Crazor, 2019-03-23:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
If I enable the interlock feature to control my window blinds motor, and the blinds are currently moving up, and while still moving, a down command is issued, the relays should both toggle immediately to reverse direction.
Right now, if I turn on the up relay and change my mind while the motor is running, and I press the down button, nothing happens, because the interlock feature decides not to turn on the second relay. Instead, the up relay should be turned off and the down relay on.

**Describe alternatives you've considered**
I can implement this with NodeRED via MQTT, but I also want this feature for the buttons in the web interface, for rule usage and for directly connected buttons.

Jason2866, 2019-03-23:
Have you looked at this fork? https://github.com/stefanbode/Sonoff-Tasmota
It has support for Shutter use-Case.

ascillato2, 2019-03-30:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-30 by ascillato2

Crazor, 2019-04-22:
Well, I know about the fork, but it doesn't work on Shelly 2.5 yet.

However, either I had something misconfigured before, or the behaviour has changed recently. I have updated my Sonoff Duals a couple of days ago and after enabling the interlocking feature, the blinds reverse direction as they should.