arendst/Tasmota #5517: TSL2561 & tasmota 6.50: reports "Analog0" beside "TSL2561 Illuminance"

heikoh81, 2019-03-24:
Hi,

I'm using Tasmota 6.50 sensors on Wemos D1 Mini with TSL2561 light sensor.
Beside "TSL2561 Illuminance" I also have an entry "Analog0".
Is this a bug?
I can not see the entry "Analog0" in the Wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-TSL2561-Luminosity-Sensor

Greetings,
Heiko
[image]
[image]

Jason2866, 2019-03-24:
As designed. Sonoff sensors bin activates Analog 0 input.

heikoh81, 2019-03-24:
Ok, thanks for the hint.
Maybe this should be added to the Wiki, and the screenshot should be updated.
In the wiki, the screenshot shows no Analog0.

Jason2866, 2019-03-24:
It is in release notes https://github.com/arendst/Sonoff-Tasmota/releases
Everyone can edit wiki, you are invited to edit :-)

meingraham, 2019-03-24:
If you create a template for your device (rather than just configuring it as a Generic module), you can control whether A0 is "active" and can eliminate it from the web GUI and from the Sensor telemetry messages.

https://github.com/arendst/Sonoff-Tasmota/wiki/Templates
https://github.com/arendst/Sonoff-Tasmota/wiki/Templates#ADC0

Mike

heikoh81, 2019-03-24:
My first try went bad, I guess.
Probably because I only have Edge on this machine.
Can't roll back history, either.

meingraham, 2019-03-24:
You had inadvertently switched the page to AsciiDoc. I reverted to markdown and updated the page to add the ADC0 information.

Please close this issue.

ascillato2, 2019-03-24:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-24 by ascillato2

heikoh81, 2019-03-24:
This was not a duplicate, the question was completely different (possible bug Vs. feature question).
However, both issues have been solved.

ascillato, 2019-03-25:
>This was not a duplicate

Sorry. It is a duplicated issue from old issues. If you search in issues, you will find that several times people had asked about analog0. Please, next time follow the template. That helps a lot.