arendst/Tasmota #5519: %06X isnt working for #define MQTT_TOPIC

Jason2866, 2019-03-24:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
using `#define MQTT_TOPIC             PROJECT"-%06X"` in my_user_config doesnt result in what it should (replacing with 4 last digits from MAC address)

[image]

[image]

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : _____
- [ x] Tasmota binary firmware version number used : 6.5..0.1 / self-compiled
- [ x] Development IDE - Compiler / Upload tools used : VSC / VSC
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
19:31:15 CMD: status 0
19:31:15 MQT: stat/sonoff-53CF5F/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff-53CF5F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"External System","Uptime":"0T00:11:36","StartupUTC":"2019-03-24T18:19:39","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":10,"SaveAddress":"F6000"}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS2 = {"StatusFWR":{"Version":"6.5.0.1(sonoff)","BuildDateTime":"2019-03-24T19:17:38","Boot":31,"Core":"STAGE","SDK":"2.2.1(cfd48f3)"}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F8AA384","040403A1","80041496","001028C2"]}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-53CF5F-3935","IPAddress":"192.168.2.169","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:53:CF:5F","Webserver":2,"WifiConfig":5}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-53CF5F","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 24 18:31:15 2019","Local":"Sun Mar 24 19:31:15 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:07","Sunset":"18:30"}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS10 = {"StatusSNS":{"Time":"2019-03-24T19:31:15"}}
19:31:15 MQT: stat/sonoff-53CF5F/STATUS11 = {"StatusSTS":{"Time":"2019-03-24T19:31:15","Uptime":"0T00:11:36","Vcc":2.952,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Replace %06X with last 4 digits from MAC address as in MQTT Client

pgollor, 2019-03-24:
> Sometimes devices hangs completly and is unresponsible!!

My Sonoff Touch EU, Sonoff, Basic, Sonoff Dual, Sonoff Power, Sonoff S20 and Sonoff T1 runs with version 6.4.1 without problems. After update to 6.5.0 some devices fall back to default MQTT topic and other devices hang and restart immediately without error notification over serial after kicking on `MQTT config`. Maybe this is the same problem?

I tried to flash a clean version (erase all flash) but i get the same problem with `#define MQTT_TOPIC "%06X"`.
After change the `MQTT_TOPIC` to `abc` everything works fine.

0n3-70uch, 2019-03-25:
+1

Hello, I can confirm this issue.
Greetings Fabi

wongnam, 2019-03-25:
> ### BUG DESCRIPTION
> using `#define MQTT_TOPIC PROJECT"-%06X"` in my_user_config doesnt result in what it should (replacing with 6 last digits from MAC address)
> Sometimes devices hangs completly and is unresponsible!!

@Jason2866 confirm, -%06X no working but no hangs.
i saw you have Hue Bridge Emulate Enable, please try to disable that it will help to reduce system hangs in this case.

arendst, 2019-03-25:
With a project name of "sonoff" one would expect to see a topic of "sonoff-123456". From your screenshots it shows it does just that: it has changed topic to sonoff-53CF5F where the 6 hex digits were derived from the MAC address.

I've tested it myself and it works as expected.

I do not see the issue here.

0n3-70uch, 2019-03-25:
Hi, please check mqtt_topic not mqtt_client.

arendst, 2019-03-25:
I did:
[image]

arendst, 2019-03-25:
A different issue is this:
[image]

most likely as a result of the %06X in the topic name. That I will investigate.

Jason2866, 2019-03-25:
But why does it hung?

arendst, 2019-03-25:
Probably because of the same reason as the previous response; corruption due to mishandling of % by my latest code changes (or legacy code as I've never tested a topic name before with %06X in it ;-).

Will investigate.

arendst, 2019-03-25:
Further investigating...

Jason2866, 2019-03-25:
Can confirm, i have no hangs anymore too. Side effect from something.....
Thx for fixing GUI
[image]
[image]

closed 2019-03-25 by Jason2866

pgollor, 2019-03-25:
Thanks, it works fine with the current dev branch.

ViniciusDS, 2020-02-20:
I have the same problem

ViniciusDS, 2020-02-20:
I have the same problem in version 8.1.0.8

ariekraakjr, 2020-07-14:
Same problem in 8.3.1(sensors)

added a PIR with:
SwitchMode1 1
SwitchTopic 0
Rule1 on Switch1#state=1 do publish stat/%topic%/PIR1 ON endon on Switch1#state=0 do Publish stat/%topic%/PIR1 OFF endon
Rule1 1

MQTT Host	192.168.178.254
MQTT Port	1883
MQTT User	DVES_USER
MQTT Client	DVES_AC6EE9
MQTT Topic	tasmota_%06X
MQTT Group Topic 1	cmnd/tasmotas/
MQTT Full Topic	cmnd/tasmota_AC6EE9/
MQTT Fallback Topic	cmnd/DVES_AC6EE9_fb/

23:47:00 MQT: tele/tasmota_AC6EE9/STATE = {"Time":"2020-07-14T23:47:00","Uptime":"0T00:03:19","UptimeSec":199,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 7581 VR","BSSId":"9C:C7:A6:CD:16:30","Channel":4,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}
23:47:00 MQT: tele/tasmota_AC6EE9/SENSOR = {"Time":"2020-07-14T23:47:00","Switch1":"OFF","BME280":{"Temperature":21.4,"Humidity":60.8,"DewPoint":13.5,"Pressure":1014.6},"PressureUnit":"hPa","TempUnit":"C"}
23:47:03 RUL: SWITCH1#STATE=1 performs "publish stat/tasmota_%06X/PIR1 ON"
23:47:03 MQT: stat/tasmota_%06X/PIR1 = ON