arendst/Tasmota #5520: Full IR receive Signal

botak13, 2019-03-24:
Hi
I have flashed a IR Remote with Tasmota and want to control via Web Browser. I receive the IR Signal in RAW Format. Some of them end with ... I have set the IR_RCV_BUFFER_SIZE 1000 and IR_RCV_TIMEOUT 50 and MQTT_MAX_PACKET_SIZE 2000 but i still dont get the complete Signal on some Remote Buttons. Anything else i can do to get the full Button Signal?

Chat could not help me.

Jason2866, 2019-03-24:
Take a look in this thread https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-450535745

botak13, 2019-03-24:
I have changed.

my_user_config.h
IR_RCV_BUFFER_SIZE 1000
IR_RCV_TIMEOUT 50
PubSubClient.h and sonoff_post.h
MQTT_MAX_PACKET_SIZE 2000

and

SetOption58 1

But the String still end with ...

Jason2866, 2019-03-25:
You havent read the comment.
You have to change https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_05_irremote.ino#L533
`uint8_t i = 0;` into `uint16_t i = 0;` because this is in original only 8 bit (0-255) which isnt enough for your use case.  With this change you have 16 bits (0-65535) for this variable

botak13, 2019-03-25:
Thank you for you help.

Now i receive.

`19:45:51 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":46,"Data":2688555987,"RawData":[418,444,396,1254,416,446,394,1254,420,444,392,446,394,1252,422,442,396,442,396,1252,422,1254,420,442,394,444,396,444,394,440,400,438,420,418,396,442,396,442,396,1252,422,440,396,1252,424,440,398,442,418,416,396,1254,422,442,394,1252,422,1254,420,442,398,440,394,1254,420,1254,392,470,394,1254,422,442,394,442,396,1252,420,444,396,1254,416,1256,422,1254,418,1254,420,442,422,1226,424],"RawDataInfo":[91,91,0]}}`

and

`19:55:31 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"PANASONIC","Bits":48,"Data":42677437,"RawData":[3308,1720,416,424,414,422,416,1258,416,424,414,1260,416,424,414,1260,416,422,416,424,414,1262,416,422,414,422,416,1264,410,1260,416,422,416,424,414,422,416,424,414,422,416,424,414,424,414,424,416,1258,418,424,414,1260,416,424,416,424,414,422,414,1260,416,424,416,1260,416,1260,416,424,416,422,416,1260,416,1262,414,422,416,1262,414,422,416,422,416,1260,414,424,414,1260,416,1260,416,1260,416,1260,414,428,412,1260, ...`

on the same Remote Button but i can not get them work with irsend..

Jason2866, 2019-03-26:
> on the same Remote Button but i can not get them work with irsend..

Could you be more specific? What have you tried? LOGs? Error messages?....

botak13, 2019-03-26:
I flashed a Tuya IR remote on Tasmota because I want to control my Denon AVR-1911 with it. For ON and OFF, I received the correct RAW string in the console. But if I want to switch the channel then I get either the Unknown or the Panasonic string. Both do not work.

I am not so well versed in the field so I need a few think impulses.

botak13, 2019-03-27:
> Received string will NOT shown completly in web frontend, if it works you get a mqtt message with 0 at the end as indicator everything is fully received.

botak13, 2019-03-27:
> 
> 
> You havent read the comment.
> You have to change https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_05_irremote.ino#L533
> `uint8_t i = 0;` into `uint16_t i = 0;` because this is in original only 8 bit (0-255) which isnt enough for your use case. With this change you have 16 bits (0-65535) for this variable
> No gurantee this works, maybe you get a stack overflow...
> Received string will NOT shown completly in web frontend, if it works you get a mqtt message with 0 at the end as indicator everything is fully received. [#2116 (comment)](https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-440716483)

The Panasonic String still ends with ...
Where can I get the full String?

Thank you.

ascillato2, 2019-03-30:
@botak13 

Hi, 

Any news on this?

botak13, 2019-03-31:
No because I cant log in on my Linux Maschine to log the IR.
Its not that easy to get the right codes.

Do you have a solution?

StalkerSVR, 2019-04-02:
Denon use NEC protocol. Today I also flashed Tuya IR remote. So, you must find lirc confit files for denon (try find lirc denon on google) and send  codes from that config. It works fine for my Denon.

ascillato2, 2019-04-02:
>No because I cant log in on my Linux Maschine to log the IR.

Ok, no problem. Closing this issue until you have the time to provide more information and the template. Thanks

closed 2019-04-02 by ascillato2

botak13, 2019-04-06:
I found a solution. First I downloaded the 2015 Denon IR Command Chart on usa.denon.com for my AVR.
I saved the file with openoffice in a * .cvs file.
I could import these into IrScrutinizer and export them as RAW.
For the result, I started with 0, and separated every number with a ,
Now I can control my Denon AVR!

Maybe it helps someone ..

NightDragon1, 2020-07-08:
> I found a solution. First I downloaded the 2015 Denon IR Command Chart on usa.denon.com for my AVR.
> I saved the file with openoffice in a * .cvs file.
> I could import these into IrScrutinizer and [...]

I was able to load the csv too but how to you proceed to get a RAW command or whatever to feed it into tasmota? I haven't managed to generate a bit stream or similar now...
Can you please give me an example command and how you transformed it?

Thx!

botak13, 2020-07-09:
Have you tried out IrScrutinizer?
Import the csv and then generate a RAW.

MangaValk, 2020-07-10:
> The Panasonic String still ends with ...
> Where can I get the full String?
>
> Thank you.

cant find this line anymore.
still have same issue with raw value being clipped.
any advice?

Jason2866, 2020-07-10:
Read the comment from Theo what you have to change
https://github.com/arendst/Tasmota/issues/2116#issuecomment-440720126

MangaValk, 2020-07-10:
thanks! figured out that console always truncates but in mqttlens it showed the whole message :D