arendst/Tasmota #5521: mqtt json has a weird character in channel

blakadder, 2019-03-24:
### BUG DESCRIPTION
_A clear and concise description of what the bug is._

STATUS json shows `"Channel":[0,0,0,0,26]` and thusly created errors when trying to parse the json value

Repeatable on another device too `"Channel":[100,49,0]`

STATUS 0 OUTPUT HERE:
```
23:40:06 CMD: status 0
23:40:06 MQT: stat/h801/STATUS = {"Status":{"Module":20,"FriendlyName":["Led Strip"],"Topic":"h801","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:40:06 MQT: stat/h801/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:43","StartupUTC":"2019-03-24T22:26:23","Sleep":100,"CfgHolder":4617,"BootCount":117,"SaveCount":927,"SaveAddress":"FA000"}}
23:40:06 MQT: stat/h801/STATUS2 = {"StatusFWR":{"Version":"6.5.0.1(10d32ac-sonoff)","BuildDateTime":"2019-03-24T16:02:46","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
23:40:06 MQT: stat/h801/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OpenWrt","HTPC"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028029","280500000100000000000000000000000000","00000248"]}}
23:40:06 MQT: stat/h801/STATUS4 = {"StatusMEM":{"ProgramSize":549,"Free":452,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
23:40:06 MQT: stat/h801/STATUS5 = {"StatusNET":{"Hostname":"h801-5717","IPAddress":"10.1.1.147","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"DC:4F:22:4B:D6:55","Webserver":2,"WifiConfig":3}}
23:40:06 MQT: stat/h801/STATUS6 = {"StatusMQT":{"MqttHost":"openwrt","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4BD655","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:40:06 MQT: stat/h801/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 24 22:40:06 2019","Local":"Sun Mar 24 23:40:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"18:01"}}
23:40:06 MQT: stat/h801/STATUS10 = {"StatusSNS":{"Time":"2019-03-24T23:40:06"}}
23:40:06 MQT: stat/h801/STATUS11 = {"StatusSTS":{"Time":"2019-03-24T23:40:06","Uptime":"0T00:13:43","Vcc":3.532,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"POWER":"OFF","Dimmer":27,"Color":"0,0,0,0,68","HSBColor":"0,0,0","Channel":[0,0,0,0,26],"CT":500,"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":2,"SSId":"HTPC","BSSId":"50:64:2B:5B:41:59","Channel":10,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

after downgrading to 
```
6.5.0.1(76149ca-sonoff)
2019-03-20T10:01:22
2_4_2/2.2.1(cfd48f3)
```
the bug is gone so it was introduced in the last few days

arendst, 2019-03-25:
Thx for testing and defining the small window of opportunity ;-)

Jason2866, 2019-03-25:
Tested, JSON string is correct

blakadder, 2019-03-25:
Fixed

closed 2019-03-25 by blakadder