arendst/Tasmota #5522: Usage of %topic% in rules

to-scho, 2019-03-25:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
One need device specific rules to have consistent MQTT topic namings like "stat/my-sonoff/my-stat when publishing MQTT in rules. I have 10+ devices using same rules and run out of var/mem when working around. 

**Describe the solution you'd like**
Allow usage of %topic%, %prefix" and %grouptopic% in rules.

**Describe alternatives you've considered**
Store %topic% in e.g. %mem1% allows same rules and easy update for a bunch of devices.

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

pgollor, 2019-03-25:
Maybe it could be the same issue like https://github.com/arendst/Sonoff-Tasmota/issues/5519

arendst, 2019-03-25:
Storing a topic in %mem% won't be implemented any time soon as currently mem variables allow up to 10 characters while %topic% could be up to 33 characters. Mems were meant for numbers only.

to-scho, 2019-03-25:
That's why I currently need multiple mems to store topic. Sorry to not been precise enough, request is not to store %topic% in mem but to allow usage of %topic% in publish command.

il300, 2019-04-07:
This functionality is also needed. I have several identical devices with the same rules. It is necessary to distinguish them. The best way through the settings. Different rules are not the best choice.

davidgurr, 2019-04-10:
Likewise - would be very useful for the same reason as mentioned by @il300

andikt, 2019-04-26:
Yes, please, this would be a valuable feature. I also have multiple devices and want to use the same rules...

gemu2015, 2019-04-30:
have a look at this tread =>https://github.com/arendst/Sonoff-Tasmota/issues/5689

its easy with a script

Jason2866, 2019-05-19:
Please close this issue. Script is supported now by Tasmota

ascillato2, 2019-05-19:
Closing this issue as there is now a workaround using scripting. Please, check the documentation for an example for doing this. Thanks

closed 2019-05-19 by ascillato2

ascillato2, 2019-12-04:
Proposed PR #7106 for adding %topic% as a new variable for rules.

PR #7106 (2019-12-04): Added new var %topic% for rules
## Description:

Added new variable `%topic%` for rules as this has been requested several times before.

Example:
```yaml
17:07:51 CMD: topic
17:07:51 MQT: stat/living/RESULT = {"Topic":"living"}
17:07:53 CMD: rule
17:07:53 MQT: stat/living/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":460,"Rules":"on event#test do publish cmnd/%topic%/power 2 endon"}
17:07:57 CMD: event test
17:07:57 MQT: stat/living/RESULT = {"Event":"Done"}
17:07:57 RUL: EVENT#TEST performs "publish cmnd/living/power 2"
17:07:57 MQT: cmnd/living/power = 2
17:07:57 MQT: stat/ [... cut]