arendst/Tasmota #5523: Couldn't send IR CODE with IRSEND Command

helmar74, 2019-03-25:
I managed to successfully flash the YTF  IR Bridge with Tasmota.
If I send a IR Code from a IR Remote it is successfully received by the Bridge:
`19:38:40 MQT: tele/sonoffIRBridge_1/RESULT = {"IrReceived":{"Protocol":"RC6","Bits":36,"Data":"0x8052100C"}}`

IF I try to send the same code to the Device, which should be controlled (TV), the Device doesn't react, although the Code is sent to Device:
`IRsend {"Protocol":"RC6","Bits":36,"Data":0x8052100C}
`

```
19:48:32 CMD: IRsend {"Protocol":"RC6","Bits":36,"Data":0x8052100C}
19:48:33 MQT: stat/sonoffIRBridge_1/RESULT = {"IRSend":"Done"}
```

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : _YTF IR Bridge____
- [X] Tasmota binary firmware version number used : _6.5.0_ / ( self-compiled ?)
- [X] Development IDE - Compiler / Upload tools used : ARDUINO____ / ____
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
12:40:35 MQT: tele/sonoffIRBridge_1/STATE = {"Time":"2019-03-25T12:40:35","Uptime":"0T01:10:14","Vcc":3.394,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:43:05 CMD: status 0
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS = {"Status":{"Module":62,"FriendlyName":["Sonoff IR Bridge 1"],"Topic":"sonoffIRBridge_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://10.1.1.50:8080/static/firmware/sonoffStandard.ino.bin","RestartReason":"Power on","Uptime":"0T01:12:44","StartupUTC":"2019-03-25T10:30:21","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":20,"SaveAddress":"F8000"}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019.03.20 13:18:35","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","0FAAE19C","00078FA0","23BA1FCE","002033C0"]}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS5 = {"StatusNET":{"Hostname":"sonoffIRBridge-1","IPAddress":"10.1.1.160","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.2","Mac":"EC:FA:BC:20:81:36","Webserver":2,"WifiConfig":4}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.50","MqttPort":1883,"MqttClientMask":"sonoffIRBridge-1","MqttClient":"sonoffIRBridge-1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 25 11:43:05 2019","Local":"Mon Mar 25 12:43:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:01","Sunset":"18:28"}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS10 = {"StatusSNS":{"Time":"2019-03-25T12:43:05"}}
12:43:05 MQT: stat/sonoffIRBridge_1/STATUS11 = {"StatusSTS":{"Time":"2019-03-25T12:43:05","Uptime":"0T01:12:44","Vcc":3.396,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-25:
Try with a (old smartphone) camera if you see the led flashing from the IR Bridge.
You can test with the orig. Remote if you see the LED flickering with.

ascillato2, 2019-03-30:
Closing this issue as there is no feedback. Please, when you have time, tell us if you could try what you were asked to test. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

roelleke, 2019-03-31:
I have exactly the same Problem. I tried to send RC6 codes and use the IRremote libary example sketch for receiving the signal. I receive a malformt signal. I believe there is a bug in the tasmota Software. The IRremote mqtt example sketch work fine and i use this sketch in the moment.

Jason2866, 2019-03-31:
@roelleke Please provide all neccessary infos (issue template) and your example.
Without we cant do anything...

arendst, 2019-03-31:
Just did a quick test with a YTF IR bridge as sender and a wemos like device as IR receiver.

Send from YTF:
```
12:31:04 CMD: IRsend {"Protocol":"RC6","Bits":36,"Data":0x8052100C}
12:31:04 SRC: WebConsole from 192.168.2.1
12:31:04 RSL: Group 0, Index 1, Command IRSEND, Data {"Protocol":"RC6","Bits":36,"Data":0x8052100C}
12:31:04 IRS: protocol_text RC6, protocol RC6, bits 36, data 2152861708 (0x8052100C), protocol_code 2
12:31:04 MQT: stat/bridgeir/RESULT = {"IRSend":"Done"}
```
Received on wemos like device:
```
13:31:03 MQT: tele/ledunia2/RESULT = {"IrReceived":{"Protocol":"RC6","Bits":36,"Data":"0x8052100C","RawData":[2664,864,488,878,494,404,486,408,492,850,1372,864,486,408,482,418,492,400,488,408,494,404,486,404,484,414,928,864,926,866,486,410,930,864,490,406,482,416,484,410,932,862,490,406,484,416,484,410,492,402,488,408,492,406,484,410,930,416,484,860,490,404,486],"RawDataInfo":[63,63,0]}}
```
So at least that seems to work.

I do not own a RC6 device.

NB. Don't mind the hour time difference ;-)). It's that time of the year around here.

roelleke, 2019-03-31:
Hallo,
Yes your command works, but you only send 32 bits and not 36. Please try this:
`IRsend {"Protocol":"RC6","Bits":36,"Data":0xC8052800C}` 
and you will see that you only receive 0xFFFFFFFF

arendst, 2019-04-01:
You're right. Will fix.

helmar74, 2019-04-01:
Maybe there's the same error with Panasonic Devices, because the length is there also greater than 32 Bits?
My Panasonic TV doesn't react on sending the code, which was recorded from the remote:
`MQT: tele/sonoffIRBridge_1/RESULT = {"IrReceived": "Protocol":"PANASONIC","Bits":48,"Data":"0x100BCBD"}}

IRsend {"Protocol":"PANASONIC","Bits":48,"Data":0x100BCBD}`

arendst, 2019-04-01:
Both IRSend and receive now support up to 64-bit values which should solve the RC6 protocol issue.

For Panasonic protocol the handling has changed from bits/data to data/bits in line with other supported protocols. This may result in users needing to redefine legacy Panasonic IRsend commands.

arendst, 2019-04-01:
Waiting for feedback

roelleke, 2019-04-01:
Hello,
yes it works now, verry fine. I tested the RC6 Protokoll and the PANASONIC with 48 bits. Both could decoded probaly. Unfortnatly I don't own a Panasonic device. With the RC6 Signal I was able to operate my Vu+ DUO box without problems. Therfore i think the Problem is solved.

helmar74, 2019-04-01:
Hi,

I can also confirm ;-)
I have a RC6 based Device and a Panasonic TV
Both Devices works with Tasmota version 6.5.0.3 (self compiled on CORE 2.5.0

@arendst : Thanks a lot for the quick fix...

Closing the issue

closed 2019-04-01 by helmar74

helmar74, 2019-04-16:
Started Collection of IR Codes for different Remotes:
https://github.com/arendst/Sonoff-Tasmota/wiki/Codes-for-IR-Remotes-(for-YTF-IR-Bridge)

hristo-atanasov, 2019-08-23:
Actually lots of ACs send about 100 so 128 is better .. No one of my ACs remotes sends less than 100 (108 approx) so it would be nice to have 128 bits .. it is truncated to 64 when I try to use irrecv .. :\

hristo-atanasov, 2019-08-23:
actually the problem is only with receiving ..
I receive:
{"IrReceived":{"Protocol":"UNKNOWN","Bits":104,"Data":"0x2000402C069FC3" ...
instead of IRrecvDump2: 0xC3F9606C02000400000000A0C0 (reversed)

s-hadinger, 2019-08-24:
Is it worth adding hvac to IRReceived? 

My focus is on the send part. Hvac codes are complex and I'm not sure how it is helping to decode hvac at Tasmota level.

hristo-atanasov, 2019-08-24:
@s-hadinger I do not need HVAC at all .. :) I wrote some, others I adapted in Python. What it worth is the ability to send me the full 104bit hex code in the MQTT message when received, because it is currently truncated .. From this code I can see what is the current state, mode, temp, etc..

BOT-CODER, 2024-01-11:
> Try with a (old smartphone) camera if you see the led flashing from the IR Bridge. You can test with the orig. Remote if you see the LED flickering with. If this fails try precompiled firmware.

Can we build IR bin with custom feature of IFSTATMENET and EXPERSION with IR enabled in ESP8266

barbudor, 2024-01-11:
As stated in the documentation, you have to build the base 'tasmota' adding everything you need including IR, IF and expression

Trying to add features to a preconfigured variant is likely to fail

BOT-CODER, 2024-01-11:
> As stated in the documentation, you have to build the base 'tasmota' adding everything you need including IR, IF and expression
> 
> Trying to add features to a preconfigured variant is likely to fail

Do you have any tutorial to include IR protocol?
And here is my IRHAC recived code (I captured this by tasmota-ir build)
```
{"Protocol":"COOLIX","Bits":24,"Data":"0xB23F40","DataLSB":"0x4DFC02","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Mode":"Cool","Power":"On","Celsius":"On","Temp":24,"FanSpeed":"Max","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}
```

In custom build it does not capture this (UNKNOWN) since protocol is not included , any tutorial to likely add this vendor protocol to my custom build 
***Consider Space since using esp8266***