arendst/Tasmota #5525: Rule using mem#state (Mem2state) trigger only works once after restart

DelroyTaylor, 2019-03-25:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
Sonoff-Tasmota 6.4.1.21 by Theo Arends
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

15:47:31 CMD: status 0
15:47:31 SRC: WebConsole from 192.168.0.27
15:47:31 RSL: Received Topic /status, Data Size 1, Data 0
15:47:31 RSL: Group 0, Index 1, Command STATUS, Data 0
15:47:31 MQT: Pool/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Pool Pump and Heater Control"],"Topic":"Pool","ButtonTopic":"button","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:47:31 MQT: Pool/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:43","StartupUTC":"2019-03-25T19:45:48","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":13,"SaveAddress":"3F7000"}}
15:47:31 MQT: Pool/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.21(sonoff)","BuildDateTime":"2019-03-25T15:27:18","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
15:47:31 MQT: Pool/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["SmartRG-a58c","CIA_SurveillanceVan"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8109","280500000100000000000000000000000000","00000200"]}}
15:47:31 MQT: Pool/stat/STATUS4 = {"StatusMEM":{"ProgramSize":581,"Free":3492,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":2,"Features":["00000809","0FDAE79C","000783A0","23B617CC","00083BC0"]}}
15:47:31 MQT: Pool/stat/STATUS5 = {"StatusNET":{"Hostname":"Pool-3036","IPAddress":"192.168.0.34","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"216.254.141.2","Mac":"CC:50:E3:08:EB:DC","Webserver":2,"WifiConfig":2}}
15:47:31 MQT: Pool/stat/STATUS6 = {"StatusMQT":{"MqttHost":"m16.cloudmqtt.com","MqttPort":15089,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_08EBDC","MqttUser":"gxkrwjot","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:47:31 MQT: Pool/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 25 19:47:31 2019","Local":"Mon Mar 25 15:47:31 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 03:00:00 2019","Timezone":99,"Sunrise":"07:10","Sunset":"19:34"}}
15:47:31 MQT: Pool/stat/STATUS10 = {"StatusSNS":{"Time":"2019-03-25T15:47:31"}}
15:47:31 MQT: Pool/stat/STATUS11 = {"StatusSTS":{"Time":"2019-03-25T15:47:31","Uptime":"0T00:01:43","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"SmartRG-a58c","BSSId":"3C:90:66:C6:A5:90","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

15:50:54 CMD: rule1
15:50:54 SRC: WebConsole from 192.168.0.27
15:50:54 RSL: Received Topic /rule1, Data Size 0, Data 
15:50:54 RSL: Group 0, Index 1, Command RULE, Data 
15:50:54 MQT: Pool/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"on Mem2#State do backlog Var5 Mem2; Sub5 3; Mem3 Var5; endon "}
15:50:55 WIF: Checking connection...
15:50:55 WIF: Connected
15:51:05 MQT: Pool/tele/STATE = {"Time":"2019-03-25T15:51:05","Uptime":"0T00:05:17","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"SmartRG-a58c","BSSId":"3C:90:66:C6:A5:90","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
15:51:08 CMD: rule1 1
15:51:08 SRC: WebConsole from 192.168.0.27
15:51:08 RSL: Received Topic /rule1, Data Size 1, Data 1
15:51:08 RSL: Group 0, Index 1, Command RULE, Data 1
15:51:08 MQT: Pool/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"on Mem2#State do backlog Var5 Mem2; Sub5 3; Mem3 Var5; endon "}
15:51:08 RUL: MEM2#STATE performs "backlog Var5 Mem2; Sub5 3; Mem3 Var5;"
15:51:08 SRC: Rule
15:51:08 RSL: Received Topic /backlog, Data Size 29, Data Var5 Mem2; Sub5 3; Mem3 Var5;
15:51:08 RSL: Group 0, Index 1, Command BACKLOG, Data Var5 Mem2; Sub5 3; Mem3 Var5;
15:51:09 SRC: Backlog
15:51:09 RSL: Received Topic /Var5, Data Size 4, Data Mem2
15:51:09 RSL: Group 0, Index 5, Command VAR, Data Mem2
15:51:09 MQT: Pool/stat/RESULT = {"Var5":"70.000"}
15:51:09 SRC: Backlog
15:51:09 RSL: Received Topic /Sub5, Data Size 1, Data 3
15:51:09 RSL: Group 0, Index 5, Command SUB, Data 3
15:51:09 MQT: Pool/stat/RESULT = {"Sub5":"67.000"}
15:51:09 SRC: Backlog
15:51:09 RSL: Received Topic /Mem3, Data Size 4, Data Var5
15:51:09 RSL: Group 0, Index 3, Command MEM, Data Var5
15:51:09 MQT: Pool/stat/RESULT = {"Mem3":"67.000"}
15:51:10 CFG: Saved to flash at 3F5, Count 15, Bytes 3584
15:51:15 WIF: Checking connection...
15:51:15 WIF: Connected
15:51:25 CMD: mem3
15:51:25 SRC: WebConsole from 192.168.0.27
15:51:25 RSL: Received Topic /mem3, Data Size 0, Data 
15:51:25 RSL: Group 0, Index 3, Command MEM, Data 
15:51:25 MQT: Pool/stat/RESULT = {"Mem3":"67.000"}
15:51:35 WIF: Checking connection...
15:51:35 WIF: Connected
15:51:36 CMD: mem2 80
15:51:36 SRC: WebConsole from 192.168.0.27
15:51:36 RSL: Received Topic /mem2, Data Size 2, Data 80
15:51:36 RSL: Group 0, Index 2, Command MEM, Data 80
15:51:36 MQT: Pool/stat/RESULT = {"Mem2":"80.000"}
15:51:37 CFG: Saved to flash at 3F4, Count 16, Bytes 3584
15:51:44 CMD: mem3
15:51:44 SRC: WebConsole from 192.168.0.27
15:51:44 RSL: Received Topic /mem3, Data Size 0, Data 
15:51:44 RSL: Group 0, Index 3, Command MEM, Data 
15:51:44 MQT: Pool/stat/RESULT = {"Mem3":"67.000"}

```

### TO REPRODUCE
Enter the following rule in the console
Rule1 on Mem2#State do backlog Var5 Mem2; Sub5 3; Mem3 Var5; endon on BUTTON1#state=3 do POWER1 2 endon on BUTTON1#state=2 do delay endon on BUTTON2#state=3 do POWER2 2 endon on BUTTON2#state=2 do delay endon 

Rule1 1

The first test of this rule works runs as expected and the result is correct.
eg.
Mem2 70
Mem2                            //result:    Mem2 = 70
Mem3                           //result:     Mem3 = 67

The second test stores the number in memory but the does not run..
eg.
Mem2 80
Mem2                            //result:    Mem2 = 80
Mem3                           //result:     Mem3 = 67

### EXPECTED BEHAVIOUR

Subtract 3 from Mem2 and store then result in Mem3.

ascillato2, 2019-03-26:
Hi,

You have a syntax error in your rule.

When you want to store a variable to VAR you should use `var5 %mem2%`

Please, go to the Tasmota console and type:

```
Rule1 on Mem2#State do backlog Var5 %Mem2%; Sub5 3; Mem3 %Var5% endon
Rule1 1
```

DelroyTaylor, 2019-03-26:
Thank you for the quick response but I still don't get the expected response. Mem2 value is now text.

20:31:26 CMD: rule1 1
20:31:26 MQT: Pool/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":448,"Rules":"on Mem2#State do backlog Var5 %Mem2%; Sub5 3; Mem3 %Var5% endon"}
20:31:43 CMD: mem2 99
20:31:43 MQT: Pool/stat/RESULT = {"Mem2":"99.000"}
20:31:43 RUL: MEM2#STATE performs "backlog Var5 %Mem2%; Sub5 3; Mem3 %Var5%"
20:31:44 MQT: Pool/stat/RESULT = {"Var5":"0.000"}
20:31:44 MQT: Pool/stat/RESULT = {"Sub5":"-3.000"}
20:31:44 MQT: Pool/stat/RESULT = {"Mem3":"0.000"}
20:33:15 CMD: mem3
20:33:15 MQT: Pool/stat/RESULT = {"Mem3":"0.000"}
20:35:09 CMD: mem2 77
20:35:09 MQT: Pool/stat/RESULT = {"Mem2":"77.000"}

ascillato, 2019-03-26:
Tested. You are right. Investigating. Thanks for reporting. :+1:

DelroyTaylor, 2019-03-26:
Hi Ascillato 

Sonoff-Tasmota 6.5.0.1 by Theo Arends also produces undesired results and Rule1 runs only once.

21:43:39 CMD: rule1 1
21:43:39 MQT: Pool/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":447,"Rules":"on Mem2#State do backlog Var5 %Mem2%; Sub5 3; Mem3 %Var5%; endon"}
21:43:59 CMD: mem2  99
21:43:59 MQT: Pool/stat/RESULT = {"Mem2":"99"}
21:43:59 RUL: MEM2#STATE performs "backlog Var5 %Mem2%; Sub5 3; Mem3 %Var5%;"
21:44:00 MQT: Pool/stat/RESULT = {"Var5":"%Mem2%"}
21:44:00 MQT: Pool/stat/RESULT = {"Sub5":"-3.000"}
21:44:00 MQT: Pool/stat/RESULT = {"Mem3":"%Var5%"}
21:44:34 CMD: mem3
21:44:34 MQT: Pool/stat/RESULT = {"Mem3":"%Var5%"}
21:44:51 CMD: mem2 88
21:44:51 MQT: Pool/stat/RESULT = {"Mem2":"88"}
21:45:02 CMD: mem3
21:45:02 MQT: Pool/stat/RESULT = {"Mem3":"%Var5%"}

Staars, 2019-03-28:
Hi,

the problem is, that 'vars_event' will be set to '>0' and stays >0.
So in the second turn the ` else if (mems_event) ` will never again be executed after ` else if (vars_event) ` in xdrv_10_rules.ino at around line 500.
You can do a short test and change both loops (mems before vars), but this only handles the symptom here. I think, we need to reset the variables in a correct way after use.

Staars, 2019-03-29:
The most simple solution for this particular problem so far, was to to encapsulate both if-statements.
Now I have this:
`00:00:35 CMD: mem2 55
00:00:35 SRC: Serial
00:00:35 RSL: Group 0, Index 2, Command MEM, Data 55
00:00:35 MQT: stat/sonoff/RESULT = {"Mem2":"55.000"}
00:00:35 RUL: MEM2#STATE performs "backlog Var5 Mem2; Sub5 3; Mem3 Var5;"
00:00:35 SRC: Rule
00:00:35 RSL: Group 0, Index 1, Command BACKLOG, Data Var5 Mem2; Sub5 3; Mem3 Var5;
00:00:35 SRC: Backlog
00:00:35 RSL: Group 0, Index 5, Command VAR, Data Mem2
00:00:35 MQT: stat/sonoff/RESULT = {"Var5":"55.000"}
00:00:35 SRC: Backlog
00:00:35 RSL: Group 0, Index 5, Command SUB, Data 3
00:00:35 MQT: stat/sonoff/RESULT = {"Sub5":"52.000"}
00:00:35 SRC: Backlog
00:00:35 RSL: Group 0, Index 3, Command MEM, Data Var5
00:00:35 MQT: stat/sonoff/RESULT = {"Mem3":"52.000"}
00:00:36 CFG: Saved to flash at F6, Count 228, Bytes 3584
00:00:46 CMD: mem2 66
00:00:46 SRC: Serial
00:00:46 RSL: Group 0, Index 2, Command MEM, Data 66
00:00:46 MQT: stat/sonoff/RESULT = {"Mem2":"66.000"}
00:00:46 RUL: MEM2#STATE performs "backlog Var5 Mem2; Sub5 3; Mem3 Var5;"
00:00:46 SRC: Rule
00:00:46 RSL: Group 0, Index 1, Command BACKLOG, Data Var5 Mem2; Sub5 3; Mem3 Var5;
00:00:46 SRC: Backlog
00:00:46 RSL: Group 0, Index 5, Command VAR, Data Mem2
00:00:46 MQT: stat/sonoff/RESULT = {"Var5":"66.000"}
00:00:47 SRC: Backlog
00:00:47 RSL: Group 0, Index 5, Command SUB, Data 3
00:00:47 MQT: stat/sonoff/RESULT = {"Sub5":"63.000"}
00:00:47 SRC: Backlog
00:00:47 RSL: Group 0, Index 3, Command MEM, Data Var5
00:00:47 MQT: stat/sonoff/RESULT = {"Mem3":"63.000"}
00:00:47 CFG: Saved to flash at F5, Count 229, Bytes 3584`

If desired, I can do a PR, but I still think, that there might be a better solution in the future.

DelroyTaylor, 2019-03-29:
Thank You Christian Staars. Your solution works.
As you suggested, I enabled #define USE_EXPRESSION under Rules in my_user_config.h. Then inserted the code you provided into the sonoff/xdrv_10_rules.ino file. 
compile upload test :+1: 
Many Thanks Christian, Adrian and Theo.

ascillato2, 2019-03-30:
Hi, great news! :+1: 

Sorry, I haven't the time to check this. please, leave this issue open as reminder for us and until the fix from @Staars got merged. Thanks.

netpok, 2019-04-04:
Actually the first example shouldn't work either (probably it was already assigned as 67) because the variables are assigned before execution.

 So to achieve the expected result you need to use an event, for example:
```Rule1 on Mem2#State do backlog Var5 %Mem2%; Sub5 3; event store endon on event#store do Mem3 %Var5% endon```

This is/was documented on the wiki somewhere but couldn't found it right now.

DelroyTaylor, 2019-04-05:
Netpok

What does this fix? Are you addressing the failure to repeat the calculation in the original code or the syntax error in the calculation?
I did get the rule to work without "event store...".
My rule1 is: Rule1 on Mem2#State do backlog Var5 Mem2; Sub5 3; Mem3 Var5; endon 
This does work and uses less characters in the Rules buffer which has a 511 character limit.
Note: I have migrated to 6.5.0.1
         I enabled #define USE_EXPRESSION
         I modified sonoff/xdrv_10_rules.ino file as suggested by Christian Staars in #5547.
Sample
20:23:52 CMD: rule1
20:23:52 SRC: WebConsole from 192.168.0.27
20:23:52 RSL: Received Topic /rule1, Data Size 0, Data 
20:23:52 RSL: Group 0, Index 1, Command RULE, Data 
20:23:53 RSL: Pool/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":287,"Rules":"on Mem2#State do backlog Var5 Mem2; Sub5 3; Mem3 Var5; endon on BUTTON1#state=3 do backlog POWER1 2; var3 2; endon on BUTTON1#state=2 do delay endon on BUTTON2#state=3 do POWER2 %var3% endon on BUTTON2#state=2 do delay endon"}
20:24:01 WIF: Checking connection...
20:24:01 WIF: Connected
20:24:21 WIF: Checking connection...
20:24:21 WIF: Connected
20:24:31 CMD: mem2 777
20:24:31 SRC: WebConsole from 192.168.0.27
20:24:31 RSL: Received Topic /mem2, Data Size 3, Data 777
20:24:31 RSL: Group 0, Index 2, Command MEM, Data 777
20:24:31 MQT: Pool/stat/RESULT = {"Mem2":"777.000"}
20:24:31 RUL: MEM2#STATE performs "backlog Var5 Mem2; Sub5 3; Mem3 Var5;"
20:24:31 SRC: Rule
20:24:31 RSL: Received Topic /backlog, Data Size 29, Data Var5 Mem2; Sub5 3; Mem3 Var5;
20:24:31 RSL: Group 0, Index 1, Command BACKLOG, Data Var5 Mem2; Sub5 3; Mem3 Var5;
20:24:31 SRC: Backlog
20:24:31 RSL: Received Topic /Var5, Data Size 4, Data Mem2
20:24:31 RSL: Group 0, Index 5, Command VAR, Data Mem2
20:24:31 MQT: Pool/stat/RESULT = {"Var5":"777.000"}
20:24:31 SRC: Backlog
20:24:31 RSL: Received Topic /Sub5, Data Size 1, Data 3
20:24:31 RSL: Group 0, Index 5, Command SUB, Data 3
20:24:31 MQT: Pool/stat/RESULT = {"Sub5":"774.000"}
20:24:32 SRC: Backlog
20:24:32 RSL: Received Topic /Mem3, Data Size 4, Data Var5
20:24:32 RSL: Group 0, Index 3, Command MEM, Data Var5
20:24:32 MQT: Pool/stat/RESULT = {"Mem3":"774.000"}
20:24:33 CFG: Saved to flash at 3F5, Count 47, Bytes 3584

netpok, 2019-04-05:
Looks like I written faster than I should as your problem lies in the event triggering and not in the value, sorry.

 I wanted to do some test, but looks like I bricked my sonoff and I won't have time to fix it until the en of week (or month if I'm unlucky).

radionerd, 2019-04-05:
Pasting additional test information here as var5#state not functioning as a rule trigger is supposed to be related to this issue, though I don't see how.
07:23:49 CMD: backlog rule1 ; var5 5; var4 4
07:23:49 SRC: WebConsole from 192.168.1.23
07:23:49 RSL: Received Topic /backlog, Data Size 22, Data rule1 ; var5 5; var4 4
07:23:49 RSL: Group 0, Index 1, Command BACKLOG, Data rule1 ; var5 5; var4 4
07:23:50 SRC: Backlog
07:23:50 RSL: Received Topic /rule1, Data Size 0, Data 
07:23:50 RSL: Group 0, Index 1, Command RULE, Data 
07:23:50 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":395,"Rules":"on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon"}
07:23:50 SRC: Backlog
07:23:50 RSL: Received Topic /var5, Data Size 1, Data 5
07:23:50 RSL: Group 0, Index 5, Command VAR, Data 5
07:23:50 RSL: RESULT = {"Var5":"5"}
07:23:50 SRC: Backlog
07:23:50 RSL: Received Topic /var4, Data Size 1, Data 4
07:23:50 RSL: Group 0, Index 4, Command VAR, Data 4
07:23:50 RSL: RESULT = {"Var4":"4"}
07:23:50 RUL: VAR4#STATE performs "backlog event display=4"
07:23:50 SRC: Rule
07:23:50 RSL: Received Topic /backlog, Data Size 15, Data event display=4
07:23:50 RSL: Group 0, Index 1, Command BACKLOG, Data event display=4
07:23:51 SRC: Backlog
07:23:51 RSL: Received Topic /event, Data Size 9, Data display=4
07:23:51 RSL: Group 0, Index 1, Command EVENT, Data display=4
07:23:51 RSL: RESULT = {"Event":"Done"}
07:23:51 RUL: EVENT#DISPLAY performs "var1 4"
07:23:51 SRC: Rule
07:23:51 RSL: Received Topic /var1, Data Size 1, Data 4
07:23:51 RSL: Group 0, Index 1, Command VAR, Data 4
07:23:51 RSL: RESULT = {"Var1":"4"}

Staars, 2019-04-08:
Please check, if this works for you.

https://github.com/arendst/Sonoff-Tasmota/pull/5604

radionerd, 2019-04-08:
Sorry this change does not appear to fix the var5 event issue.
Test results below:
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":395,"Rules":"on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon"}
05:53:40 CMD: var4 4
05:53:40 RSL: RESULT = {"Var4":"4"}
05:53:40 RUL: VAR4#STATE performs "backlog event display=4"
05:53:41 RSL: RESULT = {"Event":"Done"}
05:53:41 RUL: EVENT#DISPLAY performs "var1 4"
05:53:41 RSL: RESULT = {"Var1":"4"}
05:53:48 CMD: var5 5
05:53:48 RSL: RESULT = {"Var5":"5"} <- Change to var5 did not generate event
05:54:18 CMD: var1
05:54:18 RSL: RESULT = {"Var1":"4"} <- last event was var4
Program Version | 6.5.0.5(ce53ee0-display)
2019-04-08T14:09:57
2_4_2/2.2.1(cfd48f3)
esp12F

Staars, 2019-04-08:
Just to be sure, that we do not talk past each other: Did you compile from my fork?
I get this now:
`00:00:47 CMD: backlog rule1 ; var5 5; var4 4
00:00:47 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":395,"Rules":"on var4#state do event display=4 endon on var5#state do event display=5 endon on event#display do var1 %value% endon"}
00:00:47 RSL: stat/sonoff/RESULT = {"Var5":"5.000"}
00:00:47 RUL: VAR5#STATE performs "backlog event display=5"
00:00:48 RSL: stat/sonoff/RESULT = {"Var4":"4.000"}
00:00:48 RUL: VAR4#STATE performs "backlog event display=4"
00:00:48 RSL: stat/sonoff/RESULT = {"Event":"Done"}
00:00:48 RUL: EVENT#DISPLAY performs "var1 5"
00:00:48 RSL: stat/sonoff/RESULT = {"Var1":"5.000"}
00:00:48 RSL: stat/sonoff/RESULT = {"Event":"Done"}
00:00:48 RUL: EVENT#DISPLAY performs "var1 4"
00:00:48 RSL: stat/sonoff/RESULT = {"Var1":"4.000"}
`

radionerd, 2019-04-08:
Your test result looks perfect and acceptable to me.
Apologies, I downloaded a dev binary and tested that:
Program Version | 6.5.0.5(ce53ee0-display) 2019-04-08T14:09:57 sonoff-display-4m-dev.bin 
md5sum 8c4a63bdc77fae1489e2d8498655e257
Are you able to point me towards your binary to test? Or I can clone your repo, install the tools, compile and test. That may take me a little longer to figure out what to do. Thanks so much for looking into this.

Staars, 2019-04-08:
No problem.
Yes, just clone my repo for testing/compiling. You basically only have to use "Staars" instead of "arendst" and then follow one of the tutorials. 
Take your time, don't hurry.

radionerd, 2019-04-10:
Success.
Building from source: git clone https://github.com/staars/Sonoff-Tasmota.git 
Flashing with: platformio run --target upload --upload-port /dev/ttyUSB0
reports: Program Version	6.5.0.6(display)
 writing to var5 now generates rule var5#state events
Additionally
Building from source: git clone https://github.com/arendst/Sonoff-Tasmota.git 
reports: Program Version	6.5.0.7(display)
 writing to var5 also now generates rule var5#state events
Many Thanks

Staars, 2019-04-10:
Cool.
The PR was already integrated.

Thanks to all bug reporters, I think this thread can now be closed.

closed 2019-04-13 by DelroyTaylor