arendst/Tasmota #5526: lc technology 2ch relay

megamarco833, 2019-03-25:
hi, i have a 2ch relay based on esp-01
https://web.archive.org/web/20180602202511/http://www.chinalctech.com/index.php?_m=mod_product&_a=view&p_id=1261

i programmed the ESP-01 with tasmota and configure the module, insert the rules and set the rules on.

but when i try to toggle the relay....none is working :(

i see the wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay
and also follow this:
https://www.esp8266.com/viewtopic.php?f=160&t=13164&start=68#p74262

but my chip is not STC15F104W but Nuvoton N76: n76e003at20
so how i can program it and make it working with tasmota?
thanks

ascillato2, 2019-03-26:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

KPK75, 2019-03-26:
Baudrate set?

Jason2866, 2019-03-26:
looked at
https://github.com/arendst/Sonoff-Tasmota/pull/1282

megamarco833, 2019-03-28:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

### ISSUE DESCRIPTION - TROUBLESHOOTING
on relay lc technology 2ch with on board ESP01 and chip model Nuvoton N76: n76e003at20 (not STC15F104W as decrived in wiki, ref: https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay)
is not possible to use tasmota working
  

### REQUESTED INFORMATION
how to make it compatible

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : lc technology with 2ch and chip Nuvoton N76: n76e003at20
- [ x] Tasmota binary firmware version number used : Sonoff-Tasmota 6.4.1 by Theo Arends / (pre-compiled)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
01:47:53 CMD: status 0
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS = {"Status":{"Module":18,"FriendlyName":["ESP01_ltc_2ch","Sonoff2"],"Topic":"ESP01_ltc_2ch","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:19","StartupUTC":"2019-03-28T00:42:34","Sleep":50,"BootCount":41,"SaveCount":299,"SaveAddress":"F9000"}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sensors)","BuildDateTime":"2018-12-24T14:45:55","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"SetOption":["00C08009","558180C0","40000000"]}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":19,"ProgramFlashSize":4096,"FlashSize":1024,"FlashChipId":"146085","FlashMode":2,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS5 = {"StatusNET":{"Hostname":"ESP01_ltc_2ch-5989","IPAddress":"192.168.0.154","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"84:0D:8E:97:97:65","Webserver":2,"WifiConfig":4}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_979765","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 28 00:47:53 2019","Local":"Thu Mar 28 01:47:53 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:36","Sunset":"19:14"}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS10 = {"StatusSNS":{"Time":"2019-03-28T01:47:53","ANALOG":{"A0":91}}}
01:47:53 RSL: stat/ESP01_ltc_2ch/STATUS11 = {"StatusSTS":{"Time":"2019-03-28T01:47:53","Uptime":"0T00:05:19","SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"00:14:6C:E6:D2:0D","Channel":11,"RSSI":68}}}

megamarco833, 2019-03-28:
i have some updates by tring many time and i now have a partial solution

i discover that if follo the guide that i linked on tasmota wiki but i do not flash my chip Nuvoton and i do not remove the resistor, but simpy hold and press the S1 button on boot, the firmware become working as descrived in the wiki.

note, if you boot/power up normali, you will see a led blue lighting -> in this case the FW will NOT work
if you hold and press S1 button and then power up the board you should see: red led light on, in a while green led blinking, blue led switch off -> in this case the FW will work and the relay will toggle.

now the problem is enabling "wall switches" for relay1 and relay2
if i go to configuration-> configuration module and i set:
TX = button1
RX = button2
with this config of button the relays will not toggle :-(

if i set:
TX = NONE
Rx = button2
in this case both relay are toggling correctly. but, to do that i need to have jumper1 connectin RX1-TX1 (defaulf position) but i have to remove jumper2

now, how i can use/enable button1 for toggle relay1?
if i enable button1 on TX i will lose the toggle of both relay that became not working
thanks

Jason2866, 2019-03-28:
You cant use RX / TX for buttons since this GPIOs are used for communication to the UART!!

ascillato2, 2019-03-30:
@megamarco833

Hi,
 
Any news on this?

megamarco833, 2019-03-31:
> 
> 
> @megamarco833
> 
> Hi,
> 
> Any news on this?

as i wrote, what i discover is that:
1)i follow the guide that i linked on tasmota wiki but i do not flash my chip Nuvoton and i do not remove the resistor, but simply hold and press the S1 button on boot, the firmware become working as descrived in the wiki.

note, if you boot/power up normally, you will see a led blue lighting -> in this case the FW will NOT work
if you hold and press S1 button and then power up the board you should see: red led light on, in a while green led blinking, blue led switch off -> in this case the FW will work and the relay will toggle.

i do not know if it could be a solution, but for me is working :-)

....what now i would like to use is the possibility to use this board with 2relays with two external push button as switch.
is it possible???

I would like to use this board for blind control + exterla wall switch, is it possible?
thanks

Jason2866, 2019-03-31:
You have to mod the device to add two momentary switches for your wanted use case.
The used Esp module has two free GPIOs. GPIO 0 and GPIO 2 Both GPIOS needs to be pulled up to high for normal function of the device. With a momentary switch you have to pull to GND.

megamarco833, 2019-04-01:
> 
> 
> You have to mod the device to add two momentary switches for your wanted use case.
> The used Esp module has two free GPIOs. GPIO 0 and GPIO 2 Both GPIOS needs to be pulled up to high for normal function of the device. With a momentary switch you can pull to GND.

that's correct, but on the board there are two pins:TX and RX that should be correspond to GPIO0 and GPIO2
so i can use that two, right?
but if i set inside the configuration modul "generic" button1 and button2
i lost the function of toggle relay (web interface show the toggle, but phisically the relay don't toggle)
if i set:
TX = NONE
Rx = button2
in this case both relay are toggling correctly and if use wall switch connected at RX the relay2 will toggle. but, to do that i need to have jumper1 connectin RX1-TX1 (defaulf position) but i have to remove jumper2

now, how i can use/enable button1 for toggle relay1?
if i enable button1 on TX i will lose the toggle of both relay that became not working
thanks

Jason2866, 2019-04-02:
> on the board there are two pins:TX and RX that should not be correspond to GPIO0 and GPIO2. GPIO0 and 2 are the relays

WRONG TX and RX is connected to GPIO 1 and 3 from ESP.
**The relais arent connected to the ESP. The are connected to the MCU (Nuvoton N76) and controlled from.**

megamarco833, 2019-04-02:
> 
> 
> > on the board there are two pins:TX and RX that should not be correspond to GPIO0 and GPIO2. GPIO0 and 2 are the relays
> 
> WRONG TX and RX is connected to GPIO 1 and 3 from ESP.
> **The relais arent connected to the ESP. The are connected to the MCU (Nuvoton N76) and are controlled from.**
thanks Jason!
ok...so why if i go inside  configuration-> configuration module and i set:
TX = button1
RX = button2
with this config the relays will not phisically toggle anymore :-(
thanks

Jason2866, 2019-04-02:
Because you bring device in boot mode with your wrong configuration and the devices reacts somehow...
The ESP has only 4 usable GPIOS (0,1,2,3) 
RX/TX (GPIO 1 and 3) are used to control MCU to switch the relais via serial command codes:
```
Open the first relay：   A0 01 01 A2
Close the first Relay：  A0 01 00 A1
Open the second relay： A0 02 01 A3
Close the second Relay：A0 02 00 A2
```
So you have only left to use GPIO 0 and 2 to use for momentary switches

megamarco833, 2019-04-02:
> Because you bring device in boot mode with your wrong configuration and the devices reacts somehow...
ehmm...i use the device in boot mode because if i boot it normally the relay will not phisically toggle.
if i boot holding S1 (boot mode) the relays will phisically toggle!
i do not have STC15F104W chip, but Nuvoton chip...so what is written in the wiki to flash it is not working....
do you have other solutions?

> The ESP-01 has only 4 usable GPIOS (0,1,2,3)
> 
> So you have only GPIO 0 and 2 left to use for momentary switches
> See wiki https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay too
> **GPIO 0 and 2 are set as VIRTUAL relays to get relais defined in Tasmota. This relais has nothing to do with the real ones!** You could use non existent GPIOs for this needed setting too

so, if i understand well, i can set virtual relay in tasmota:
GPIO1 and GPIO3 as Relay1 and Relay 2 (not like in the wiki)
and use GPIO0 and GPIO2 for momentary switch like button1 and button2

or, because RX/TX (GPIO 1 and 3) are used to control MCU, i can't use 1 and 3 for relay and i must set only 0 and 2 for relays ?
thanks so much for your help

Jason2866, 2019-04-02:
Set GPIO 4 as virtual relais1 and GPIO 5 as virtual relais2. Rules like in wiki
So you have GPIO 0 and 2 free for connecting momentary switches

ascillato2, 2019-04-13:
Closing issue as there is no feedback. Please, when you have time, tell us if the proposed solution works for you. Thanks

closed 2019-04-13 by ascillato2

soumaxetuirk, 2020-01-20:
> Closing issue as there is no feedback. Please, when you have time, tell us if the proposed solution works for you. Thanks

Hi, I recently got the same model as you from Aliexpress. mine is working perfectly fine.I want to give you and other a detailed guide for LC Tech 2 Channel relay.

The 2 channel relay board is programmed to work with 2 modes.
**Mode 1: (BLUE LED D5 is ON)**
The factory firmware supposed to command ESP8266 Chip to connect to your Wifi network.
after 20 sec green led will start to blink (0.5 Sec).at this time you can connect to your ESP8266 chip which is flushed already by Tasmota.But board will not accept serial commands sent by ESP8266, Hence you will not be able to control relay.

**Mode 2: (RED LED D7 is On)**
in this mode the board is supposed to send command to ESP8266 to create a mobile hotspot. Also if no ESP8266 is present then this mode will allow you to control the board using Serial ports directly.That is the reason why Tasmota works in this mode only. 

note:
if you are stuck in "mode 1 ", then hold the S1 button while powering on the board. Red LED will glow for 1 or 2 sec, during that time remove the finger from S1 button. (or else it will go back to mode 1).

GREEN LED has 3 meanings.
OFF - board is configuring with ESP8266
BLINK (0.5 sec) - Connecting to wifi for rest configuration
BLINK (2 sec) - Configured and stable

The key(S1 and S2) are available only when the LED(D6) slow blink every 2S or on all the time.

GPIO1 & GPIO3 used for serial communication with board so create 2 virtual relay with GPIO0 & GPIO2.(Tasmota>configuration>configure template>generic(18),GPIO0 as relay1,GPIO2 as relay2)
once done, create a rule to send serial commands to board to use the relays.
Open the first relay：   A0 01 01 A2
Close the first Relay：  A0 01 00 A1
Open the second relay： A0 02 01 A3
Close the second Relay：A0 02 00 A2

So the rule is as follows:
```
Rule1
 on System#Boot do Backlog Baudrate 115200; SerialSend5 0 endon
 on Power1#State=1 do SerialSend5 A00101A2 endon
 on Power1#State=0 do SerialSend5 A00100A1 endon
 on Power2#State=1 do SerialSend5 A00201A3 endon
 on Power2#State=0 do SerialSend5 A00200A2 endon
```

after a restart the device will work .
The relay present in board are cut-off (2 way ) relay so I use cut-off (2 way) wall switches. In a similar way as the conventional "stair case dual switch " scenario.

megamarco833, 2020-01-22:
> after a restart the device will work .
> The relay present in board are cut-off (2 way ) relay so I use cut-off (2 way) wall switches. In a similar way as the conventional "stair case dual switch " scenario.
hi, thanks for supporting, i have again some trouble with module lc technology with two channels, i need to know:
1) inside configuration what GPIO you address to relay1 and relay2?
2)what jumper configuration do you have on module? jumper of TX1-RX1 is connected? jumper of TX-RX is also connected?
I see that inside your rule1 you use Backlog Baudrate 115200, inside wiki is written 9600, there is any reason why you changed?
3) are you able to use momentary switches? if yes, do you connect to pin RX and TX?
what you set inside module configuration as gpio for button1 and button2?
thanks so much
Marco

soumaxetuirk, 2020-01-23:
> > after a restart the device will work .
> > The relay present in board are cut-off (2 way ) relay so I use cut-off (2 way) wall switches. In a similar way as the conventional "stair case dual switch " scenario.
> > hi, thanks for supporting, i have again some trouble with module lc technology with two channels, i need to know:
> 
> 1. inside configuration what GPIO you address to relay1 and relay2?
>    2)what jumper configuration do you have on module? jumper of TX1-RX1 is connected? jumper of TX-RX is also connected?
>    I see that inside your rule1 you use Backlog Baudrate 115200, inside wiki is written 9600, there is any reason why you changed?
> 2. are you able to use momentary switches? if yes, do you connect to pin RX and TX?
>    what you set inside module configuration as gpio for button1 and button2?
>    thanks so much
>    Marco

1. GPIO0 > Relay1
   GPIO2 > Relay 2
2.RX-RX1 & TX-TX1
 According LC Tech Data Sheet, it is 115200 . I tried with 9600 but it is not working. But 115200 works.
3. I have already described the use of two buttons (s1 & S2) present in the board.There are no extra GPIO present to create button 1 or button 2.
4.one more thing i want to mention that, as we send serial commands to board while turning on & off by virtual relay.if the board gets power failure,on restart it will not remember the states of 2 relays.they will stay off. 
So I compile a binary by removing extra sensors and adding one "if else" function.
also created a second rule which will save relay's states in mem1 & mem2 .And in every boot it will send serial command to board as per the last state saved.

Carminepz, 2020-03-01:
perché a ogni comando risultato
stat/tasmota/RESULT = {"Command":"Unknown"}   ???

ascillato, 2020-03-01:
Which command?

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Carminepz, 2020-03-01:
ok , scusa

Carminepz, 2020-03-02:
Configurazione EXTRA:
{"NAME":"LC-ESP01-2R-5V","GPIO":[21,148,22,149,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
LED D7= Rosso
LED D6= Verde lampeggio (0,5 sec)

Nota:
se sei bloccato in "modalità 1", tieni premuto il tasto S1 mentre accendi la scheda. Il LED rosso si illuminerà per 1 o 2 secondi, durante quel tempo rimuovere il dito dal pulsante S1. (altrimenti tornerà alla modalità 1).

Il LED VERDE ha 3 significati.
OFF - la scheda si sta configurando con ESP8266
BLINK (0,5 sec) - Connessione al wifi per la configurazione di riposo
BLINK (2 sec) - Configurato e stabile

I tasti (S1 e S2) sono disponibili solo quando il LED (D6) lampeggia lentamente ogni 2 secondi o sempre acceso.

nacansino, 2020-04-18:
Hi, I know this is an old hair but I felt the need to share my individual findings here.
I also have the same LC-tech Relay2 module with the NUVOTON ic on it. Exact product [here](https://www.aliexpress.com/item/32862098154.html?spm=a2g0s.9042311.0.0.70854c4didRPU1).

After a couple of trial and errors, my observations are consistent with @soumaxetuirk . I managed to find a workaround on how to make the NUVOTON ic recognize the serial messages after it enters **Mode 1**. So this solution I'm gonna tell you **should work** even though the device is not in Mode 2, hence saving you of having to press S1 button while booting.

So here's what you're gonna do.

1. Use the device template [here](https://templates.blakadder.com/LC-Relay-ESP01-2R-5V.html). I will paste the code here so you don't have to go to the link:
`{"NAME":"LC-ESP01-2R-5V","GPIO":[0,148,0,149,0,0,0,0,21,22,0,0,0],"FLAG":0,"BASE":18}`
One way to configure your board with this configuration is to go to **Configuration - Configure Other** in the Tasmota web portal. Paste the configuration above to the **Template** field then click **Save**. This will reset the device. What this does is it configures the pins of the device for use in the target hardware (the LC-Tech Relay 2).

2. After the reset, you will get this screen on your web portal:
[image]

3.  We will add the following rules on our Tasmota:

`on System#Boot do Backlog Baudrate 115200`
`on SerialReceived#Data=41542B5253540D0A do SerialSend5 5749464920434f4e4e45435445440a5749464920474f542049500a41542b4349504d55583d310a41542b4349505345525645523d312c383038300a41542b43495053544f3d333630 endon`
`on Power1#State=1 do SerialSend5 A00101A2 endon`
`on Power1#State=0 do SerialSend5 A00100A1 endon`
`on Power2#State=1 do SerialSend5 A00201A3 endon`
`on Power2#State=0 do SerialSend5 A00200A2 endon`

Here's what the above code does line per line:
- Sets the serial baud rate to 115200 (this seems to be the default for the Nuvoton LCTech Relay)
- This sends a certain stream of serial messages (in hex) below after receiving `AT+RST` (41542B5253540D0A in hex) from the NUVOTON devices. This message seems to make the NUVOTON enter listening mode. The long stream of hex messages for sending is equivalent to the ff. in ASCII:
`WIFI CONNECTED`
`WIFI GOT IP`
`AT+CIPMUX=1`
`AT+CIPSERVER=1,8080`
`AT+CIPSTO=360`
I will call this the **bypass key**.
-The PowerX#State=xxx... messages are triggers to send serial messages to the NUVOTON chip.

Do not forget to enable the rule.

4. There's a catch. After the device receives the bypass key, it wouldn't immediately respond to commands. The ESP has to wait for the ff. return messages echoed back to Serial first:
`AT+CIPMUX=1`
`AT+CIPSERVER=1,8080`
`AT+CIPSTO=360`
After these messages are sent back by Nuvoton to the ESP, the green LED beside the green LED will start blinking once a second. From here you can check that 

That's it. I have tested this on four LCTech boards and they work flawlessly. So I think they should work on other Nuvoton boards too. Let me know if this worked out for you guys. I am willing to help.

megamarco833, 2020-04-18:
@nacansino thanks for sharing this!
before your post i never succeed on having this device working
now i tested with your solution and now it works!!!
fantastic, thanks so much

ps. is it possible to use push button1 and button2 to control relay1 and relay2?
thanks again

nacansino, 2020-04-19:
@megamarco833 That's fantastic news! With that, I think this solution is
applicable to any LCTech Nuvoton devices from the same manufacturer.

Regarding the buttons 1 and 2, I will verify if those buttons are
physically connected to GPIOs of the ESP (I think they aren't).

If it is not, then we have to verify if pressing those buttons will trigger
Nuvoton to send serial messages to ESP, and using those signals we can
create a separate rule that triggers the relays. This, or we have to
manually rewire the board to connect the buttons to the GPIO of the ESP and
make a rule for the relay.

meingraham, 2020-04-19:
@nacansino 

Could you add this information to the Tasmota docs? All users are able to contribute. This would be a great addition to the device's article.

nacansino, 2020-04-19:
@meingraham 

That's a good idea. Where exactly do you think I can add this?
I just started using Tasmota yesterday so I haven't threaded yet on the whole docs.

meingraham, 2020-04-19:
@nacansino 

https://tasmota.github.io/docs/devices/LC-Technology-WiFi-Relay/

nacansino, 2020-04-20:
@meingraham Submitted a Pull request! Thanks!

megamarco833, 2020-04-20:
thanks again for support!!!
@nacansino  and @meingraham 
there is also:
https://templates.blakadder.com/LC-Relay-ESP01-2R-5V.html
what's the better place to add ?
thanks

megamarco833, 2020-04-30:
> @megamarco833 That's fantastic news! With that, I think this solution is applicable to any LCTech >Nuvoton devices from the same manufacturer.
> Regarding the buttons 1 and 2, I will verify if those [...]

hi @nacansino 
did you find a way to use two switches to control relay1 and relay2?
thanks again for support on this!

nacansino, 2020-05-01:
@megamarco833 
Unfortunately, no. I tried to press those switches and they seem to move the IC to a different state. :/ What are you trying to accomplish anyway?
I personally would pivot to using another board as a smart switch. I find this board inefficient as it requires a separate DC supply on top of the AC line it's connected to.

megamarco833, 2020-05-01:
@nacansino i thought that can be used tx and rx pin exposed to control relays :-)
i would like to use this board to control roller shutter because it quite small and it's perfect having two relays: 1=up 2=down
what is missing are the wall switches to control both relays :)
thanks

soumaxetuirk, 2020-05-04:
> @nacansino i thought that can be used tx and rx pin exposed to control relays :-)
> i would like to use this board to control roller shutter because it quite small and it's perfect having two relays: 1=up 2=down
> what is missing are the wall switches to control both relays :)
> thanks

As I mentioned before also, you can use a staircase (2 way) light switch with the relay.