arendst/Tasmota #5530: MTU issue

revk, 2019-03-26:
Web server not working via limited MTU remote link (i.e. my VPN).

This may be a library issue, or maybe it is something the tasmota s/w can set if we have a config option, but it seems that the MTU in the initial SYN from the client is ignored. I have tried prepping for 1460 and 1500 and MTU with no luck.

I have a dump showing TCP MSS in initial SYN of 1360 but first full packet from device is 1460 Tcp segment length, and oddly shows a TCP checksum error too.

Version 6.5.0 on a Sonoff basic.

Is there any way the Tasmota can have an MTU/MSS config?

Jason2866, 2019-03-26:
Have you tried to compile a version by yourself with a different lwip setting?
Try lwip setting higher bandwidth. Which core do you use?
Without `Status 0` it is hard....

revk, 2019-03-26:
Bandwidth is not the issue, but can you clarify where lwip settings are, thanks.
It looks like a faulty TCP stack, but trying to see if any workaround.

Jason2866, 2019-03-26:
If you use Arduino IDE you find the infos here https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
When using platformio as IDE you have to change in platformio.ini Infos are here
https://github.com/arendst/Sonoff-Tasmota/wiki/PlatformIO

Jason2866, 2019-03-27:
See https://github.com/esp8266/Arduino/issues/5902

closed 2019-03-27 by revk