arendst/Tasmota #5531: Constant URL for reboot

einarmiger0815, 2019-03-26:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-26:
I didnt change over a long time....

einarmiger0815, 2019-03-26:
For sure. Please test for yourself:

In Version 5.X it was http://ipadress/rb?"

In version 6.4.x it was "http://ipadress/?rstrt=", now in 6.5.x it is "http://ipadress/?rst=". Unfortunately

curzon01, 2019-03-26:
Use [official commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) like [Reset 1](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management) by http request:

`http://sonoff/cm?cmnd=Reset%201` or
`http://sonoff/cm?cmnd=Reset%2099`

This has never changed as long as ca can remmeber

einarmiger0815, 2019-03-26:
Okay, will look for the simple reboot-command and try...

Jason2866, 2019-03-26:
5.x is that oooooold

jziolkowski, 2019-03-26:
5.x was used by Noah on the Ark...

einarmiger0815, 2019-03-26:
I know, it was an example ...

Jason2866, 2019-03-26:
Please read and respect next time ;-) https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md

einarmiger0815, 2019-03-26:
For sure. Nevertheless, the question is not unjustified, since the repeated change of the link is a fact.

But now I know what I can use. Thank you, that fits.

closed 2019-03-26 by einarmiger0815