arendst/Tasmota #5532: Help - Papertail logging setup

sibero80, 2019-03-26:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No problem, just a question regarding a current feature.

**Describe the solution you'd like**
¿Has someone had experience with https://papertrailapp.com for Tasmota Logging?

**Describe alternatives you've considered**
I dont want to use my RPI for logging.

**Additional context**
Have tried different methods offered by https://papertrailapp.com without success.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-26:
Have you read https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md ?

blakadder, 2019-03-26:
No

closed 2019-03-26 by blakadder