arendst/Tasmota #5535: MCP230xx - sensor29 commands error out

kobuki, 2019-03-27:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Compiled sonoff-sensors and enabled the following:

```
#define USE_I2C
#define USE_MCP230xx
  #define USE_MCP230xx_ADDR 0x20
  #define USE_MCP230xx_OUTPUT
```

The chip is available:

```
20:14:22 CMD: i2cscan
20:14:22 RSL: stat/remote-test/RESULT = {"I2CScan":"Device(s) found at 0x20"}
```

But any sensor29 command just returns an error:

```
20:11:52 CMD: sensor29
20:11:52 RSL: stat/remote-test/RESULT = {"Command":"Error"}
20:12:03 CMD: sensor29 reset
20:12:03 RSL: stat/remote-test/RESULT = {"Command":"Error"}
...
20:21:15 CMD: sensor29 0,?
20:21:15 RSL: stat/remote-test/RESULT = {"Command":"Error"}
```

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : NodeMCU/generic module
- [x] Tasmota binary firmware version number used : sonoff remote-test Version 6.5.0.2(sensors)-2_4_2 / self-compiled
- [x] Development IDE - Compiler / Upload tools used : VSCode + PIO
- [x] Provide the output of command ``status 0`` :
```
20:19:18 CMD: status 0
20:19:18 RSL: stat/remote-test/STATUS = {"Status":{"Module":18,"FriendlyName":["remote-test"],"Topic":"remote-test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:19:18 RSL: stat/remote-test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:07:59","StartupUTC":"2019-03-27T19:11:19","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":21,"SaveAddress":"F7000"}}
20:19:18 RSL: stat/remote-test/STATUS2 = {"StatusFWR":{"Version":"6.5.0.2(sensors)","BuildDateTime":"2019-03-27T20:10:35","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:19:18 RSL: stat/remote-test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["altair",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:19:18 RSL: stat/remote-test/STATUS4 = {"StatusMEM":{"ProgramSize":555,"Free":448,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","0FDEE394","2407A3A4","B7FFBFCC","005ABBC0"]}}
20:19:18 RSL: stat/remote-test/STATUS5 = {"StatusNET":{"Hostname":"remote-test-7337","IPAddress":"10.11.12.99","Gateway":"10.11.12.1","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.1","Mac":"5C:CF:7F:00:1C:A9","Webserver":2,"WifiConfig":4}}
20:19:18 RSL: stat/remote-test/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_001CA9","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:19:18 RSL: stat/remote-test/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 27 19:19:18 2019","Local":"Wed Mar 27 20:19:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:38","Sunset":"19:12"}}
20:19:18 RSL: stat/remote-test/STATUS10 = {"StatusSNS":{"Time":"2019-03-27T20:19:18","ANALOG":{"A0":0}}}
20:19:18 RSL: stat/remote-test/STATUS11 = {"StatusSTS":{"Time":"2019-03-27T20:19:18","Uptime":"0T00:07:59","SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xx:xx:xx:BC:9F:DB","Channel":1,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Provide the output of console when you experience your issue if apply : N/A

What could be the issue? Is it a faulty MCP chip? Do I miss a compile option?

Jason2866, 2019-03-27:
Yes, you have to compile yourself. Is not included in any precompiled version
https://github.com/arendst/Sonoff-Tasmota/blob/development/RELEASENOTES.md

kobuki, 2019-03-27:
@Jason: please see the info in my initial post:

> Tasmota binary firmware version number used : sonoff remote-test Version 6.5.0.2(sensors)-2_4_2 / **self-compiled**

I even specified the required flags!

kobuki, 2019-03-27:
In any case, `STATUS10` should list all the inputs on the MCP23017 I'm using. At least that's what I saw in other people's `status 0` output. Something's not quite right, but I'm not sure what. It's as if my node didn't initialize MCP23017 functionality despite explicitly enabling it. Maybe a bug?

kobuki, 2019-03-27:
OK, my mistake. At an update I let git overwrite my `platformio.ini` and it lost the `-D USE_CONFIG_OVERRIDE` flag. All is working now, sorry for the noise.

closed 2019-03-27 by kobuki