arendst/Tasmota #5537: ap mode not working

gitterman, 2019-03-27:
I accidentally entered a wrong wifi password and now I am stuck.
I had flashed an OBI v2 with the latest 6.5.0 firmware.
Then I booted the device and connected to the AP with my tablet.
I opened the 192.168.4.1 with opera and did a search for my network.
After I had entered the wrong wifi password the device did a reboot but of course it could not connect.
So I tried to reset the device by pressing the ON/OFF button for 40s but nothing happened.
I tried to push the button quickly 4 times but no AP mode appeared either.
Maybe the button of the OBI is not recognized as it is possibly connected to a different port than usual?
I then did a re-flash of the device but no change either.
How can I get the device working again?

ascillato2, 2019-03-28:
Hi, 

In that case, you need to erase all flash with esptool.py as explained in the wiki and flash again. Sorry. 

Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-28 by ascillato2

gitterman, 2019-03-28:
Thanks for pointing me to the solution.
I did a search before I opened the issue but could not find anything appropriate.
You have marked this as duplicate and it would be nice if you could add the original issue for which #5537  is a duplicate.

Jason2866, 2019-03-28:
One chance before reflash. Set your WLAN to open. Means not using a  password. So your device will connect and you can change pwd to correct one. Set back you WLAN to a secured after this change again.

gitterman, 2019-03-28:
I did an erase and a re-flash and now I got it working.
The tool I use is esptool from the arduino IDE on windows and the command to erase was:
 esptool.exe -vv -cd nodemcu -cb 57600 -ca 0x00000 -cp COM6 -bz 2M -ce