arendst/Tasmota #5540: Memory leak in Version 6.5.0.2(7c89ae8-sonoff)-2_4_2

Jason2866, 2019-03-28:
### BUG DESCRIPTION
free memory is decreasing -> device is restarting
Tried with core 2.4.2. and core 2.30 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : NodeMCU
- [ x] Tasmota binary firmware version number used : 6.5.0.2(7c89ae8-sonoff)-2_4_2 / (pre-compiled)
- [ x] Development IDE - Compiler / Upload tools used : Esptool.py
- [ x] Provide the output of command ``status 0`` :
```
00:00:00 Project sonoff BueroKlima Version 6.5.0.2(7c89ae8-sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-5D3631-5681...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-5D3631-5681 with IP address 192.168.2.122
00:00:05 UPP: Multicast (re)joined
12:00:06 MQT: Attempting connection...
12:00:06 MQT: Connected
12:00:06 MQT: tele/sonoff-5D3631/LWT = Online (retained)
12:00:06 MQT: cmnd/sonoff-5D3631/POWER = 
12:00:06 MQT: tele/sonoff-5D3631/INFO1 = {"Module":"Generic","Version":"6.5.0.2(7c89ae8-sonoff)","FallbackTopic":"cmnd/sonoff-5D3631_fb/","GroupTopic":"sonoffs"}
12:00:06 MQT: tele/sonoff-5D3631/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5D3631-5681","IPAddress":"192.168.2.122"}
12:00:06 MQT: tele/sonoff-5D3631/INFO3 = {"RestartReason":"Software/System restart"}
12:00:06 UPP: Multicast (re)joined
12:00:14 MQT: tele/sonoff-5D3631/STATE = {"Time":"2019-03-28T12:00:14","Uptime":"0T00:00:15","Vcc":2.689,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":8,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:00:14 MQT: tele/sonoff-5D3631/SENSOR = {"Time":"2019-03-28T12:00:14","AM2301":{"Temperature":23.7,"Humidity":44.6},"BME280":{"Temperature":24.1,"Humidity":36.0,"Pressure":989.0},"PressureUnit":"hPa","TempUnit":"C"}
12:02:00 MQT: tele/sonoff-5D3631/UPTIME = {"Time":"2019-03-28T12:02:00","Uptime":"0T00:02:01"}
12:04:32 CMD: status 0
12:04:32 MQT: stat/sonoff-5D3631/STATUS = {"Status":{"Module":18,"FriendlyName":["BueroKlima"],"Topic":"sonoff-5D3631","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:33","StartupUTC":"2019-03-28T10:59:59","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":23,"SaveAddress":"FA000"}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS2 = {"StatusFWR":{"Version":"6.5.0.2(7c89ae8-sonoff)","BuildDateTime":"2019-03-28T12:01:38","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":1,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5D3631-5681","IPAddress":"192.168.2.122","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:5D:36:31","Webserver":2,"WifiConfig":5}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-5D3631","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 28 11:04:32 2019","Local":"Thu Mar 28 12:04:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:59","Sunset":"18:36"}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS10 = {"StatusSNS":{"Time":"2019-03-28T12:04:32","AM2301":{"Temperature":23.8,"Humidity":44.3},"BME280":{"Temperature":24.1,"Humidity":36.0,"Pressure":989.0},"PressureUnit":"hPa","TempUnit":"C"}}
12:04:32 MQT: stat/sonoff-5D3631/STATUS11 = {"StatusSTS":{"Time":"2019-03-28T12:04:32","Uptime":"0T00:04:33","Vcc":2.693,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":8,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
Configure BME280 on D6 (SCL), D7 (SDA), D5 AM2301 (2)

### EXPECTED BEHAVIOUR
Memory not decreasing. 

### SCREENSHOTS

```
12:09:41 MQT: tele/sonoff-5D3631/LWT = Online (retained)
12:09:41 MQT: cmnd/sonoff-5D3631/POWER = 
12:09:41 MQT: tele/sonoff-5D3631/INFO1 = {"Module":"Generic","Version":"6.5.0.2(7c89ae8-sonoff)","FallbackTopic":"cmnd/sonoff-5D3631_fb/","GroupTopic":"sonoffs"}
12:09:41 MQT: tele/sonoff-5D3631/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5D3631-5681","IPAddress":"192.168.2.122"}
12:09:41 MQT: tele/sonoff-5D3631/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x40236b93 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
12:09:41 UPP: Multicast (re)joined
```

arendst, 2019-03-28:
Investigating.

arendst, 2019-03-28:
What happens if you turn emulation off (``emulation 0``)

Jason2866, 2019-03-28:
Yes, it has to be in emulation. If turned off, memory is not decreasing

Jason2866, 2019-03-28:
The bug only occurs if NO relais is defined. 
I defined now a relais -> no more memory decrease

Jason2866, 2019-03-28:
Error occurs only if NO relais is defined

arendst, 2019-03-28:
Great! I know what I did to make this happen. I also have a solution but I'm still searching for the exact cause as it is related to _udp.parsePacket(). For some reason I have to call this regularly even if no device is configured (that was what I changed) as otherwise it will steal memory for every UDP message coming in resulting in your memory leak experience.

Almost there...

arendst, 2019-03-28:
Solved! The parsePacket() function needs to be called regularly as this will flush previous UDP packets received. I didn't do it when no device was configured resulting in the memory flooding by received UDP packets.

While I was testing I always had a device configured.

Thx for finding this one!

Jason2866, 2019-03-28:
Thx for fast fixing :-)