arendst/Tasmota #5541: New ESP8266 dimmer - cannot get it to work

bigcookie, 2019-03-28:
I found this dimmer on Amazon: https://www.amazon.de/TEEKAR-Timing-Funktion-Fernbedienung-Lichtschalter-Neutralleitung/dp/B07M78J15Q/
which looks quite nice. For experimental purposes, I purchased it and it is based on a ESP8266 module with 1MB flash. The module is marked as HYLE2S. The flash chip is an 8MBit type.
The dimmer control works via a ST Microelectronics MCU: stm8s003f3. The MCU is connected to RX/TX of the ESP8266.

I desoldered the module, backed the original firmware up and flashed it with Tasmota. I tried several IDs - seems not to work. Also on/off not working. I will try to work with the templates, but I assume the commands might be different for the MCU...

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Teekar Smart Dimmer
- [x] Tasmota binary firmware version number used : 6.5.0 / (downloaded from https://github.com/arendst/Sonoff-Tasmota/releases
- [x] Development IDE - Compiler / Upload tools used : - / esptool
- [ ] Provide the output of command ``status 0`` :
```
15:45:52 CMD: status 0
15:45:52 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:45:52 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:37","StartupUTC":"2019-03-28T14:27:15","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":25,"SaveAddress":"FB000"}}
15:45:52 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:45:52 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndroLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280502000100000000000000000000000000","00000000"]}}
15:45:52 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
15:45:52 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"SH-Dimmer1","IPAddress":"192.168.20.38","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"68:C6:3A:EB:C5:3B","Webserver":2,"WifiConfig":4}}
15:45:52 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EBC53B","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:45:52 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 28 14:45:52 2019","Local":"Thu Mar 28 15:45:52 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:36","Sunset":"19:14"}}
15:45:52 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-28T15:45:52"}}
15:45:52 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-28T15:45:52","Uptime":"0T00:18:37","Vcc":3.207,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":126,"POWER":"ON","Dimmer":32,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"AndroLAN","BSSId":"44:4E:6D:5F:22:03","Channel":5,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

bigcookie, 2019-03-28:
[Teekar_image_1M.bin.zip](https://github.com/arendst/Sonoff-Tasmota/files/3018550/Teekar_image_1M.bin.zip)

Original backup of firmware uploaded

Jason2866, 2019-03-28:
This is not a Tasmota issue. Please read https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md

bigcookie, 2019-03-28:
I know this, but the question is, howto start contribution if I cannot send a message to the original developer of the Tuya support ia Github? Any hint is highly welcome (e.g. how to send a message to a github member - never figured this out...).
Anyway thanks for the reply

Jason2866, 2019-03-28:
Tuya Convert can be found here https://github.com/ct-Open-Source/tuya-convert/issues
Tasmota is in NO way related to Tuya Convert nor does Tasmota claim to support other devices
as being listed at Tasmota wiki (see Readme).

bigcookie, 2019-03-28:
Thanks. got in contact with the original developer. Flashing was not a problem (Tuya-convert), I flashed tasmota on it, but I cannot get it working - potentially a different protocol. Feel free to close this topic

ascillato2, 2019-03-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-28 by ascillato2

erdii, 2019-05-31:
@bigcookie would you mind sharing what you learnt about the teekar, especially the pinout? 👍

bigcookie, 2019-06-03:
Hello @erdii ,
I stopped working on this and discarded the pictures. Sorry. In general, I did the following:
removed the ESP module (which is a standard Tuya one) and checked all GPIOs. My findings can be found here: https://github.com/arendst/Sonoff-Tasmota/issues/469#issuecomment-477997302

In general the ESP module was sending brightness values to the MicroController. IT didnt receive anything back. Thus in this case, the full logic of handling faceplate etc is done in the ESP, the MC is just setting the brightness to the light.
The Tuya dimmer solution in Tasmota is based on the MC keeping the state if I recall correctly and handling the inputs of the faceplate. The ESP is just the connection to the Wifi world.

My summary: Tasmota as is cannot be used for this dimmer AND it is not a simple alteration of the existing code - at least if you want to use the faceplate :-).

I hope this helps. I still have the device, but I am now using one of the standard Tuya dimmers.
-> The standard ones are awful as I cannot "feel" the point to switch the light on.
-> The Teekar was much better from haptic design.

regards

erdii, 2019-06-03:
Wow thank you @bigcookie for all this information! I'll try to integrate to get the esphome firmware to work with this dimmer. Will get back to you :)

bigcookie, 2019-06-05:
@erdii : let me know - I am happy to test any firmware on the dimmer. It is now lying around, waiting to be MQTT integrated in my network ;-). Let me know if you need people to try it out.

erdii, 2019-06-19:
@bigcookie i have mixed news for you

sooo... I managed to kill my dimmer... destroyed it while trying to take out the pcb holding the esp chip. I already had it out once to flash bare esphome firmware, but then I wanted to flash the original firmware to sniff the touch input protocol. Took it out and, boom, nothing working anymore.

What I managed to accomplish: code a "Light" output component for esphome -> dimmer exposed.
What I not managed:
* sniffing the touch input protocol
* code a simple component for touch led output
* wiring together one big component that combines touch input, touch led output and dimmer output on device (this should be a breeze when the touch procotol is implemented and can be read)

If you want, I can post my light code and a small manual on how to create an esphome firmware with my custom class included.

bigcookie, 2019-06-19:
Hi @erdii ,

not sure, where you are living. I currently not using my dimmer as it doesnt work in my setup :-(. If you are sitting in Germany, I could send you mine. Abroad might be too expensive for this device in comparison with purchasing it.
Alternative idea: you can get a ESP PCB and replace it (if the ESP is broken). I still have the 1M original firmware backup... if you needed... :-).
Unfortunately I dont have the proper tools to sniff the protocols (multimeter, no digital oscilloscope or other things).

When I tried to handle the device: I connected the ESP to a breakout board and wired the dimmer hardware to the board - just as an idea to avoid continuous soldering. The effort was a bit higher, but probably worth a try (if you didnt do this already).

bigcookie, 2019-06-19:
And please share our code (I probably cannot develop it further, but if other people are interested...).

erdii, 2019-06-19:
good idea! I desoldered the board for flashing and then resoldered it into the dimmer pcb.
I actually do live in Germany, could you send me an email to the address in my github profile?

For sniffing I bought this: http://dangerousprototypes.com/docs/Bus_Pirate but wasn't able to try it yet, because the dimmer died 💀 

Ok! I will setup a repo for code sharing!

miruim, 2019-07-10:
hi.... Any news. I am in same situation.. can not control it

bigcookie, 2019-07-11:
I am not a (trained/good) developer and sent my dimmer to @erdii . I hope he will have time to take a look at it...

Jason2866, 2019-09-25:
Tuya switches and dimmers and more devices are supported via TuyaMCU
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#TuyaMCU
and
https://github.com/arendst/Sonoff-Tasmota/wiki/TuyaMCU

erdii, 2019-09-26:
Hey all, sadly i had quite some stuff going on in my private life and could not allocate time to reverse engineer the dimmer for esphome :(

myqel, 2019-11-17:
This issue is closed, but I want to add in two sentences my current investigations, as I have this dimmer purchased a couple of days ago:
Yes, the ST MCU doesn't send anything to the Esp 8285 Tuya Tywe2s Clone (no shielding). GPIO12 seems to be the button (measured), GPIO14 is connected to "LEDs" (measured) of the connector towards the touch/LED PCB. GPIO04 and 05 have a kind of PWM each, but in combination to each other (sync'd) (MCU->ESP?). These signals are permanent available, even in OFF mode. GPIO12 seems to be a clock signal with 10 pulses, no synchronized with GPIO4/5 signals. This signal appears with dimmer value change only. I have oscilloscope screenshots. The ESP->MCU message for dimming is "55 AA 00 01 05 00 00 xx yy" xx=value, yy=checksum, on/off messages are available, too. I have to thing about it as the signals seem not to be a serial protocol (I2C, SPI) as checked the pin function levels of the ESP8266 so far. Cheers; Meikel

myqel, 2019-11-17:
Some additional short facts: The SPI flash on the clone ESP PCB is 1 MB of size. As the MCU does not send anything to the ESP (but it is physically connected, I checked it (Pin 2 of the ST MCU)), I was able to flash back the original backup file in-curcuit with a 3.3V FTDI adapter. I supplied the whole dimmer PCB with a breadboard 3.3V supply (no mains connection as there is no decoupling! BTW: Bad soldering of the 230V srew contacts, I would recomment to re-solder them). The touch control of my device here changes randomly since a while. Maybe it got broken in meantime or it catches disturbances by the wiring or supply. Regards; Meikel

Abbadon89, 2019-11-17:
Good job. I don't have much experience to help but i believe that this good product become tasmotized :D

vaxus1, 2019-12-11:
hei. nice job so far. I bought this dimmer time ago. want to upgrade it to tasmota but did not get to it so far.

blakadder, 2019-12-12:
I got the dimmer in the previous post. 
Hooked up to the rx/tx pins while the original firmware was on it and got the codes used to control the dimming function. Now to figure out how do the mcu and esp communicate exactly. 

When Tasmota is installed none of the on board functions work (touch pad or button)

myqel, 2019-12-12:
The Teekar dimmer consists of three software pieces to be combined in the ESP:
1) The communication with the dimmer MCU: This is close to TUYA, same protocol, but only direction from ESP to MCU using a different command compared to the TUYA white book, see the link bigcokie gave a little bit earlier in this thread. Button is on GPIO12 and can be served by Tasmota SW.
2) The 8 LEDs are served by a HC164 shift register IC on the 2nd PCB, the connector says 'LEDS' for the shifted bits and 'CLK' for the shiftig clock. Look for the same link bigcookie gave (great work) in this thread (https://github.com/arendst/Tasmota/issues/469#issuecomment-477997302)
3) The communication with the touch sensor IC. This seems to be an I2C link I assume, because the other pins of the connector are called 'SDA' and 'SCL'. Unfortunately the IC on the 2nd PCB has no label. (Something like microchip AT42QT1070). And with a fast read through the ESP µC docs the GPIO pins connected to the lines 'SDA' and 'SCL' seems not the hardware I2C. Maybe a Soft-I2C? Or I missed something in the docs and it is the hardware I2C.
So I have to log data on the I2C signals to reverse engineer the I2C address(es) and commands. Up to now I had no time to do so, I hope, the christmas vacation time give a chance to investigate further.

Regards, miqel

blakadder, 2019-12-13:
I2C is on IO4 - SDA and IO5 - SCL, My chip only has HLS028 labelled on it.

tasmota finds the i2c device with i2cscan
`{"I2CScan":"Device(s) found at 0x56"}`

myqel, 2019-12-21:
So, I managed to grab a couple of I2C data.
Unfortunately my dimmer seems to catch disturbances from environment through the attached cables or it is broken. I cannot get it in a stable condition without changing the dimming value all the time.
There seems to be a poll with 50Hz (20ms) from µC to cap IC even in off mode.
I2C address my data logger decoded is 0xD8.
@blakadder : could a place a zip file for you with text files somewhere on your web site?
In the meantime I look into data sheets of different cpacitive sensor ICs, maybe there is a hit.

blakadder, 2019-12-21:
simplest way is for you to open an issue there and post a link to a hastebin or wherever

epicurean70, 2020-01-21:
Any updates on this? Still no luck on configuring this dimmer?

Abbadon89, 2020-05-24:
Anyone? please dont let this topic die...

dollop80, 2020-11-24:
Just started digging.
I've found the following:
1. The ESP polls the IC with address 0x56 each 20ms. The data transfer consists of the following reads/writes
[image]
2. Each time the touch is pressed there are three consecutive messages that contain the code of the touched sensor: the codes are the following: K1 = 0x02, K2 = 0x04, K3 = 0x08, K5 = 0x10, K6 = 0x20, K7 = 0x40
[image]
3. For K8 sensor the pattern is different:
[image]
4. Just after a sensor has been touched and the described earlier pattern appeared in the I2C bus the ESP controls the LED using clock and led pin (it does it twice with interval of 600us):
[image]
This is a close up for the first LED:
[image]
This is for the fourth LED:
[image]

So, now I need to figure out how to add this to tasmota :)

dollop80, 2020-11-27:
Hi, I've modified tasmota 9.1.0 code and added the new template. The dimmer works beautifully with tasmota now: dimmer, front panel leds, touch, everything is ok. The only thing I'm missing is fade. May be will add it later...
[tasmota_teekar.zip](https://github.com/arendst/Tasmota/files/5606779/tasmota_teekar.zip)

blakadder, 2020-11-27:
can you provide code modifications so I can compile my own version?

dollop80, 2020-11-27:
Hi,
Sure, check my fork https://github.com/dollop80/Tasmota/tree/Teekar
Use "teekar" branch

dollop80, 2020-11-28:
Sorry, forgot to track some files in the repository. Now its OK. Good luck

daniel-a-h, 2021-02-01:
@dollop80 thank you for your effort! I have one of these teekars sitting around and will probably be using your code.

Question: has it been merged into the upstream and is it Included in tasmota 9.2? I'm asking because there we also have the auto detect feature for home assistant.

@blakadder: does this mean the documentation for the templates is outdated?

blakadder, 2021-02-01:
Someone needs to properly merge the new drivers without using anew module to the upstream