arendst/Tasmota #5543: Support for SANT-1CH / WiFi & RF 433 MHz

S474N, 2019-03-28:
Is possible to add support/firmware for SANT-1CH / WiFi & RF 433 MHz.

- input 5V MicroUSB
- output relay 220V
- WiFi & RF 433 MHz

This device is fully working with eWeLinK, which identify it as:
Brand/Model: SANTTONG / SANTTON_2
Firmware version: 3.0.0

Photo of device:
http://temp.s474n.com/SANT_1CH.jpg

Jason2866, 2019-03-28:
Since we dont have the device we dont know...
Try it, if you got it working you findings are welcome
See https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md

S474N, 2019-03-28:
Understand, but it's over my strength. :( 

This is the device:
https://www.aliexpress.com/item/SONOFF-5V-12V-220V-RF-433MHz-Wifi-Switch-Wireless-Relay-Module-Remote-Switch-for-Android-IOS/32978091649.html

Or do you know about another device, which can be controlled via WiFi (and to be implemented to Node-RED/HomeBridge) and RF together and input is 5V or 12V and separated from output? I'd like to use it as garage door opener. Thank you for reply.

Jason2866, 2019-03-28:
Using a cheap rf433 device with fixed code isnt a secure solution. It needs just ONE code sniffing...
If you want it to do this way get a sonoff-sv and a sonoff rf-bridge.
You can connect them via NodeRed (or just using rules from Tasmota)

S474N, 2019-03-28:
And Sonoff remotes have floating code?

NodeRED I need, because it's way to HomeBridge and HomeKit :) Actually I have this device in HomeKit, but as "Switch" and I want "GarageDoor" :)

Jason2866, 2019-03-28:
No, it is all fixed code...

S474N, 2019-03-28:
OK, therefore there is same to use my device or original Sonoff :)

Does exists some device with normal price and with floating code?

ascillato2, 2019-03-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-28 by ascillato2

S474N, 2019-03-28:
Sorry for chat, but support for SANT-1CH is still valid ;)

Jason2866, 2019-03-29:
It is closed because https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md

bluefoxlee, 2019-06-09:
@S474N Got my Sant-1ch wifi version today, and it works fine with Tasmota. Try look close at back of PCB, and there are same GND/VCC/TX/RX just in different order. should not be a big issue to flash it.

Will try to update to wiki later (since I never use wiki before).

S474N, 2022-02-15:
@bluefoxlee ok, these:
[image]

And settings after flashing?

S474N, 2022-02-26:
@bluefoxlee can you please add here template settings? Thank you!