arendst/Tasmota #5544: Latest dev b8dfc6b compile error

DavidFW1960, 2019-03-28:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

Compiling latest dev b8dfc6b

Errors:
```
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development b8dfc6b/sonoff/settings.ino: In function 'void SettingsDefaultSet2()':
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development b8dfc6b/sonoff/settings.ino:818:29: error: 'WORKING_PERIOD' was not declared in this scope
Settings.novasds_period = WORKING_PERIOD;
^
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development b8dfc6b/sonoff/settings.ino: In function 'void SettingsDelta()':
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development b8dfc6b/sonoff/settings.ino:1061:33: error: 'WORKING_PERIOD' was not declared in this scope
Settings.novasds_period = WORKING_PERIOD;
```
### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : _____
- [ x] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ x] Development IDE - Compiler / Upload tools used : Visual Studio Code PlatformIO
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-03-29:
Hi,

Sorry, I could not reproduce it. Besides Tasmota is passing Travis compilation ok for all languages and version explicited in platformio.ini file (see compilation badge in readme) 

Please, try to erase all files and download again in a fresh folder. Thanks

DavidFW1960, 2019-03-29:
I did that Adrian... I always download the full zip file to a fresh folder. I'll try again.

DavidFW1960, 2019-03-29:
I downloaded again and got the exact same result.
I also re-compiled the firmware from a couple of days ago and got no error.

I always download the fill zip file and unzip to a new folder and then make the necessary changes in my_user_config.h and platformio.ini
This has always worked.... (the 6.4.1.x problem excepted)

wongnam, 2019-03-29:
I only saw this.

[image]

DavidFW1960, 2019-03-29:
Same error it seems and compilation fails...

Jason2866, 2019-03-29:
There is no error when compiling latest dev version using VSC.
Your setup has to be faulty.
[image]
@wongnam 
This is just a cosmetic issue. Fixed with open PR https://github.com/arendst/Sonoff-Tasmota/pull/5524

arendst, 2019-03-29:
Define WORKING_PERIOD 5 should be in your my_user_config.h file.

As of the introduction of command Sensor20 yesterday it is mandatory. For those who commented it out I'll re-add it to settings.ino someday...

DavidFW1960, 2019-03-29:
Ah. Thanks. I’ll take a look at that. That would explain it. I’ll confirm tomorrow.

ascillato2, 2019-03-30:
Closing issue as it is not a Tasmota bug. 

Please, if you need further assistant, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-30 by ascillato2

DavidFW1960, 2019-03-31:
Seems @arendst has added it back in settings.ino already. Thanks... it's now compiling with latest dev branch