arendst/Tasmota #5545: Adding a new template for a new module

Rambylon, 2019-03-29:
**Have you look for this feature in other issues and in the wiki?**
Yes, i did look in the other issues and in wiki but i didn't find solution as this device was not using GPIO pins for control. It uses Serial Data for controling through RX and TX ports. The "Tuya Dimmer" module works for only one channel control from Tasmota(the first one) but it receives state change from the device for all channels, though it shows only one channel in Tasmota ( CH1 on/off in device-> turns on/off in tasmota, CH4-> on/off in device-> turns on/off in Tasmota, and respectively for all the 4 channels).

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
The problem is that this device (Wallpad 4CH Touch Wifi Switch) is Tuya based and it uses Serial data for communicating with a secondary MCU for switching the relays, rather than using on-board GPIO pins for relay control. So the Module configuration in Tasmota was useless as they can only configure the GPIO pins for I/O. It can be controlled via the console and MQTT using the serial codes but it can not transmit the state changes in MQTT.And can not be controlled from the Tasmota GUI.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I have identified the Serial codes used to communicate with the secondary MCU for relay control (both command codes and response codes). I can successfully use the serial codes for testing the relay controls in the Tasmota Console. The solution that i wish is the implementation of these codes to create a new template (or module) in Tasmota for this device. So that the device can get controlled from the Tasmota GUI and MQTT.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
If the above mentioned idea is not possible, then i wish a method for transmitting the state data of the device through MQTT. As the switches are not configured, getting the state data from the device through MQTT is not possible. 

**Additional context**
_Add any other context or screenshots about the feature request here._
The link to the product in Aliexpress is https://www.aliexpress.com/item/New-Arrival-4-Gang-Wifi-Switch-Wallpad-White-Tempered-Glass-UK-EU-Standard-4-Gang-Wifi/32892658176.html
[image]
[image]
[image]
The Serial Codes for this device
[tuya_4gang_wallpad_uk.xlsx](https://github.com/arendst/Sonoff-Tasmota/files/3021109/tuya_4gang_wallpad_uk.xlsx)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-02:
You can use rules to get the device usable with Tasmota. Solution is similar to this https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay

Rambylon, 2019-04-05:
@Jason2866  when in generic template the serial communication doesn't work. when configuring GPIO 0 and GPIO 2 for relays while in tuya dimmer template it doesn't even show the relay control in the main menu.

Jason2866, 2019-04-05:
You have to use module generic. I dont understand why serial communication doesnt work for you.
I use serial bridge function in module generic....

Rambylon, 2019-04-05:
dont know. it only send and recieve serial data in "Tuya Dimmer" template.

thirug010, 2019-04-06:
I bought the US version of the Tuya  4 Ch touch switch, it does not use the Tuya serial, it directly use the gpio's as below.

{"GPIO0":"25 (Relay5)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"19 (Button3)","GPIO4":"23 (Relay3)","GPIO5":"17 (Button1)","GPIO12":"20 (Button4)","GPIO13":"24 (Relay4)","GPIO14":"22 (Relay2)","GPIO15":"21 (Relay1)","GPIO16":"18 (Button2)"}.

[image]
[image]
[image]
[image]

Rambylon, 2019-04-07:
where did you buy it? can you send the link?

thirug010, 2019-04-07:
https://www.alibaba.com/product-detail/SESOO-Smart-home-Amazon-alexa-compatible_60601674427.html?spm=a2700.7724838.2017115.1.7e6b1505r3ArLm

the same is available in the Aliexpress but i dont have the link you get it
from alibaba chat

thirug010, 2019-04-07:
https://www.alibaba.com/product-detail/SESOO-Smart-home-Amazon-alexa-compatible_60601674427.html?spm=a2700.7724838.2017115.1.7e6b1505r3ArLm

the same is available in the Aliexpress but i dont have the link, you can get it
from Alibaba chat

Rambylon, 2019-04-07:
thank you. how is the touch response in the switch? does it respond immediately or it takes a slight delay to react?

thirug010, 2019-04-07:
when I generic and GPIO it has 1 sec delay but after defining the template in tasmota it works fast.

{"NAME":"Tuya 4 Channel","GPIO":[255,255,255,19,23,17,0,0,20,24,22,21,18],"FLAG":0,"BASE":18}

GPIO0 is the Wifi Icon LED if you want to control.

Rambylon, 2019-04-07:
good work

blakadder, 2019-04-08:
I think this thread has reached a conclusion and therefore its being closed

Rambylon, 2019-04-09:
what the hell? what is that conclusion? do you know how it can be controlled @blakadder ? when i came to you for help, you wont help me with my problem. ok i left you alone. Now why did you close this? what the hell is wrong with you?

thirug010, 2019-04-10:
Hi

I have done some analysis and found the following, have special Tuya
Devices not fully supported by Tasmota Tuya code,
as Tuya get more popular we may expect some new devices.

My findings,

1. I*t assume only one power for Tyua Serial devices.*
update devices_present = 4 in sonoff.ino, will allow you to add button or
extra power for current Tuya dimmer with GPIO's
devices_present = 1 fro dimmer. In my case it 3 (dimmer 2 , dimmer power 1
, other power 3)
else if (TUYA_DIMMER == my_module_type) // if TUYA_XXXXXX if you defined
new template
{
devices_present = 3; // devices_present = Setoption; to better support for
new devices
}
2. *It assume the Power is always Device 1, but in reality it  can change
based on the device. *
   standard Tyua dimmer power is 1but various for device, as per the
Protocol tuya_buffer[6] dimmer id

*Example: *I have the Tyua device  has Fan Dimmer and Light switch (link :
https://www.amazon.com/gp/product/B07M77KCYJ/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1
)

in my device

MCU to ESP --> 55AA010700050*3*0100010112

    case TUYA_CMD_STATE:
  *  ......*
      *  if(tuya_buffer[6] == 1 ) // Device -->1 Dimmer Power *
        {
          if((power || Settings.light_dimmer > 0) && (power !=
tuya_buffer[10])) {
            *ExecuteCommandPower(1*, tuya_buffer[10], SRC_SWITCH);  // send
SRC_SWITCH? to use as flag to prevent loop from inbound states from
faceplate interaction
          }
        }
        *if(tuya_buffer[6] == 3 ) // Device -->3 Light Switch *
        {
            *ExecuteCommandPower(3*, tuya_buffer[10], SRC_SWITCH);  // send
SRC_SWITCH? to use as flag to prevent loop from inbound states from
faceplate interaction
        }
        //in your case you need add 1,2,3,4 for tuya_buffer[6] and remove
the Dimmer part as it is device 2
ESP to MCU --> 55AA000600050*3*0100010110, I used XdrvMailbox.notused to
pass the deviceId  from "Sonoff.ino" and "xdrv_04_light.ino"
bool TuyaSetPower(void)
{
  bool status = false;
  uint8_t deviceId = XdrvMailbox.notused;
  uint8_t rpower = XdrvMailbox.index;
  int16_t source = XdrvMailbox.payload;

  AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TYA: SetDevicePower.rpower=%d, deviceId=
%d"), rpower, deviceId);

  if (source != SRC_SWITCH && TuyaSerial) {  // ignore to prevent loop from
pushing state from faceplate interaction
if(deviceId == 1) // 3 is dimmer
{
  TuyaSendBool(1, rpower);
}
if(deviceId == 3) // 3 is dimmer
{
  TuyaSendBool(3, rpower);
}
status = true;
  }
  return status;
}
Thanks
Thiru

Rambylon, 2019-04-19:
im very well confused and im not expert with tasmota recoding. can you help me out a bit? i would be very grateful

thirug010, 2019-04-20:
Hi,

      I am updating the code based on the your Serial Data in the Excel
file, I will get back to you ASAP.

Thanks
Thiru

Rambylon, 2019-04-20:
thank you @thirug010 are you tamil?

thirug010, 2019-04-20:
Yes

thirug010, 2019-04-21:
Hi @Rambylon,

I have updated the 6.5.0 code to support 4 gang Switch. I tested with Wemo + serial Terminal for the communication, 

1. Tasmota Web UI ON /OFF send the Serial to the Terminal ( tested for all switches )
2.. MQTT power command trigger the Serial to the Terminal 
3.  Serial Data from your Excel switch ON /OFF the Tasmota / MQTT. 

update the sonoff.ino and xdrv_16_tuyadimmer.ino from the .zip folder and compile and update the firmware. 

[image]

Thanks 
Thiru  

[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/3100800/sonoff.zip)

Rambylon, 2019-04-21:
thank you @thirug010  i flashed the update through ota. and the firmware now recognizes 4 switches. switching them works perfectly when done one by one. but when more than two switches are on and when we try to turn off one it doesnt work. the relay doesnt click and the display shows off for a second and goes back to its previous state.
Like this 
[image]
why do you think this happens? and share me a way to contact you personally other than this issue. im also from your state.

thirug010, 2019-04-21:
Can you give me the Console log with option 4 I will check and update you

My email : thirug010@gmail.com

Thanks
Thiru

Rambylon, 2019-04-21:
the console log with setoption4 @thirug010 

from web gui
```
09:26:59 SRC: WebGui from 192.168.0.131
09:26:59 TYA: Send Dim Level skipped due to 0 or already set. Value=0
09:26:59 TYA: TX Packet: "55aa0006000501010001010e"
09:27:00 MQT: stat/tuya/POWER1 = {"POWER1":"OFF"}
09:27:00 MQT: stat/tuya/POWER1 = OFF
09:27:00 TYA: RX Packet: "55aa0007000501010001010f"
09:27:00 TYA: RX Device -1 --> On State
09:27:00 SRC: Switch
09:27:00 TYA: Send Serial Packet Dim Value=25 (id=1)
09:27:00 TYA: TX Packet: "55aa0006000801020004000000192d"
09:27:00 MQT: stat/tuya/POWER1 = {"POWER1":"ON"}
09:27:00 MQT: stat/tuya/POWER1 = ON
09:27:00 SRC: WebGui from 192.168.0.131
09:27:00 TYA: TX Packet: "55aa0006000502010001010f"
09:27:00 MQT: stat/tuya/POWER2 = {"POWER2":"OFF"}
09:27:00 MQT: stat/tuya/POWER2 = OFF
09:27:01 TYA: RX Packet: "55aa00070005020100010110"
09:27:01 TYA: RX Device -2 --> On State
09:27:01 SRC: Switch
09:27:01 MQT: stat/tuya/POWER2 = {"POWER2":"ON"}
09:27:01 MQT: stat/tuya/POWER2 = ON
```

from mqtt

```
09:28:05 SRC: MQTT
09:28:05 RSL: Received Topic cmnd/tuya/POWER3, Data Size 3, Data OFF
09:28:05 RSL: Group 0, Index 3, Command POWER, Data OFF
09:28:05 TYA: TX Packet: "55aa00060005030100010110"
09:28:05 MQT: stat/tuya/POWER3 = {"POWER3":"OFF"}
09:28:05 MQT: stat/tuya/POWER3 = OFF
09:28:05 TYA: RX Packet: "55aa00070005030100010111"
09:28:05 TYA: RX Device -3 --> On State
09:28:05 SRC: Switch
09:28:05 MQT: stat/tuya/POWER3 = {"POWER3":"ON"}
09:28:05 MQT: stat/tuya/POWER3 = ON
09:28:05 SRC: MQTT
09:28:05 RSL: Received Topic cmnd/tuya/POWER4, Data Size 3, Data OFF
09:28:05 RSL: Group 0, Index 4, Command POWER, Data OFF
09:28:05 TYA: TX Packet: "55aa00060005040100010111"
09:28:05 MQT: stat/tuya/POWER4 = {"POWER4":"OFF"}
09:28:05 MQT: stat/tuya/POWER4 = OFF
09:28:05 TYA: RX Packet: "55aa00070005040100010112"
09:28:05 TYA: RX Device -4 --> On State
09:28:05 SRC: Switch
09:28:05 MQT: stat/tuya/POWER4 = {"POWER4":"ON"}
09:28:05 MQT: stat/tuya/POWER4 = ON
09:28:06 MQT: tele/tuya/STATE = {"Time":"2019-04-21T09:28:06","Uptime":"0T01:23:52","Vcc":3.186,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER1":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","POWER2":"ON","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"Beacon 2.4G","BSSId":"D8:0D:17:6B:5F:7A","Channel":6,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:07"}}
```

MikeSaint, 2019-05-13:
@Rambylon 

Did you get any further with a working Tuya touch switch in Tasmota? I have a different 4 gang touch switch (non-dimmer) but the serial codes you provided are identical.

blakadder, 2019-05-13:
@MikeSaint Take a look at https://blakadder.github.io/templates/oil_diffuser_550ml.html for an example on how to get some control of the switches

Rambylon, 2019-05-13:
@MikeSaint no. There is no further development in the matter. Still hoping for further development in the tuya driver.

thirug010, 2019-05-15:
Hi All,

Me and @Rambylon is working in the separate thread to resolve the issue with Custom Tuya based devices (1 dimmer + 1 relay aka Fan Light switch , 4 Relay Tuya serial switch .. ). 
We will keep you posted once all the issues are resolved 

Thanks 
Thiru

MikeSaint, 2019-05-15:
That's great news @thirug010   If you want me to do any testing on the Tuya touch switches (3 & 4 gang) I have, please let me know.

blakadder, 2019-05-16:
reopening since the code is being worked on

thirug010, 2019-05-16:
We are able to test code in the Tuya 4Ch serial based device it is working hard some slowness working to fix it and need adjust it to the 3 ch fan light (1Dimmer 1 relay) devices. 

Will keep you posted. 

@MikeSaint could you send the serial data for 3 gang device for the testing.

Thanks
Thiru

thirug010, 2019-05-17:
I have updated the tasmota to support the custom tuya devices, It can be flashed with USB or OTA with minimal firmware  _**But not with Tuya-Convert directly ( it is 516kb bin file)**_ 

You can the following options to update the code
1 . Use the attached (.zip) files  and update your Sonoff-Tasmota-6.5.0 Branch 
2 . Use cloned the version of the Sonoff-Tasmota-6.5.0 [here](https://github.com/thirug010/Sonoff-Tasmota-6.5.0-New-Tuya-Devices) 
_**Important** this code is not tested on all tuya devices, update it on your own risk, make sure compare the serial data, if you need help please reach out to me_

### Summary of Changes: 
**./sonoff**
_settings.h
sonoff.ino
xdrv_01_webserver.ino
xdrv_04_light.ino
xdrv_16_tuyadimmer.ino_
**./tools**
_decode-config.py_

This code will support up-to 8 Tuya devices as per the TUYA-MCU Protocols. it can be all 8 Relays or 1 dimmer + 6 relays but only one Dimmer can be used at a time.  

- To Enable or Disable Dimmer control use SetOption63 --> 1-Enable 0-Disable 
- To Set No of devices / Relays / Dimmer  use SetOption38 (_**Note:** require re-boot to apply the change_)
   Example 4 Gang SetOption38 - 4 

Here is sample devices and SetOptions .

**Steps use Tuya Dimmer as Switch (4Gang) ..** _Thanks to @Rambylon for Testing and Support_
1. Make sure the Serial Data is matching with your device [tuya_4gang_wallpad_uk.xlsx](https://github.com/arendst/Sonoff-Tasmota/files/3192203/tuya_4gang_wallpad_uk.xlsx)
2. Reset Current Configuration --> it will make the device as basic sonoff 
3. Update the new firmware 
4. SetOption63 as 0 --> will disable the dimmer slider control 
5. SetOption38 as 4 
6. Select Module as Tuya dimmer
[image] 

**Steps use Tuya Dimmer as Fan Dimmer and Light Switch (1 Dimmer, 1 Relay)**
1. Make sure the Serial Data is matching with your device 
[Fan Dimmer Data.zip](https://github.com/arendst/Sonoff-Tasmota/files/3192220/Fan.Dimmer.Data.zip)
2. Reset Current Configuration --> it will make the device as basic sonoff 
3. Update the new firmware 
4. SetOption63 as 1 --> will enable the dimmer slider control 
5. SetOption38 as 3 --> 1-SSR, 2-Dimmer, 3-Relay 
6. SetOption34 as 2 --> (Optional it will be set based on the feedback from MCU on the 1st dim packet) 
7. Select Module as Tuya dimmer 

**Notes:**
- Use Power 1 and Dimmer Control to turn ON/OFF and set the Dimmer Value
- Use Power 3 for the control Light Switch (relay)
- Power 2 is just dimmer control Just sent the dimmer packets. toggle buttn does not work 
[image]

**Steps use Tuya Dimmer as Switch (3Gang)** _Not Tested_
1. Make sure the Serial Data ( first 3 switch)  is matching with your device [tuya_4gang_wallpad_uk.xlsx] (https://github.com/arendst/Sonoff-Tasmota/files/3192203/tuya_4gang_wallpad_uk.xlsx)
2. Reset Current Configuration --> it will make the device as basic sonoff  
3. Update the new firmware 
4. SetOption63 as 0 --> will disable the dimmer slider control 
5. SetOption38 as 3
6. Select Module as Tuya dimmer 

[Tuya-Updates.zip](https://github.com/arendst/Sonoff-Tasmota/files/3192113/Tuya-Updates.zip)
Cloned Repo: https://github.com/thirug010/Sonoff-Tasmota-6.5.0-New-Tuya-Devices

ascillato2, 2019-05-19:
Closing this issue as the PR has been made and it is being reviewed to be merged.

Thanks a lot to everyone who has helped on this :+1:

closed 2019-05-19 by ascillato2

MaliMrav, 2019-07-06:
> Hi I have done some analysis and found the following, have special Tuya Devices not fully supported by Tasmota Tuya code, as Tuya get more popular we may expect some new devices. My findings, 1. I*t [...]

Hi All,
I too had to analyze thirug010  the code as I have tuya switches 1,2,3 and 4 gang and for a while now I've been looking at a solution to process the tuya messages.

I must say I am very impressed at the coding techniques. Whoever coded the TuyaSerialInput(); and the TuyaPacketProcess(); my hat is off to you! 

I don't have a full understanding of the whole Tasmota environment so I have one question but before that, I have a couple of comments...

 @thirug010 - also some great work on the analysis and suggestion for improving and incorporating more than just your standard tuya dimmers.
here are some hygienic changes I suggest:

your code:
* if(tuya_buffer[6] == 1 ) // Device -->1 Dimmer Power * { if((power || Settings.light_dimmer > 0) && (power != tuya_buffer[10])) { *ExecuteCommandPower(1*, tuya_buffer[10], SRC_SWITCH); 
if(tyua_buffer[6] ==3) ...etc

no need for if statements. Just pass the tuya_buffer[6] to the ExecuteCommandPower.

so it becomes:
******
case TUYA_CMD_STATE:
      if (tuya_buffer[5] == 5) {  // on/off packet

        AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TYA: RX - %s State"),tuya_buffer[10]?"On":"Off");

        if((power) && (power != tuya_buffer[10])) {
          ExecuteCommandPower(tuya_buffer[6], tuya_buffer[10], SRC_SWITCH);  // send SRC_SWITCH? to use as flag to prevent loop from inbound states from faceplate interaction
        }
      }
*****

Same thing for the TuyaSetPower(); function

instead of:
 if (source != SRC_SWITCH && TuyaSerial) { // ignore to prevent loop from pushing state from faceplate interaction 
if(deviceId == 1) // 3 is dimmer { TuyaSendBool(1, rpower); } 
if(deviceId == 3) // 3 is dimmer { TuyaSendBool(3, rpower); } status = true; } return status; }
...

the function becomes:
*****
bool TuyaSetPower(void)
{
  bool status = false;
  uint8_t deviceId = XdrvMailbox.notused;
  uint8_t rpower = XdrvMailbox.index;
  int16_t source = XdrvMailbox.payload;

  if (source != SRC_SWITCH && TuyaSerial) {  // ignore to prevent loop from pushing state from faceplate interaction

    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TYA: SetDevicePower.rpower=%d, deviceId= %d"), rpower, deviceId);

    TuyaSendBool(deviceId, rpower);

    status = true;
  }
  return status;
}
*****

just a little bit more elegant, that's all.

and now for my question...

Q1. Everything works but I cannot find which parameter to pass for the the switch number (01, 02, 03 or 04)
so the reply needs to be...

                              S#              ON
55aa 00 06 0005 01 01 0001 01 (checksum)
55aa 00 06 0005 02 01 0001 01 (checksum)
55aa 00 06 0005 03 01 0001 01 (checksum)
55aa 00 06 0005 04 01 0001 01 (checksum)

                              S#              OFF
55aa 00 06 0005 01 01 0001 00 (checksum)
55aa 00 06 0005 02 01 0001 00 (checksum)
55aa 00 06 0005 03 01 0001 00 (checksum)
55aa 00 06 0005 04 01 0001 00 (checksum)

So I need to pass it the switch number 01, 02, 03 and o4. Currently whatever toggle I press on the main page the switch number (S#) is 01.

MaliMrav, 2019-07-06:
> I have updated the tasmota to support the custom tuya devices, It can be flashed with USB or OTA with minimal firmware _**But not with Tuya-Convert directly ( it is 516kb bin file)**_
> 
> You can the following options to update the code
> 1 . Use the attached (.zip) files and update your Sonoff-Tasmota-6.5.0 Branch
> 2 . Use cloned the version of the Sonoff-Tasmota-6.5.0 [here](https://github.com/thirug010/Sonoff-Tasmota-6.5.0-New-Tuya-Devices)
> _**Important** this code is not tested on all tuya devices, update it on your own risk, make sure compare the serial data, if you need help please reach out to me_
> ### Summary of Changes:
> 
> **./sonoff**
> _settings.h sonoff.ino xdrv_01_webserver.ino xdrv_16_tuyadimmer.ino_
> **./tools**
> _decode-config.py_
> 
> This code will support up-to 8 Tuya devices as per the TUYA-MCU Protocols. it can be all 8 Relays or 1 dimmer + 6 relays but only one Dimmer can be used at a time.
> 
>     * To Enable or Disable Dimmer control use SetOption63 --> 1-Enable 0-Disable
> 
>     * To Set No of devices / Relays / Dimmer  use SetOption38 (_**Note:** require re-boot to apply the change_)
>       Example 4 Gang SetOption38 - 4
> 
> 
> Here is sample devices and SetOptions .
> 
> **Steps use Tuya Dimmer as Switch (4Gang) ..** _Thanks to @Rambylon for Testing and Support_
> 
>     1. Make sure the Serial Data is matching with your device [tuya_4gang_wallpad_uk.xlsx](https://github.com/arendst/Sonoff-Tasmota/files/3192203/tuya_4gang_wallpad_uk.xlsx)
> 
>     2. Reset Current Configuration --> it will make the device as basic sonoff
> 
>     3. Update the new firmware
> 
>     4. SetOption63 as 0 --> will disable the dimmer slider control
> 
>     5. SetOption38 as 4
> 
>     6. Select Module as Tuya dimmer
>        [image]
> 
> 
> **Steps use Tuya Dimmer as Fan Dimmer and Light Switch (1 Dimmer, 1 Relay)**
> 
>     1. Make sure the Serial Data is matching with your device
>        [Fan Dimmer Data.zip](https://github.com/arendst/Sonoff-Tasmota/files/3192220/Fan.Dimmer.Data.zip)
> 
>     2. Reset Current Configuration --> it will make the device as basic sonoff
> 
>     3. Update the new firmware
> 
>     4. SetOption63 as 1 --> will enable the dimmer slider control
> 
>     5. SetOption38 as 3 --> 1-SSR, 2-Dimmer, 3-Relay
> 
>     6. SetOption34 as 2 --> (Optional it will be set based on the feedback from MCU on the 1st dim packet)
> 
>     7. Select Module as Tuya dimmer
> 
> 
> **Notes:**
> 
>     * Use Power 1 and Dimmer Control to turn ON/OFF and set the Dimmer Value
> 
>     * Use Power 3 for the control Light Switch (relay)
> 
>     * Power 2 is just dimmer control Just sent the dimmer packets. toggle buttn does not work
>       [image]
> 
> 
> **Steps use Tuya Dimmer as Switch (3Gang)** _Not Tested_
> 
>     1. Make sure the Serial Data ( first 3 switch)  is matching with your device [tuya_4gang_wallpad_uk.xlsx] (https://github.com/arendst/Sonoff-Tasmota/files/3192203/tuya_4gang_wallpad_uk.xlsx)
> 
>     2. Reset Current Configuration --> it will make the device as basic sonoff
> 
>     3. Update the new firmware
> 
>     4. SetOption63 as 0 --> will disable the dimmer slider control
> 
>     5. SetOption38 as 3
> 
>     6. Select Module as Tuya dimmer
> 
> 
> [Tuya-Updates.zip](https://github.com/arendst/Sonoff-Tasmota/files/3193413/Tuya-Updates.zip)
> Cloned Repo: https://github.com/thirug010/Sonoff-Tasmota-6.5.0-New-Tuya-Devices

Looks like you've already done that. That's great!
Ok one more thing...

This is the status update from a 4 Gang Tuya switch panel with "Cooperative Processing by the MCU and Wi-Fi Module". i.e.
[image]

the MCU sends this status for a 4 gang (no dimmer)

55AA 01 07 0005 01 01 0001 00 0F
55AA 01 07 0005 02 01 0001 00 10
55AA 01 07 0005 03 01 0001 00 11
55AA 01 07 0005 04 01 0001 00 12
55AA 01 07 0008 07 02 0004 00000000 1C
55AA 01 07 0008 08 02 0004 00000000 1D
55AA 01 07 0008 09 02 0004 00000000 1E
55AA 01 07 0008 0A 02 0004 00000000 1F
55AA 01 07 0008 65 02 0004 00000000 7A
55AA 01 07 0008 66 02 0004 00000000 7B
55AA 01 07 0008 67 02 0004 00000000 7C
55AA 01 07 0008 68 02 0004 00000961 E7

so I'm not user what 07, 08, 09 and 0A are. I think they have something to do with dimmer (not implemented in this switch

but 65, 66, 67 and 68 are nothing to do with the dimmer and the code is misinterpreting it as a dimmer packet because the code length (0x00 0x08) in byte 5.

I then discovered that it looks like one of them is the voltage and the other is the current

55AA 01 07 0008 65020004000000007A
55AA 01 07 0008 66020004000000E560
55AA 01 07 0008 6702000400000241BF
55AA 01 07 0008 680200040000096EF4

and it looks like the two last bytes are a word, i.e.:

word(tuya_buffer[12],tuya_buffer[13]);

So how hard is it to a) stop misinterpreting those as dimmer settings and b) perhaps process them as voltage (V) and current (mA)?

Rambylon, 2019-07-06:
@MaliMrav i dont think analogue operation(process them as voltage (V) and current (mA)) works when the tuya processes serial data. i guess its a software and hardware limitation.

MaliMrav, 2019-07-07:
[image]
The tuya switches I have have a sensor that is read by the MCU and passed on to the tuya Smartlife app, via the last two messages 67 and 68.. as per pic
Ive managed to get the Tasmota to display it via Addlog to confirm.
Hence, my issue is that when the code reads the last 4 lines it interprets them as dimmer values and plays havoc with the switches, randomly switching them off as they read a 0 or a duty less than 25%, I assume.

Anyhow, if disable the dimmer functions, the device behaves as it should.

thirug010, 2019-07-07:
@MaliMrav,

Send me the packets for the 67 and 68 will update the code, and send me the link for the device / switch has power information 

Thanks 
Thiru

thirug010, 2019-07-07:
@MaliMrav, Thanks for your finding, I will use the serial data to update the code, Please share the details about the switch, I would try get and try it myself

thanks 
thiru

thirug010, 2019-07-07:
@MaliMrav , Random switching due to the unknown packets on the non-dimmer device was fixed already in the dev branch, Please take the latest dev branch code.

MikeSaint, 2019-07-08:
@MaliMrav Great information on the power usage and the diagram on "Cooperative Processing by the MCU and Wi-Fi Module".

I have flashed 3 devices to the dev branch Tasmota and it's working fine as a serial switch (apart from SetOption38 to 4 makes them reboot but not retain the setting). But that's not the end of the world.

Recovering from a failed install / bad setting / user error is my problem of the week.

Mine are all TYWE3S based Tuya 3 and 4 gang serial switches.

Has anyone worked out how to USB flash one of these serial 4 gang switches if the OTA flash fails and it goes into a mode where it doesn't connect to WiFi or create it's own SSID for management (aka sort of bricked)?

3 devices worked fine and are fully functional. I've got one that was running Tasmota basic and flashed to the new Dev branch of full Tasmota but failed once it was set to be a Tuya Dimmer and rebooted. The other failed because I stupidly put some version of ESPEasy firmware on it that didn't have OTA upgrade capability or my WiFi settings (my bad).

Being serial, the buttons don't respond as GPIO-0 would on a Sonoff device. I've tried loads of combinations that would work 100% on a Sonoff device using ESPEasy on a USB to TTL serial stick, but these serial switches don't seem to behave in the same way where we'd simply ground GPIO-0 to put it into flash mode.

Grounding GPIO-0 on these units during startup doesn't seem to do anything at boot. I'm pretty sure that is my issue.

Has anyone worked out how to flash one of these? Could you please share the process and exactly which TYWE3S pins were used?

[image]

MaliMrav, 2019-07-08:
Yes, I've worked it out

The reason why it fails with usb is because the MCU sends data every 5 seconds on the same line as your USB serial (RXD0). So to successfully update your firrmware via USB you need to cut the RXD0 track, write your firmware and then re-solder the track back on

MaliMrav, 2019-07-08:
> @MaliMrav, Thanks for your finding, I will use the serial data to update the code, Please share the details about the switch, I would try get and try it myself
> 
> thanks
> thiru

Hi Thiru,

Sure, not a problem. Will do that tonight. It's from a company called makegood. I'll up[load photos

MaliMrav, 2019-07-08:
> @MaliMrav , Random switching due to the unknown packets on the non-dimmer device was fixed already in the dev branch, Please take the latest dev branch code.

In the dev branch the SetOption63 --> 0 and SetOption38 --> 4 doesn't seem to work.
So the options to set the switches and remove the dim is only in the tuya-new, right?

Do I now need to hard-code the switch count?

arendst, 2019-07-08:
SetOption38 is wrong and is deleted on latest dev branch. Use SetOption41 instead as documented here #6039

meingraham, 2019-07-08:
@arendst is this functionality no longer supported then?

`set IRReceive protocol detection sensitivity minimizing UNKNOWN protocols`

arendst, 2019-07-08:
@Mike, it is still supported as documented. The problem was both IRReceive and Tuya used the same storage location. That has been changed now as IRReceive continues to use SetOption38 and Tuya uses SetOption41.

meingraham, 2019-07-08:
OK. So you were referring to the improper use of SetOption38 in the Tuya PR. SetOption38 as documented for IR remains. Check. All's well.

MaliMrav, 2019-07-10:
> @MaliMrav, Thanks for your finding, I will use the serial data to update the code, Please share the details about the switch, I would try get and try it myself
> 
> thanks
> thiru

Hi Thiru,

So the switches can be found at http://www.szmakegood.com/

the pics of the front and back of the 3 and 4 Gang Switches
[image]
[image]
[image]
[image]

MaliMrav, 2019-07-10:
> SetOption38 is wrong and is deleted on latest dev branch. Use SetOption41 instead as documented here #6039

ok so the switch count is fixed; Great!

What about removing the dimmer slider, SetOption63 --> 0 ?

Further more, I still have the status lines interpreted as dimmer settings. My workaround is pretty much a hack to disable the reading of what the function incorrectly assumes it to be a dim value

 else if (tuya_buffer[5] == 08 {  // dim packet <--- this is a wrong assumption because there are other use cases where the 6th byte is 08, not just dimmer values.
in my case there are 8 lines that are status values: (Voltage, power, and some I have not worked out yet.)
(see my previous post for more details.

so I've just disabled it by looking for a different value, for now:
 else if (tuya_buffer[5] == 0) {  //from 8 to 0

Regards,
MaliMrav

tyjtyj, 2019-07-21:
I need some help. I connected the serial at 9600 baudrate. I am getting U▒U▒U▒U▒U▒sOGcriojabRHDvFH1.0.0▒U▒U▒U▒U...  what is the correct baud rate for TYEW3S ?

christianzappa, 2019-08-08:
> I have updated the tasmota to support the custom tuya devices, It can be flashed with USB or OTA with minimal firmware _**But not with Tuya-Convert directly ( it is 516kb bin file)**_
>
> You can the [...]

How do youupload the ZIP file OTA using the Tasmota WebGUI?

thirug010, 2019-08-08:
@christianzappa,
Please upgrade to latest firmware it has the changes now 

thanks 
Thiru

christianzappa, 2019-08-08:
> @christianzappa,
> Please upgrade to latest firmware it has the changes now
> 
> thanks
> Thiru

Thanks @thirug010 
OK, half working, what setoptions do I need to configure, also should I use Tuya dimmer for the module and template? 2 and 3 gang switch here

thirug010, 2019-08-08:
@christianzappa , Please share screenshot of the Tasmota web ui, 

thanks 
thiru

christianzappa, 2019-08-08:
@thirug010 sure. I've just got a WiFi issue atm. After resetting Tasmota and inputting my SSID and Pass, the unit isn't connecting to my wifi or generating an access point.

MikeSaint, 2019-08-08:
On my units I had to fully power them off for the wifi to connect once the SSID and password were set. Not just reboot, remove power completely.

SetOption 38 is not used for Tuya switches. I wonder if it’s possible to edit that post and change it as it confusing when people think its fact. It’s SetOption41, but on my units running the latest production release of Tasmota it doesn’t change the number of switches but I don’t care as it works fine and I just ignore switches 5 & 6.

Regards
Mike

+6421349514
mike@saint.co.nz<mailto:mike@saint.co.nz>

meingraham, 2019-08-08:
@MikeSaint - `SetOption38`? That's an IR related option.

thirug010, 2019-08-09:
@MikeSaint,

This post was old it had the old setoptions correct option is updated it is SetOption41 is the correct option

Please use the latest dev branch code to avoid the 6 switch issue 

Thanks 
Thiru

christianzappa, 2019-08-09:
Tried all the reset button tricks, and completely removed it, not just switching the lighting circuit off at the breaker, same problem, not generating an AP or connecting to the one I programmed. The Wi-Fi LED just blinks away with no result...

MaliMrav, 2019-08-09:
> How do youupload the ZIP file OTA using the Tasmota WebGUI?

You can actually do it with Tuya-convert. The only
Thing is it’s a three stage process:
1) do the tuya-convert
2) load (via Web GUI) the minimal version of Tasmota
3) load this firmware

Tobias007, 2019-11-05:
I flashed the Sonoff.bin - set the Tuya MCU device - however I get a endless loop in the console.

Seems to be a restart of MCU detected - but why?

12:35:58 TYA: Heartbeat
12:36:04 WIF: Checking connection...
12:36:04 WIF: Connected
12:36:09 TYA: Send "55aa00000000ff"
12:36:09 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA000000010000"}
12:36:09 TYA: Heartbeat
12:36:09 TYA: Detected MCU restart
12:36:10 TYA: Set WiFi LED 3 (0)
12:36:10 TYA: Send "55aa000300010306"
12:36:10 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA0003000002"}
12:36:10 TYA: RX WiFi LED set ACK
12:36:20 TYA: Send "55aa00000000ff"
12:36:20 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA000000010000"}
12:36:20 TYA: Heartbeat
12:36:20 TYA: Detected MCU restart
12:36:21 TYA: Set WiFi LED 3 (0)
12:36:21 TYA: Send "55aa000300010306"
12:36:21 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA0003000002"}
12:36:21 TYA: RX WiFi LED set ACK
12:36:24 WIF: Checking connection...
12:36:24 WIF: Connected
12:36:31 TYA: Send "55aa00000000ff"
12:36:31 RSL: tele/sonoff/RESULT = {"TuyaMcuReceived":"55AA000000010101"}
12:36:31 TYA: Heartbeat

[6 more replies]