arendst/Tasmota #5546: Is there Any timer to calculate relay Uptime ?

hwarang098tkd, 2019-03-29:
I was wondering if there is inner timer for relays(power). I want to sen to my openhab the min that a relay is on or off!!! I know that i can do it with code in openhab but is safer to do this work the client.

Jason2866, 2019-03-29:
No, maybe it can be done via rules.

hwarang098tkd, 2019-03-29:
Any suggestion?

Jason2866, 2019-03-29:
No, not really. Never thought about this...
You find examples for rules here https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

ascillato2, 2019-03-30:
Hi,

Sorry, there is not an easy way to do that without modifying the code. 

Anyway, that uptime is meant to be managed in your home automation software. 

Adding label feature request (hold over) for future reference. 

Thanks for sharing your ideas.

closed 2019-03-30 by ascillato2