arendst/Tasmota #5551: DS18x20 sensors gone after upgrade to 6.5.0.3

pascalsaul, 2019-03-29:
### ISSUE DESCRIPTION - TROUBLESHOOTING
DS18B20 sensor(s) gone after upgrade to 6.5
  
### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff TH
- [X] Tasmota binary firmware version number used : 6.4.1.19 / self-compiled
- [X] Development IDE - Compiler / Upload tools used : PlatformIO / Webinterface
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
``
th_dbe01/cmnd/status 0
th_dbe01/stat/STATUS {"Status":{"Module":4,"FriendlyName":["th_dbe01"],"Topic":"th_dbe01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
th_dbe01/stat/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.12/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:58","StartupUTC":"2019-03-29T20:10:34","Sleep":100,"CfgHolder":4617,"BootCount":183,"SaveCount":2454,"SaveAddress":"FA000"}}
th_dbe01/stat/STATUS2 {"StatusFWR":{"Version":"6.5.0.3(sonoff)","BuildDateTime":"2019-03-29T20:40:35","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
th_dbe01/stat/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"192.168.1.17","LogPort":514,"SSId":["XXX","XXX"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A0009","280500000100000000000000000000000000","00000201"]}}
th_dbe01/stat/STATUS4 {"StatusMEM":{"ProgramSize":443,"Free":560,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","06482384","040003A1","21003CC6","000000C0"]}}
th_dbe01/stat/STATUS5 {"StatusNET":{"Hostname":"th_dbe01","IPAddress":"192.168.1.71","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:9E:C6:5E","Webserver":2,"WifiConfig":4}}
th_dbe01/stat/STATUS6 {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"th_dbe01","MqttClient":"th_dbe01","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
th_dbe01/stat/STATUS7 {"StatusTIM":{"UTC":"Fri Mar 29 20:17:32 2019","Local":"Fri Mar 29 21:17:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
th_dbe01/stat/STATUS10 {"StatusSNS":{"Time":"2019-03-29T21:17:32"}}
th_dbe01/stat/STATUS11 {"StatusSTS":{"Time":"2019-03-29T21:17:32","Uptime":"0T00:06:58","Vcc":3.498,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"POWER":"0","Wifi":{"AP":1,"SSId":"Peppelenwei 65","BSSId":"A4:18:75:51:51:11","Channel":1,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:06"}}}
``
```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
Nothing to see...

CONSOLE OUTPUT HERE:
Before update:
```
th_dbe01/tele/LWT Online
homeassistant/switch/9EC65E_RL_1/config {"name":"th_dbe01","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"0","pl_on":"1","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9EC65E_RL_1","device":{"identifiers":["9EC65E"]}, "~":"th_dbe01/"}
homeassistant/sensor/9EC65E_DS18B20_Temperature/config {"name":"th_dbe01 DS18B20 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9EC65E_DS18B20_Temperature","device":{"identifiers":["9EC65E"]}, "~":"th_dbe01/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20'].Temperature}}"}
homeassistant/sensor/9EC65E_status/config {"name":"th_dbe01 status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"9EC65E_status","device":{"identifiers":["9EC65E"],"name":"th_dbe01","model":"Sonoff TH","sw_version":"6.4.1.19(sonoff)","manufacturer":"Tasmota"}, "~":"th_dbe01/tele/"}
th_dbe01/tele/LWT Online
th_dbe01/cmnd/POWER (null)
th_dbe01/tele/HASS_STATE {"Version":"6.4.1.19(sonoff)","Module":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:08:24","BootCount":182,"SaveCount":2452,"IPAddress":"192.168.1.71","RSSI":"94","LoadAvg":6630}
```
After update:
```
th_dbe01/tele/LWT Online
th_dbe01/cmnd/POWER (null)
th_dbe01/tele/INFO1 {"Module":"Sonoff TH","Version":"6.5.0.3(sonoff)","FallbackTopic":"cmnd/th_dbe01_fb/","GroupTopic":"sonoffs"}
th_dbe01/tele/INFO2 {"WebServerMode":"Admin","Hostname":"th_dbe01","IPAddress":"192.168.1.71"}
th_dbe01/tele/INFO3 {"RestartReason":"Software/System restart"}
th_dbe01/stat/RESULT {"POWER":"0"}
th_dbe01/stat/POWER 0
homeassistant/switch/9EC65E_RL_1/config {"name":"th_dbe01","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"0","pl_on":"1","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9EC65E_RL_1","device":{"identifiers":["9EC65E"]},"~":"th_dbe01/"}
homeassistant/sensor/9EC65E_status/config {"name":"th_dbe01 status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"9EC65E_status","device":{"identifiers":["9EC65E"],"name":"th_dbe01","model":"Sonoff TH","sw_version":"6.5.0.3(sonoff)","manufacturer":"Tasmota"},"~":"th_dbe01/tele/"}
th_dbe01/tele/STATE {"Time":"2019-03-29T21:10:50","Uptime":"0T00:00:16","Vcc":3.510,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"POWER":"0","Wifi":{"AP":1,"SSId":"Peppelenwei 65","BSSId":"A4:18:75:51:51:11","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}
th_dbe01/tele/HASS_STATE {"Version":"6.5.0.3(sonoff)","BuildDateTime":"2019-03-29T20:40:35","Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)","Module":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:00:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":183,"SaveCount":2454,"IPAddress":"192.168.1.71","RSSI":"98","LoadAvg":12}
```

blakadder, 2019-03-29:
How did you upgrade to 6.5? OTA upgrade or precompiled .bin? 
Gone as in not visible in webui?

pascalsaul, 2019-03-29:
> How did you upgrade to 6.5? OTA upgrade or precompiled .bin?
> Gone as in not visible in webui?

As you've read the post it is a precompiled bin ;) 

Gone in WebUI and MQTT.

ascillato, 2019-03-30:
Please, try **sonoff-sensors.bin** Thanks

blakadder, 2019-03-30:
Your post says self compiled which created the confusion. 
Did the sensors.bin resolve the issue?

arendst, 2019-03-30:
There is gone and there is gone...

You experience missing DS18B20 in the Hass discovery message right after a restart. Does it also not appear at teleperiod time?

Cooper-Dale, 2019-03-30:
Hello, I have the same issue. Afrer upgrade to self compiled 6.5.1 my DS18B20 is gone (WEB and also MQTT = sensor is not detected). I have also tried precompiled 6.5.0 sonoff-sensors.bin and still nothing.

arendst, 2019-03-30:
Mine still work fine.

I suppose you did try a power cycle first before commenting...

Cooper-Dale, 2019-03-30:
I tried power cycle many times.

arendst, 2019-03-30:
In that case check your DS18B20 power circuit as the latest firmware might draw more power form your sonoff basic resulting in not enough power for the sensor. Check your DS18B20 resistor

Cooper-Dale, 2019-03-30:
Mine is standard 4K7 one
https://imgur.com/YVZ3dRu

arendst, 2019-03-30:
Mine too and works on a wemos. Haven't tried the basic though but that used to work...

arendst, 2019-03-30:
Still working fine on TH:
```
14:52:40 MQT: tele/th1/LWT = Online (retained)
14:52:40 MQT: cmnd/th1/POWER = 
14:52:40 MQT: tele/th1/INFO1 = {"Module":"Sonoff TH","Version":"6.5.0.3(sonoff)","FallbackTopic":"cmnd/DVES_07C8FF_fb/","GroupTopic":"sonoffs"}
14:52:40 MQT: tele/th1/INFO2 = {"WebServerMode":"Admin","Hostname":"th1","IPAddress":"192.168.2.174"}
14:52:40 MQT: tele/th1/INFO3 = {"RestartReason":"Software/System restart"}
14:52:40 MQT: stat/th1/POWER1 = {"POWER1":"ON"}
14:52:40 MQT: stat/th1/POWER1 = ON
14:52:41 MQT: domoticz/in = {"idx":172,"nvalue":1,"svalue":"","Battery":83,"RSSI":10}
14:52:48 MQT: tele/th1/STATE = {"Time":"2019-03-30T14:52:48","Uptime":"0T00:00:17","Vcc":3.429,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":49,"POWER1":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}
14:52:48 MQT: domoticz/in = {"idx":207,"nvalue":0,"svalue":"21.56","Battery":82,"RSSI":10}
14:52:48 MQT: tele/th1/SENSOR = {"Time":"2019-03-30T14:52:48","DS18B20":{"Temperature":21.56},"TempUnit":"C"}
```

Cooper-Dale, 2019-03-30:
I tried remove sensor ant test it on generic ESP (own compiled 6.4.1.15) and it is working, soledered back to Sonoff and nothing.
I also tried to downgrade to 6.4.1 - sensors and also did not hepl.

Cooper-Dale, 2019-03-30:
What a big coincidence, i resoldered sensor and header and now it is working. But bad timinig, Sonoff + sensor was working for more than year, without touch and header lost its connenction just in time of upgrade. 
So for me it is closed now. Thank you

closed 2019-03-30 by ascillato2

pascalsaul, 2019-03-30:
With the default sonoff-sensor.bin it does work but that is 6.5.0 compiled with 2.3.0. I need to test my own compile with 2.3.0 and see what happens then since I used 2.5.0.

ascillato, 2019-03-30:
Precompiled bins with core 2.5.0:

http://thehackbox.org/tasmota/020500/

pascalsaul, 2019-03-30:
> Precompiled bins with core 2.5.0:
> 
> http://thehackbox.org/tasmota/020500/

Doesn't through OTA, too big ;)

``
th_dbe01/tele/STATE {"Time":"2019-03-30T23:04:35","Uptime":"0T00:00:16","Vcc":3.219,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"POWER":"0","Wifi":{"AP":1,"SSId":"Peppelenwei 65","BSSId":"A4:18:75:51:51:11","Channel":1,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:06"}}
th_dbe01/tele/SENSOR {"Time":"2019-03-30T23:04:35","DS18B20":{"Temperature":18.9},"TempUnit":"C"}
th_dbe01/tele/HASS_STATE {"Version":"6.5.0.3(sonoff)","BuildDateTime":"2019-03-30T21:38:30","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T00:00:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":193,"SaveCount":2481,"IPAddress":"192.168.1.71","RSSI":"96","LoadAvg":9}
``

Compiled with 2_3_0 and it does work...

ascillato, 2019-03-30:
You have to use minimal first and then the full version

pascalsaul, 2019-04-01:
Even then it doesn't work...

Program Version | 6.5.0.3(a380eb9-sensors)
2019-04-01T17:04:38
2_5_0/3.0.0-dev(c0f7b44)

pascalsaul, 2019-04-06:
``Program Version		6.5.0.5(sonoff)
Build Date & Time	2019-04-06T18:46:06
Core/SDK Version	2_5_0/3.0.0-dev(c0f7b44)
``
This one does work on all my devices and sensors now...