arendst/Tasmota #5552: Ifan02 and buttons usage

calcipher49, 2019-03-29:
I will to use button (monostable) on Sonoff Ifan02 to toggle fan speed.
But all 4 buttons (key1 to 4) are mapped on template  with "Virtual button X as feedback from RC".
Seems RC means Remote Command.
Is this mandatory or canIe modify it to use button and also keep Remote usage?

Thanks for clarification (I know can use switch as workaround but I prefer button if possible).

blakadder, 2019-03-29:
You cannot set the button to do two things at once. Why don't you want to use switch instead of button?

calcipher49, 2019-03-29:
blakadder-> Button is more convenient for this usage. Also I don't understand why buttons are reserved in template for Ifan02...

blakadder, 2019-03-30:
If you really need a button and not a switch then create your own template.

ascillato2, 2019-03-30:
Closing this issue as it has been answered.

Please, if you need further assistant, please ask here or in the Tasmota support chat. 

Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-30 by ascillato2