arendst/Tasmota #5554: Multiple reboot of the module when using the PZEM-004 sensor

stepanovalex, 2019-03-30:
When the PZEM-004 sensor is connected, the module repeatedly reboots (exeption, watchdog).

Module operation time from several seconds to several hours. After that - a new reboot cycle.

The number of reboots is extremely large (hundreds of times more than other modules).

Does not depend on the firmware version (now on the last current one)

Info from Information page:

Версия программы | 6.5.0(release-sonoff)
-- | --
Дата & Время сборки | 2019-03-19T12:40:37
Версия Core/SDK | 2_3_0/1.5.3(aec24ac9)
Uptime | 0T00:21:59
Кол-во записей Flash | 5902 at 0xF8000
Количество загрузок | **5625**
Причина перезагрузки | **Exception**

I have three such devices. Everywhere the situation is similar.

My friend has the same problem.

arendst, 2019-03-30:
Solutions:
- Only use hardware serial as core 2.3.0 does not support software serial with interrupts
- Use core 2.4.2 or higher

stepanovalex, 2019-03-30:
How?

arendst, 2019-03-30:
See release notes, readme and wiki

stepanovalex, 2019-03-30:
Thank you! 

I flashed my module with new kernel - I'll tell you the results later.

closed 2019-03-30 by ascillato2