arendst/Tasmota #5555: Reboot on IRsend command with invalid input

alessio-papi, 2019-03-30:
### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

Hi,

trying to send an invalid IR code using the console I'm getting a device reboot with the following exception:

18:05:01 MQT: tele/remote_transmitter/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40239bc6 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}  

the command was:

IRsend {}

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (i.e. Sonoff Basic) : NodeMCU (ESP8266)
- [ ] Tasmota binary firmware version number used : 6.5.0 (Core/SDK 2_4_2/2.2.1) / pre-compiled 
- [ X] Development IDE - Compiler / Upload tools used : Arduino IDE 1.8.8 / esptool.py 2.5.1
- [ ]X Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

18:15:13 CMD: status 0
18:15:13 MQT: stat/remote_transmitter/STATUS = {"Status":{"Module":18,"FriendlyName":["Remote transmitter"],"Topic":"remote_transmitter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:15:13 MQT: stat/remote_transmitter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-IT.bin","RestartReason":"Exception","Uptime":"0T00:04:36","StartupUTC":"2019-03-30T17:10:37","Sleep":50,"CfgHolder":4617,"BootCount":30,"SaveCount":43,"SaveAddress":"F9000"}}
18:15:13 MQT: stat/remote_transmitter/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T13:12:24","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:15:13 MQT: stat/remote_transmitter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Attic","Home"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
18:15:13 MQT: stat/remote_transmitter/STATUS4 = {"StatusMEM":{"ProgramSize":525,"Free":476,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000410","0FDAE394","000783A0","23B617CE","00003BC0"]}}
18:15:13 MQT: stat/remote_transmitter/STATUS5 = {"StatusNET":{"Hostname":"remote_transmitter-1506","IPAddress":"192.168.0.XXX","Gateway":"192.168.0.XXX","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.XXX","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
18:15:13 MQT: stat/remote_transmitter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.XXX","MqttPort":XXXX,"MqttClientMask":"XXXXX","MqttClient":"XXXX","MqttUser":"XXXX","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:15:13 MQT: stat/remote_transmitter/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 30 17:15:13 2019","Local":"Sat Mar 30 18:15:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:32","Sunset":"19:17"}}
18:15:13 MQT: stat/remote_transmitter/STATUS10 = {"StatusSNS":{"Time":"2019-03-30T18:15:13"}}
18:15:13 MQT: stat/remote_transmitter/STATUS11 = {"StatusSTS":{"Time":"2019-03-30T18:15:13","Uptime":"0T00:04:36","Vcc":2.997,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Attic","BSSId":"XX:XX:XX:XX:XX:XX","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

No error in console, only excpetion on reboot

18:22:12 MQT: tele/remote_transmitter/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40239bc6 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-30:
Can confirm sending wrong! command string `irsend {}` results in a exception

arendst, 2019-03-30:
Then do not send invalid JSON data.

I will implement some small check but ArduinoJson should have handled it. It doesn't hence my template workaround ;-)

alessio-papi, 2019-03-31:
Sure, avoid the invalid JSON :-) 
Only for reference: I ran into the problem due to a Home Assistant misconfigured command topic.

curzon01, 2019-03-31:
unfortunately {} is a valid json :-) (see [ECMA-404](http://www.ecma-international.org/publications/files/ECMA-ST/ECMA-404.pdf))

arendst, 2019-03-31:
But ArduinoJson doesn't like it. In fact it hates all permutations of {\<spaces\>}

Jason2866, 2019-03-31:
Hmm, sending NEC code does work `IRSend {"PROTOCOL":"NEC","BITS":32,"DATA":"551541285"}`
```
13:38:06 CMD: irsend {"PROTOCOL":"NEC","BITS":32,"DATA":"551541285"}
13:38:06 MQT: stat/sonoff-17DBAE/RESULT = {"IRSend":"Done"}
```
If i try `irsend {"PROTOCOL":"RC6","BITS":36,"DATA":"551541285"}` i get a 'Exception
```
00:00:00 Project sonoff WeMini 192-168-2-150 Version 6.5.0.3(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-17DBAE-7086...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-17DBAE-7086 with IP address 192.168.2.150
13:39:58 MQT: Attempting connection...
13:39:58 MQT: Connected
13:39:58 MQT: tele/sonoff-17DBAE/LWT = Online (retained)
13:39:58 MQT: cmnd/sonoff-17DBAE/POWER = 
13:39:58 MQT: tele/sonoff-17DBAE/INFO1 = {"Module":"Generic","Version":"6.5.0.3(sonoff)","FallbackTopic":"cmnd/sonoff-17DBAE_fb/","GroupTopic":"sonoffs"}
13:39:58 MQT: tele/sonoff-17DBAE/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-17DBAE-7086","IPAddress":"192.168.2.150"}
13:39:58 MQT: tele/sonoff-17DBAE/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40252a00 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
13:40:06 MQT: tele/sonoff-17DBAE/STATE = {"Time":"2019-03-31T13:40:06","Uptime":"0T00:00:15","Vcc":3.088,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:1E:69:50","Channel":4,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:40:06 MQT: tele/sonoff-17DBAE/SENSOR = {"Time":"2019-03-31T13:40:06","AM2301":{"Temperature":23.7,"Humidity":29.8},"BME280":{"Temperature":27.1,"Humidity":25.0,"Pressure":978.7,"SeaPressure":1022.6},"PressureUnit":"hPa","TempUnit":"C"}
```
Status 0
```
00:00:00 Project sonoff WeMini 192-168-2-150 Version 6.5.0.3(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-17DBAE-7086...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-17DBAE-7086 with IP address 192.168.2.150
13:39:58 MQT: Attempting connection...
13:39:58 MQT: Connected
13:39:58 MQT: tele/sonoff-17DBAE/LWT = Online (retained)
13:39:58 MQT: cmnd/sonoff-17DBAE/POWER = 
13:39:58 MQT: tele/sonoff-17DBAE/INFO1 = {"Module":"Generic","Version":"6.5.0.3(sonoff)","FallbackTopic":"cmnd/sonoff-17DBAE_fb/","GroupTopic":"sonoffs"}
13:39:58 MQT: tele/sonoff-17DBAE/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-17DBAE-7086","IPAddress":"192.168.2.150"}
13:39:58 MQT: tele/sonoff-17DBAE/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40252a00 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
13:40:06 MQT: tele/sonoff-17DBAE/STATE = {"Time":"2019-03-31T13:40:06","Uptime":"0T00:00:15","Vcc":3.088,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:1E:69:50","Channel":4,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:40:06 MQT: tele/sonoff-17DBAE/SENSOR = {"Time":"2019-03-31T13:40:06","AM2301":{"Temperature":23.7,"Humidity":29.8},"BME280":{"Temperature":27.1,"Humidity":25.0,"Pressure":978.7,"SeaPressure":1022.6},"PressureUnit":"hPa","TempUnit":"C"}
```

arendst, 2019-03-31:
Exception 0 mostly point to a power problem. How do you power your IR led(s)?

arendst, 2019-03-31:
I see no exception on YTF:
```
13:43:45 CMD: irsend {"PROTOCOL":"RC6","BITS":36,"DATA":"551541285"}
13:43:45 SRC: WebConsole from 192.168.2.1
13:43:45 RSL: Group 0, Index 1, Command IRSEND, Data {"PROTOCOL":"RC6","BITS":36,"DATA":"551541285"}
13:43:45 IRS: protocol_text RC6, protocol RC6, bits 36, data 551541285 (0x20DFDA25), protocol_code 2
13:43:46 MQT: stat/bridgeir/RESULT = {"IRSend":"Done"}
```

Jason2866, 2019-03-31:
With a 9v extra power supply (old modified TV remote...) Maybe the power supply of used wemos mini is bad. Will replace the 3.3V reg on wemos. Thx for hint!

ascillato2, 2019-03-31:
Closing this issue as it is solved. Thanks for reporting and Theo for fixing it so fast :+1:

closed 2019-03-31 by ascillato2