arendst/Tasmota #5558: Teckin SP22 restarts and loses configuration every time

marcor78, 2019-03-30:
I have flashed several Teckin SP22 smart plugs OTA with Tuya-convert. Some of them working only if configured as Blitzwolf (45), some of them only if configured as Teckin (52), but that's ok. Just one of them working both as Teckin (52) or Gosund Sp1 (55), but it restarts often and as soon as it restarts and connects to wifi again, it just loses the configuration, reverting to Sonoff Basic (1) therefore stops working completely until I configure module again to Gosund or Teckin. It may last few hours at most, sometimes it resets after minutes. Completely unuseful right now..
the device is flashed with 6.5.0 firmware, and I tried both core 2.3.0 than 2.4.2 with no result if not 2.4.2 even worse about this issue (making the smart plug restarting even more often)

ascillato2, 2019-03-30:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

marcor78, 2019-03-30:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Teckin SP22 flashed tasmota 6.5.0. core 2.3.0. restarts frequently and loses configuration becoming unusable
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Teckin SP22
- [X] Tasmota binary firmware version number used : 6.5.0 (Core/SDK 2_3_0/1.5.3) / pre-compiled
- [X] Development IDE - Compiler / Upload tools used : Tuya-Convert OTA on Raspberry Pi3 B+
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
00:50:20 CMD: status 0
00:50:20 MQT: stat/teckin_06/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"teckin_06","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:50:20 MQT: stat/teckin_06/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:14:02","StartupUTC":"2019-03-30T23:36:18","Sleep":50,"CfgHolder":4617,"BootCount":54,"SaveCount":273,"SaveAddress":"FB000"}}
00:50:20 MQT: stat/teckin_06/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:50:20 MQT: stat/teckin_06/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNCAP336AA3",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","280500000100000000000000000000000000","00000000"]}}
00:50:21 MQT: stat/teckin_06/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
00:50:21 MQT: stat/teckin_06/STATUS5 = {"StatusNET":{"Hostname":"teckin_06-6804","IPAddress":"192.168.1.76","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"DC:4F:22:ED:BA:94","Webserver":2,"WifiConfig":5}}
00:50:21 MQT: stat/teckin_06/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.92","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EDBA94","MqttUser":"marcomqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:50:21 MQT: stat/teckin_06/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 30 23:50:21 2019","Local":"Sun Mar 31 00:50:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:30","Sunset":"19:18"}}
00:50:21 MQT: stat/teckin_06/STATUS10 = {"StatusSNS":{"Time":"2019-03-31T00:50:21"}}
00:50:21 MQT: stat/teckin_06/STATUS11 = {"StatusSTS":{"Time":"2019-03-31T00:50:21","Uptime":"0T00:14:03","Vcc":3.157,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"TNCAP336AA3","BSSId":"A4:B1:E9:33:6A:A3","Channel":13,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:33"}}}

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

I experienced the issue again right in this moment. After about 3 minutes from the log seen above, device restarted on its own, then console restarted like this:

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 TNCAP336AA3 in mode 11N as teckin_06-6804...
00:00:10 WIF: Connected
00:00:10 HTP: Web server active on teckin_06-6804 with IP address 192.168.1.76
00:00:11 MQT: Attempting connection...
00:53:23 MQT: Connected
00:53:23 MQT: tele/teckin_06/LWT = Online (retained)
00:53:23 MQT: cmnd/teckin_06/POWER = 
00:53:23 MQT: tele/teckin_06/INFO1 = {"Module":"Sonoff Basic","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/DVES_EDBA94_fb/","GroupTopic":"sonoffs"}
00:53:23 MQT: tele/teckin_06/INFO2 = {"WebServerMode":"Admin","Hostname":"teckin_06-6804","IPAddress":"192.168.1.76"}
00:53:24 MQT: tele/teckin_06/INFO3 = {"RestartReason":"Power on"}
00:53:24 MQT: stat/teckin_06/RESULT = {"POWER":"ON"}
00:53:24 MQT: stat/teckin_06/POWER = ON (retained)
00:53:27 MQT: stat/teckin_06/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"teckin_06","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:53:27 MQT: stat/teckin_06/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:00:16","StartupUTC":"2019-03-30T23:53:11","Sleep":50,"CfgHolder":4617,"BootCount":56,"SaveCount":277,"SaveAddress":"F7000" [... cut ...]
00:53:27 MQT: stat/teckin_06/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 30 23:53:27 2019","Local":"Sun Mar 31 00:53:27 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:30","Sunset":"19:18"}}
00:53:27 MQT: stat/teckin_06/STATUS10 = {"StatusSNS":{"Time":"2019-03-31T00:53:27"}}
00:53:27 MQT: stat/teckin_06/STATUS11 = {"StatusSTS":{"Time":"2019-03-31T00:53:27","Uptime":"0T00:00:16","Vcc":3.156,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":252,"POWER":"ON","Wifi":{"AP":1,"SSId":"TNCAP336AA3","BSSId":"A4:B1:E9:33:6A:A3","Channel":13,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:10"}}}

```

ascillato, 2019-03-31:
Thanks a lot for completing the template. It gave us a lot of information.

You said that you have configured your module as Blitzwolf (45), Teckin (52) and Gosund Sp1 (55). Those 3 have the button configured in different GPIOs, so you should use the correct one only, because using the incorrect GPIO for button1 could trigger undesired behaviours like turning on/off your relay and also can trigger the config reset. Tasmota has a feature that if you push the button for more than 40 seconds, it will reset all the config to default. This can be disabled using commands setoption1 and setoption13 (more info in the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption-overview)

Another thing is that you said is that the log is just after the failure. The log said restart reason POWER ON, that means that your device was left without power and was powered again, not a software reset. So, if you didn't disconnect it and you didn't have a power loss, that could be that the power regulator inside the device is faulty (hardware issue). Your boot count: 56, is high, so may be is your issue if you have flashed the device recently.

Another feature that Tasmota has is a boot loop protection. If any config makes Tasmota to reset after a couple of seconds of being powered, and that loop happens few times in a row, Tasmota will reset the config (see setoption36). If your power regulator is faulty, could also trigger this config reset.

 So, you can try `setoption1 1` and `setoption36 0` in order to disable all config reset features and test again.

You should increase the weblog to 4 and also, if you can, try to set a syslog for full information.

If your issue is still happening, seems that your hardware is just faulty. Sorry.

marcor78, 2019-03-31:
thanks for your support!
the fact is that both Teckin (52) and Gosund Sp1 (55) are working with my device, so I guess I should try each one with the features you suggested. I'll test it in the next hours and will give a feedback if the issue is still happening. Me too I fear a hardware failure unfortunately. Thanks so much!

ascillato, 2019-03-31:
After you select Teckin or Gosund, try the button to check if the selection was ok or not.

ascillato, 2019-03-31:
both devices have the button connected to different GPIO

marcor78, 2019-03-31:
with Teckin the button is working as expected, with Gosund the button is not working but the device works via web UI. I'm setting to Teckin now and testing for the next hours. Thanks again!

marcor78, 2019-03-31:
well, after nearly 24 hours the issue didn't appear again, so I guess the issue could be considered resolved. The only problem now is that the device while switched on sometimes (let's say every 10 minutes randomly) turns off and immediately on again, like not keeping state in case of lost wifi connection. But I guess this is another issue

ascillato, 2019-03-31:
Have you checked the log?
Seems that your device is indeed turning off. Sorry but the hardware seems to be faulty.

ascillato2, 2019-03-31:
Closing this issue as it is hardware related.

Please, if you have any questions, just ask here or in the chat. Thanks

closed 2019-03-31 by ascillato2