arendst/Tasmota #5561: PMS5003 - wifi disconnects

jara1, 2019-03-31:
Hello,

I'm using couple of devices running 6.5.0 with 2.4.2 core. I was experiencing wifi diconnectes on all of them, but after configuring them to sleep 0, they are stable. Only one which has PMS5003 connected is still disconnecting. If I disable PMS5003 on this device, it is working correctly. I suspect that asynchronous messages comming from PMS5003 are causing this.

Is this possible? Would be possible to implement query-response approach for PMS5003?

ascillato, 2019-03-31:
Hi, 

The wifi disconnects are related to an incompatibility of the core with your router, or a lack of enough power on your device. 

Are you powering your PMS5003 with an external source? If you are using the same that your esp8266 device, may be not enough. 

Also you can try core 2.3.0 

See wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#Wi-Fi-issues-arduino-core-versions-and-expressif-sdk

And also, if you need further help, please complete the troubleshooting template in order to have more information so as to properly help you. Thanks

ascillato2, 2019-03-31:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-31 by ascillato2