arendst/Tasmota #5563: Disable Access Point after setup

eeevs, 2019-03-31:
Hi,
I can't image that I'm the first person with this problem.  I looked at:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Wi-Fi
I used  image : 6.4.1(sonoff) 

After configuration, and connecting to my local network, the access point is still available. I really don't like the idea of an open access point in my network. 

How do I solve this or is it possible it add this option?

Thanks.

Jason2866, 2019-03-31:
No, there isnt any AP after successfull seting up wifi credentials.
Since you havent filled out issue template we cant help....

ascillato, 2019-03-31:
Sorry, I also can't reproduce your issue.

Have you modified Tasmota? Have you erase all flash with esptool.py as explained in the wiki? 

Please, to let us help you, please complete the troubleshooting template. 

Thanks

ascillato2, 2019-03-31:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-01 by ascillato2

synthomat, 2020-12-07:
Is it possible that after a reboot the ESPs open an access point until they join the actually configured WiFi?
I could at least see open APs on my configured ESPs when I experimented with WiFi frequencies. I assume the ESPs have rebooted or something else made them open up the AP again…

ascillato, 2020-12-07:
To set your desired behavior, please see `wificonfig` command in the docs. http://Tasmota.com

synthomat, 2020-12-07:
@ascillato ah awesome, thanks! I must have overlooked it during my search.

njordan77, 2024-04-09:
I have the same issue with SHELLY Plug S on Tasmota 13.4. even though i'm connected over weeks with several reboots, i have always a ShellyPlus1-xxxxxx Wifi Network running, the worst its OPEN.

I read that this should not happen when i'm connected to my network sucessfully, which is the case. otherwise i would not be able to swithc the device on/off via web interface.

Any idea what is wrong here?? Kind Regards Norbert