arendst/Tasmota #5564: WiFi HOSTNAME is changing each time rest the module

mikroelektron, 2019-04-01:
Hi,
I'm using Tasmota last version, and I used ESP core 2.4.2, I note that there is a problem in WiFi hostname, When i run the module first time, I see the host-name like  (sonoff-last 6 character of mac address) and sometimes i see it likes (ESP-last 6 charactor of mac address) and sometimes i see it like both of them.  How i can display fixed Host-name each time even i restart the module several times.

ascillato, 2019-04-01:
Hi,

As you were told in the chat, you need to full erase the flash. Please, use esptool.py and the exact command for full erase as explained in the wiki. 

Thanks

ascillato2, 2019-04-01:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-04-01:
#5566

closed 2019-04-01 by ascillato2