arendst/Tasmota #5565: I can't find ES version in latest releases?

GoNzCiD, 2019-04-01:
Why I can't see ES version for latest releases since 6.1.1?

Thanks

Jason2866, 2019-04-01:
What do you mean with ES?

ascillato2, 2019-04-01:
He is right. I have just checked. The spanish version is not in release binaries.

Seems that Theo just forgot to upload it

arendst, 2019-04-01:
Checking...

ascillato2, 2019-04-01:
As a workaround until Theo add it, you can download mine from https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v6.5.0.1/sonoff_KNX-ES.bin

arendst, 2019-04-01:
As far as I know there has never been an ES version. For Spanish use the AR (Argentina) version.

ascillato2, 2019-04-01:
Ah ok,

I was also confused as in platformio.ini the env for Spanish is [env:sonoff-ES]

ascillato2, 2019-04-01:
I think the name should change to ES for Spanish in future releases in order to let AR reserved for Arabic. So far there isn't an arabic translation for Tasmota but who knows, may be we could have.

ascillato2, 2019-04-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-01 by ascillato2

arendst, 2019-04-01:
Changed Argentina to Spain.