arendst/Tasmota #5566: WiFi Hostname changed after resting the sonoff device

mikroelektron, 2019-04-01:
I'm using the latest Tasmota version 6.5, with ESP core 2.4.2.
I used the ESPtools for each time that i want to upload a new firmware, first i make full erase and then i upload the bin code.
### BUG DESCRIPTION
The problem is the WIFI host-name that appears in WiFi list, is changed after resting the module by applying 4 fast clicks on sonoff button. 
I have attached 2 pictures, as shown in first one, the host-name is sonoff-with last 6 char of mac address. After connect with it and make the configuration of wifi, then the module will be connected. I do reset by clicking the button 4 fast times. i got the result that clear in second picture. I do that several times and i got same result. Even the host-name is changed to ESP-with last 6 char of mac address, its still working if you want to connect with it. But i want to get fixed Host-name each time even i reset the module several times. 

[image]
[image]

ascillato2, 2019-04-01:
Confirmated. Thanks for reporting.

arendst, 2019-04-01:
The hostname has always been derived using the following code:
```
  Format(mqtt_topic, Settings.mqtt_topic, sizeof(mqtt_topic));
  if (strstr(Settings.hostname, "%") != nullptr) {
    strlcpy(Settings.hostname, WIFI_HOSTNAME, sizeof(Settings.hostname));
    snprintf_P(my_hostname, sizeof(my_hostname)-1, Settings.hostname, mqtt_topic, ESP.getChipId() & 0x1FFF);
  } else {
    snprintf_P(my_hostname, sizeof(my_hostname)-1, Settings.hostname);
  }
```
So if hostname is defined with "%s_%04d" it will replace %s with formatted topic appended with the last four decimal characters of the MAC address.

What you experience is redundant naming caused by SDK which is out of control by Tasmota and core. The SDK might change hostname with what you see as ESP_\<6 character\> MAC address. This is not fabricated by Tasmota.

Sorry can't help here.

mikroelektron, 2019-04-01:
I also have used core 2.3.0 and 2.5 and i got same result.

closed 2019-04-01 by ascillato2