arendst/Tasmota #5567: WS2812 and mqtt, uncontrolled LED glow

thrusty66, 2019-04-01:
Hi, I have a problem with mqtt and one ws2812 LED.
I built a LED strip into a picture frame as a status indicator. The LEDs are controlled by a NodeMCU and tasmota 6.41. Everything worked well.
Now I have now and then a wildly shining LED. Apparently without a reason an LED lights up in any color. Maybe it's a tasmota problem. But it only appeared after the pimatic update yesterday. I have changed from 0.9.42 to 0.9.47, with node 4.6.2 and npm 2.15.11. 
Now I looked at the problem in the terminal with mosquitto_sub. 
```
pi@rpistrech:~ $ mosquitto_sub -v -t cmnd/node1/led12
cmnd/node1/led12 000000
cmnd/node1/led12 true
cmnd/node1/led12 false
cmnd/node1/led12 true
cmnd/node1/led12 false
```

First he shows me the right attitude. After some time I get the output "true". A little later he shows me "false". With "false" this LED goes on. I have set this LED to 000000 via a terminal and sent retain with it. Even without success.
Can someone help me and tell me why that is. And how I can change it.
It concerns only this one LED of 26LEDs

Also in the console of tasmota he shows me this change. But I don't know where it comes from. I don't have a rule that sets this LED to this color. 
```
17:50:00 MQT: stat/node1/RESULT = {"Led12":"FA0000"}
17:50:25 MQT: stat/node1/RESULT = {"Led12":"000000"}

19:28:09 MQT: stat/node1/RESULT = {"Led12":"000000"}
19:28:25 MQT: stat/node1/RESULT = {"Led12":"FA0000"}
19:30:11 MQT: stat/node1/RESULT = {"Led12":"000000"}
19:30:40 MQT: stat/node1/RESULT = {"Led12":"FA0000"}
```

Edit
Meanwhile, I think that the problem has nothing to do with pimatic. It’s probably just a coincidence that it just showed up after the upgrade.
I have 3 rules that address this LED12. Now I have changed these rules to LED13. That didn’t solve the problem. The problem seems to be mqtt or tasmota. Although I didn’t change anything at mqtt or tasmota.
Can anyone help me here?

ascillato2, 2019-04-01:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-04-01:
Please, also pay attention to make the output of the log requested in the template using weblog to 4 for more debug information. Thanks

thrusty66, 2019-04-01:
I have now changed the firmware to tasmota 6.5.0. Unfortunately the problem is the same, it didn't work.

thrusty66, 2019-04-01:
I don't understand this new template yet. How do I get this template into my request?

ascillato, 2019-04-01:
Follow the link I have provided in my previous comment and copy and paste the template, completing the requested information.

Remember to go to the Tasmota console and type weblog 4 in order to have more debug information. After that, try again your logs because it will have a entry like SRC: with the source of the command that Tasmota is running.

KPK75, 2019-04-01:
Because the command `LEDx RRGGBB` is used to set led number "x" to desired color. And you are sending  wrong value of `FALSE` which is interpreted as FA0000.

thrusty66, 2019-04-02:
My problem is I don't send anything. The LED obviously switches alone. I don't know where the signal false and true comes from.

ascillato, 2019-04-02:
Please,

type `weblog 4` in the Tasmota console. Then wait for that to happen and post here the output of the console

thrusty66, 2019-04-02:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
I use a NopdeMCU with WS2812 Strip for a status display of different devices in the Smarthome. Not all LEDs are addressed. The signal is transmitted via mqtt (mosquitto).
The problem: LED 12 has been on and off sporadically for a week now. All other LEDs work as they should.
The output of mosquitto_sub and tasmota console can be seen in the first thread.
As far as I can understand, the signal to switch LED 12 does not come from my system. I can't find a solution anywhere. 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : NodeMCU
- [ ] Tasmota binary firmware version number used : 6.5.0 / (pre-compiled )
- [ ] Development IDE - Compiler / Upload tools used : --- / NodeMCU-Flasher
- [ ] Provide the output of command ``status 0`` :
```
15:50:08 CMD: status 0
15:50:08 SRC: WebConsole from 192.168.178.69
15:50:08 RSL: Received Topic /status, Data Size 1, Data 0
15:50:08 RSL: Group 0, Index 1, Command STATUS, Data 0
15:50:08 MQT: stat/node1/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"node1","ButtonTopic":"node1","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"node11","SwitchMode":[1,1,1,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":0}}
15:50:08 MQT: stat/node1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T20:52:12","StartupUTC":"2019-04-01T17:57:56","Sleep":50,"CfgHolder":4617,"BootCount":135,"SaveCount":3657,"SaveAddress":"F8000"}}
15:50:08 MQT: stat/node1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:50:08 MQT: stat/node1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Octavia","indebuurt2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000089","280500000100000000000000000000000000","00000001"]}}
15:50:08 MQT: stat/node1/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
15:50:08 MQT: stat/node1/STATUS5 = {"StatusNET":{"Hostname":"node1-1502","IPAddress":"192.168.178.48","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"5C:CF:7F:5A:45:DE","Webserver":2,"WifiConfig":3}}
15:50:08 MQT: stat/node1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A45DE","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:50:08 MQT: stat/node1/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 02 14:50:08 2019","Local":"Tue Apr 02 15:50:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:26","Sunset":"19:21"}}
15:50:08 MQT: stat/node1/STATUS10 = {"StatusSNS":{"Time":"2019-04-02T15:50:08","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON"}}
15:50:08 MQT: stat/node1/STATUS11 = {"StatusSTS":{"Time":"2019-04-02T15:50:08","Uptime":"0T20:52:12","Vcc":2.789,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":0,"Color":"000000","HSBColor":"0,0,0","Channel":[0,0,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Octavia","BSSId":"34:81:C4:CC:94:38","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```

```
15:49:06 CMD: weblog 4
15:49:06 SRC: WebConsole from 192.168.178.69
15:49:06 RSL: Received Topic /weblog, Data Size 1, Data 4
15:49:06 RSL: Group 0, Index 1, Command WEBLOG, Data 4
15:49:06 MQT: stat/node1/RESULT = {"WebLog":4}
15:49:06 WIF: Checking connection...
15:49:06 WIF: Connected

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-04-02:
Thank you very much for sharing all the information.

In the output of the console now you see the SRC: MQTT so the command that makes your undesired behavior is sent by your broker. It is not from Tasmota side.

thrusty66, 2019-04-02:
But why would my broker do that? Can anyone help me? What can I do to stop it? 
I have restarted mosquitto. Without success. The same behaviour with LED 12.

Is there a possibility to lock an LED on the WS2812 directly via mqtt?

ascillato, 2019-04-02:
Something in your network is sending that to Tasmota. You need to check your other softwares that are using mqtt.

That can't be locked in Tasmota. You will need to solve it in the software that is sending that. Sorry

thrusty66, 2019-04-02:
Okay, thanks. I'll try to find something in other forums. I will leave this issue open for now. Maybe somebody can help me, or I will find a solution somewhere.

ascillato2, 2019-04-02:
Closing this issue as it is not a Tasmota Software Bug. Sorry. Your issue seems to be in pimatic.

If you have further questions, please just ask. Thanks

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-02 by ascillato2

thrusty66, 2019-04-20:
Edit

Hello, here's another supplement to my problem. I still don't know how the problem came about and what it was exactly. But finally I know for sure that it was neither pimatic nor tasmota. 
I reinstalled the Mosquito Brocker on a proxmox drive. So with a different IP.  Maybe it was a retain problem. 
So that this topic can get the label SOLVED I think.