arendst/Tasmota #5568: Configute Timer incomplete Webinterface

toto79, 2019-04-01:
### BUG DESCRIPTION
Webinterface for Timers with incomplete Template.
I flashed a Sonoff S20 and a nodemcu V2 completely new, all Options works fine except the "configure Timer".
The problems comes with: 20190330-180834-929292f-sonoff-4m.bin
Still ok is: 6.5.0.3(eb9617e-sonoff)

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff S20 and Nodemcu V2
- [x] Tasmota binary firmware version number used : 6.5.0.3 / (pre-compiled)
- [x] Upload tools used : espflasher
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE FOR S20 (I soldered a 4MB Winbond IC on the S20):
19:25:23 MQT: stat/SonIR4MB/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"SonIR4MB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:25:23 MQT: stat/SonIR4MB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-4m.bin","RestartReason":"Software/System restart","Uptime":"0T01:10:57","StartupUTC":"2019-04-01T16:14:26","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":49,"SaveAddress":"3FA000"}}
19:25:23 MQT: stat/SonIR4MB/STATUS2 = {"StatusFWR":{"Version":"6.5.0.3(a380eb9-sonoff)","BuildDateTime":"2019-04-01T17:07:44","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:25:23 MQT: stat/SonIR4MB/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mesh2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
19:25:23 MQT: stat/SonIR4MB/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":3552,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
19:25:23 MQT: stat/SonIR4MB/STATUS5 = {"StatusNET":{"Hostname":"Son-IR-4MB","IPAddress":"192.168.178.61","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"DC:4F:22:xx:xx:xx","Webserver":2,"WifiConfig":4}}
19:25:23 MQT: stat/SonIR4MB/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.39","MqttPort":1883,"MqttClientMask":"Son-IR-4MB","MqttClient":"Son-IR-4MB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:25:23 MQT: stat/SonIR4MB/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 01 17:25:23 2019","Local":"Mon Apr 01 19:25:23 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:28","Sunset":"20:20"}}
19:25:23 MQT: stat/SonIR4MB/STATUS10 = {"StatusSNS":{"Time":"2019-04-01T19:25:23"}}
19:25:23 MQT: stat/SonIR4MB/STATUS11 = {"StatusSTS":{"Time":"2019-04-01T19:25:23","Uptime":"0T01:10:57","Vcc":3.394,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"34:81:C4:DE:xx:xx","Channel":1,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
```
STATUS 0 OUTPUT HERE FOR Nodemcu V2 absolut original:
19:32:08 MQT: stat/SonNodeMCU/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"SonNodeMCU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:32:08 MQT: stat/SonNodeMCU/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020500/sonoff-4m.bin","RestartReason":"Software/System restart","Uptime":"0T14:18:56","StartupUTC":"2019-04-01T03:13:12","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":21,"SaveAddress":"3F9000"}}
19:32:08 MQT: stat/SonNodeMCU/STATUS2 = {"StatusFWR":{"Version":"6.5.0.3(880bbe3-sonoff)","BuildDateTime":"2019-03-31T18:08:46","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
19:32:08 MQT: stat/SonNodeMCU/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mesh2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000001"]}}
19:32:08 MQT: stat/SonNodeMCU/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":3524,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
19:32:08 MQT: stat/SonNodeMCU/STATUS5 = {"StatusNET":{"Hostname":"Son-NodeMCU","IPAddress":"192.168.178.63","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:69:2B:FC","Webserver":2,"WifiConfig":4}}
19:32:08 MQT: stat/SonNodeMCU/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.39","MqttPort":1883,"MqttClientMask":"SonNodeMCU","MqttClient":"SonNodeMCU","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:32:08 MQT: stat/SonNodeMCU/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 01 17:32:08 2019","Local":"Mon Apr 01 19:32:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:28","Sunset":"20:20"}}
19:32:08 MQT: stat/SonNodeMCU/STATUS10 = {"StatusSNS":{"Time":"2019-04-01T19:32:08"}}
19:32:08 MQT: stat/SonNodeMCU/STATUS11 = {"StatusSTS":{"Time":"2019-04-01T19:32:08","Uptime":"0T14:18:56","Vcc":2.984,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"34:81:C4:xx:xx:xx","Channel":1,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash devices wioth the Archive 20190330-180834-929292f-sonoff-4m.bin and take a look at configure Timers in WebIf

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I like to see 16 Timers with Time Options and Days like in 6.5.0.1(8782053-sonoff).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

arendst, 2019-04-01:
Verified. Wil check tomorrow.

arendst, 2019-04-01:
Issue caused by change from define MAX_TIMERS to const uint8_t MAX_TIMERS.

I'll investigate.

toto79, 2019-04-01:
thx for your fast answer. let me know if i could test something more.

toto79, 2019-04-02:
Fix works fine.

closed 2019-04-02 by toto79