arendst/Tasmota #5570: BSD29 power calibration

tyjtyj, 2019-04-02:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
Based on template https://blakadder.github.io/templates/bsd29.html
BSD29 power calibration wont work.
Steps done
Use 2 bulb load the power to 30W. Calibrate the power according to https://github.com/arendst/Sonoff-Tasmota/wiki/Power-monitoring-calibration

On 2 bulb fully calibrated to 30W (15W + 15W). Power shows 30W - correct
03:24:13 MQT: tele/bsd1/SENSOR = {"Time":"2019-04-02T03:24:13","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.104,"Yesterday":0.085,"Today":0.019,"Period":3,"Power":30,"ApparentPower":30,"ReactivePower":0,"Factor":1.00,"Voltage":239,"Current":0.125}}

On 1 bulb which is 15W . Power shows 14W - correct or near enough 
03:28:21 MQT: tele/bsd1/SENSOR = {"Time":"2019-04-02T03:28:21","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.106,"Yesterday":0.085,"Today":0.021,"Power":16,"ApparentPower":30,"ReactivePower":25,"Factor":0.54,"Voltage":239,"Current":0.125}}

on 3 buld which is 37W . Power remains at 30W
03:29:14 MQT: tele/bsd1/SENSOR = {"Time":"2019-04-02T03:29:14","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.107,"Yesterday":0.085,"Today":0.021,"Period":3,"Power":30,"ApparentPower":30,"ReactivePower":0,"Factor":1.00,"Voltage":239,"Current":0.125}}

I tested load it to 100W, it still showing to 30W. ApparentPower":30,"ReactivePower":0 remain the same all the way. Notice the Factor changed. 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : BSD29
- [0.6.5 ] Tasmota binary firmware version number used : pre-compiled
- [x ] Development IDE - Compiler / Upload tools used : N/A
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
03:24:44 MQT: stat/bsd1/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"bsd1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:24:44 MQT: stat/bsd1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:47","StartupUTC":"2019-04-02T02:03:57","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":65,"SaveAddress":"FB000"}}
03:24:44 MQT: stat/bsd1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-basic)","BuildDateTime":"2019-03-19T12:25:14","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
03:24:44 MQT: stat/bsd1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["J n J",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
03:24:44 MQT: stat/bsd1/STATUS4 = {"StatusMEM":{"ProgramSize":432,"Free":568,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F082380","000483A0","21021706","000001C0"]}}
03:24:44 MQT: stat/bsd1/STATUS5 = {"StatusNET":{"Hostname":"bsd1-4983","IPAddress":"192.168.11.209","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.4.4","Mac":"DC:4F:22:FE:D3:77","Webserver":2,"WifiConfig":4}}
03:24:44 MQT: stat/bsd1/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxxxxxx","MqttPort":1883,"MqttClientMask":"snf_%06X","MqttClient":"snf_FED377","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:24:44 MQT: stat/bsd1/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 02 02:24:44 2019","Local":"Tue Apr 02 03:24:44 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:26","Sunset":"19:21"}}
03:24:44 MQT: stat/bsd1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
03:24:44 MQT: stat/bsd1/STATUS10 = {"StatusSNS":{"Time":"2019-04-02T03:24:44","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.104,"Yesterday":0.085,"Today":0.019,"Power":30,"ApparentPower":30,"ReactivePower":0,"Factor":1.00,"Voltage":239,"Current":0.125}}}
03:24:44 MQT: stat/bsd1/STATUS11 = {"StatusSTS":{"Time":"2019-04-02T03:24:44","Uptime":"0T00:20:47","Vcc":3.141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxJ","BSSId":"60:45:CB:B1xx:xx","Channel":11,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-04-02:
Try to calibrate again. Load for calibration should be 60W or more (old classic bulb no LED!)
Do exact as described in wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration

tyjtyj, 2019-04-02:
The 60w is example not requirement. 'A resistive load device' was tge requirements.
I did the same calibration for sonoffpow nad it works. So this is some to do with the bug? I will find 60w device n try again but i think it is not the cuase. 
The reason if i calibrate using 15w. Max will be 15w regardless the load. If i calibrate using 37w, the max stays at 37w. Any load below calibrated load will shows correctly. Any load beyond it will stay at calibrated load.
I had killawatt meter and sonoffpow use for verification.

tyjtyj, 2019-04-02:
Further monitored. Calibated to 2075W.

When I put a 30W load to it. 
As you can see the current shows 8.9A while and Power is 29W. Does not even makes sense. 
This should be a bug I believe 

14:14:06 MQT: tele/bsd1/STATE = {"Time":"2019-04-02T14:14:06","Uptime":"0T00:00:18","Vcc":3.088,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"POWER":"ON","Wifi":{"AP":1,"SSId":"J n J","BSSId":"40:16:7E:C1:31:D9","Channel":11,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:08"}}
14:14:06 MQT: tele/bsd1/SENSOR = {"Time":"2019-04-02T14:14:06","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.094,"Yesterday":0.085,"Today":0.008,"Period":0,"**Power":29**,"ApparentPower":2118,"ReactivePower":2118,"Factor":0.01,"Voltage":238,"**Current":8.914**}}

Jason2866, 2019-04-02:
Thx for testing recalibration. So the template or your device is wrong/faulty/different.
Sorry to say we cant help you, because we dont have this device (the template for the device was user contributed).

tyjtyj, 2019-04-02:
Thanks for reply. Can you point me which portion of code does the calculations. It think it just something is not correct in calculation P/I/W is calculated. So having 9A and 30W means something is wrong with the calculations.

Jason2866, 2019-04-02:
If it is a HLW8012, BL0937 or HJL-01 power metering IC it is in file https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xnrg_01_hlw8012.ino

Jason2866, 2019-04-02:
If the current is static it is probably not measured. 
So i think the GPIO configuration for your device has to be different.

arendst, 2019-04-02:
There is nothing wrong with the code. Your hardware template configuration is wrong.

Find out the correct information about your device and change the energy metering GPIO's accordingly. Learn about the Energy Metering GPIO functionality.

arendst, 2019-04-02:
Ah, if current is static it means that it provides the (normally static)  voltage at that time.

You will need to change the Current/Voltage GPIO to the inverted one.

tyjtyj, 2019-04-02:
@Jason2866 and @arendst appreciate your help on this. 
I have no idea where to find out about  Energy Metering GPIO . I guess I just leave it what is have right now since the Wattage is correct as long a I calibrate it to really high Wattage device. 
Any device using lower that the calibrated Wattage will be correct so far.
2000W near the limit of the 10A device and I can live with that until I know who does those number in the template means 

Does CF1, CF, SEL mean anything ? I just know SELi and the i means inverted. 

```
Current config  --- 
GPIO4 = BL0937(CF) -- i think this is voltage ?
GPIO5 = HLWBL CF1 --- no idea
GPIO15 = HLWBL SEL(130) -- no idea
```

anyway to get RAW reading of GPIO data? maybe i can start from there.

arendst, 2019-04-02:
Change HLWBL SEL to HLWBL SELi.

This will swap the Current/Voltage signals.

arendst, 2019-04-02:
Choose three from the following energy GPIO's:
- BL0937 CF or HLW8012 CF is the power input signal depending on type of energy monitoring hardware chip.
- HLWBL CF1 is the toggling Current/Voltage input signal
- HLWBL SEL or HLWBL SELi control the type of signal on the HLWBL CF1 input

tyjtyj, 2019-04-02:
This was the template 
{"NAME":"BSD29","GPIO":[0,0,0,0,134,132,0,0,21,17,56,131,0],"FLAG":0,"BASE":45}

I swap 131 and 130. Does not seems to change anything. 

@AS137430 you provided this template with help on Adobo from discord. Do you have device teardown ? or Does your device work correctly ?

AS137430, 2019-04-03:
@tyjtyj : I haven't noticed the issue when the load W changes as I only did the calibration with 1 load. I must have messed up somehow. 
GPIO5 or 15 are probably wrong? You can try to change it to other pins. I haven't opened up the device. I will also try to figure it out.

tyjtyj, 2019-04-04:
> 
> 
> @tyjtyj : I haven't noticed the issue when the load W changes as I only did the calibration with 1 load. I must have messed up somehow.
> GPIO5 or 15 are probably wrong? You can try to change it to other pins. I haven't opened up the device. I will also try to figure it out.

I got to this. So far quite accurate up to 30W. 
{"NAME":"BSD29","GPIO":[0,0,0,131,134,132,0,0,21,17,56,0,0],"FLAG":0,"BASE":52}

Calibrate to 30W. I will try larger load when possible. Does those number makes more sense ?
No load
16:36:43 MQT: tele/bsd2/SENSOR = {"Time":"2019-04-04T16:36:43","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.004,"Yesterday":0.079,"Today":0.004,"Power":0,"ApparentPower":2,"ReactivePower":2,"Factor":0.12,"Voltage":234,"Current":0.007}}

Just Kill-A-Watt similar meter.
16:37:05 MQT: tele/bsd2/SENSOR = {"Time":"2019-04-04T16:37:05","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.004,"Yesterday":0.079,"Today":0.004,"Power":2,"ApparentPower":2,"ReactivePower":0,"Factor":1.00,"Voltage":234,"Current":0.007}}

This is data for 10W bulb + Kill-A-Watt similar meter. 
16:41:00 MQT: tele/bsd2/SENSOR = {"Time":"2019-04-04T16:41:00","ENERGY":{"TotalStartTime":"2019-04-01T15:07:12","Total":0.005,"Yesterday":0.079,"Today":0.005,"Period":1,"Power":12,"ApparentPower":13,"ReactivePower":4,"Factor":0.95,"Voltage":235,"Current":0.054}}

tyjtyj, 2019-04-05:
Confirmed and calibrated for 2 device
{"NAME":"BSD29","GPIO":[0,0,0,131,134,132,0,0,21,17,56,0,0],"FLAG":0,"BASE":52}

Thanks everyone helps and great work on tasmota

closed 2019-04-05 by tyjtyj

fisht99, 2021-08-24:
Hi all,

I know this is old but I had the same issue. This was resolved by calibrating the voltage first followed by the current.

I used ohms law to calculate what the current should be eg. 

150w globe = i = P/V

i = 150 / 240

i = 625ma

CurrentSet 625

Then tweak the power with PowerSet 150

Works like a charm now.