arendst/Tasmota #5571: Power statistics are strange

yuuya45, 2019-04-02:
### BUG DESCRIPTION
Power usage seems not to be reset.
Energy Today always shows the same value as Energy Total.
Energy Yesterday always shows 0.

Did I make a mistake in configuration settings?
I have attached user_config_override.h.

Settings by other commands are as follows.
・ Calibration (VoltageSet, PowerSet, CurrentSet)
・ I just executed "SetOption 19 1" to have Home Assistant recognize it.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Pow R2 / S31
- [x] Tasmota binary firmware version number used : 6.4.1.21 / self-compiled
- [x] Development IDE - Compiler / Upload tools used : ATOM
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
{"Status":{"Module":41,"FriendlyName":["電動カミソリ"],"Topic":"sd-s-ut-sink1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T18:05:15","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":32,"SaveAddress":"F4000"}}
{"StatusFWR":{"Version":"6.4.1.21(sonoff)","BuildDateTime":"2019-03-10T12:07:55","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wx05-NetworkRepeater","r-175-ut-iot"],"TelePeriod":300,"Resolution":"559F80C0","SetOption":["000A8009","280500000100000000000000000000000000","00000220"]}}
{"StatusMEM":{"ProgramSize":531,"Free":472,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE7B4","240783A1","23B617CE","00003BC0"]}}
{"StatusNET":{"Hostname":"sd-s-ut-sink1","IPAddress":"192.168.100.191","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"BC:DD:C2:EA:AF:D0","Webserver":2,"WifiConfig":4}}
{"StatusMQT":{"MqttHost":"192.168.100.254","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EAAFD0","MqttUser":"DVES_USER","MqttType":1,"MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
{"StatusTIM":{"UTC":"Thu Jan 01 18:05:54 1970","Local":"Thu Jan 01 18:05:54 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"22:05","Sunset":"07:09"}}
{"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
{"StatusSNS":{"Time":"1970-01-01T18:05:54","ENERGY":{"TotalStartTime":"2019-03-10T12:09:16","Total":0.041,"Yesterday":0.000,"Today":0.041,"Power":1.105,"ApparentPower":4.189,"ReactivePower":4.000,"Factor":0.26,"Voltage":105.016,"Current":0.040}}}
{"StatusSTS":{"Time":"1970-01-01T18:05:54","Uptime":"0T18:05:15","Vcc":3.418,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"r-175-ut-iot","BSSId":"B0:C7:45:35:AF:80","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### ADDITIONAL CONTEXT
user_config_override.h
```
// -- Master parameter control --------------------
#undef  CFG_HOLDER
#define CFG_HOLDER        4617                   // [Reset 1] Change this value to load SECTION1 configuration parameters to flash

// -- Project -------------------------------------
#undef PROJECT
#define PROJECT                "sd-s-ut-sink1"          // PROJECT is used as the default topic delimiter
#undef MODULE
#define MODULE                 SONOFF_S31      // [Module] Select default model from sonoff_template.h

// -- Wifi ----------------------------------------
#undef WIFI_IP_ADDRESS
#define WIFI_IP_ADDRESS        "192.168.100.191"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or enter a static IP address
#undef WIFI_GATEWAY
#define WIFI_GATEWAY           "192.168.100.1"     // [IpAddress2] If not using DHCP set Gateway IP address
#undef WIFI_DNS
#define WIFI_DNS               "192.168.100.1"     // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#undef STA_SSID1
#define STA_SSID1              "wx05-NetworkRepeater"                // [Ssid1] Wifi SSID
#undef STA_PASS1
#define STA_PASS1              "****************"                // [Password1] Wifi password

// -- Syslog --------------------------------------
#undef WEB_LOG_LEVEL
#define WEB_LOG_LEVEL          LOG_LEVEL_INFO    // [WebLog] (LOG_LEVEL_NONE, LOG_LEVEL_ERROR, LOG_LEVEL_INFO, LOG_LEVEL_DEBUG, LOG_LEVEL_DEBUG_MORE)

// -- Ota -----------------------------------------
#undef OTA_URL
#define OTA_URL                "http://thehackbox.org/tasmota/release/sonoff.bin"  // [OtaUrl]

// -- MQTT ----------------------------------------
#undef MQTT_USE
#define MQTT_USE               1                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)
#undef MQTT_HOST
#define MQTT_HOST              "192.168.100.254"                // [MqttHost]

// -- HTTP ----------------------------------------
#undef WEB_PASSWORD
#define WEB_PASSWORD           "**********"                // [WebPassword] Web server Admin mode Password for WEB_USERNAME (empty string = Disable)
#undef FRIENDLY_NAME
#define FRIENDLY_NAME          PROJECT          // [FriendlyName] Friendlyname up to 32 characters used by webpages and Alexa

// -- mDNS ----------------------------------------
#undef MDNS_ENABLED
#define MDNS_ENABLED           1                 // [SetOption55] Use mDNS (0 = Disable, 1 = Enable)

// -- Time - Up to three NTP servers in your region
#undef NTP_SERVER1
#define NTP_SERVER1            "ntp1.jst.mfeed.ad.jp"       // [NtpServer1] Select first NTP server by name or IP address (129.250.35.250)
#undef NTP_SERVER2
#define NTP_SERVER2            "ntp2.jst.mfeed.ad.jp"    // [NtpServer2] Select second NTP server by name or IP address (5.39.184.5)
#undef NTP_SERVER3
#define NTP_SERVER3            "ntp.nict.jp"  // [NtpServer3] Select third NTP server by name or IP address (93.94.224.67)

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#undef TIME_DST_HOUR
#define TIME_DST_HOUR          0                 // Hour (0 to 23)
#undef TIME_DST_OFFSET
#define TIME_DST_OFFSET        +0              // Offset from UTC in minutes (-780 to +780)

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#undef TIME_STD_HOUR
#define TIME_STD_HOUR          9                 // Hour (0 to 23)
#undef TIME_STD_OFFSET
#define TIME_STD_OFFSET        +540               // Offset from UTC in minutes (-780 to +780)

// -- Location ------------------------------------
#undef LATITUDE
#define LATITUDE               **.******         // [Latitude] Your location to be used with sunrise and sunset
#undef LONGITUDE
#define LONGITUDE              ***.********          // [Longitude] Your location to be used with sunrise and sunset

// -- Application ---------------------------------
#undef APP_TIMEZONE
#define APP_TIMEZONE           9                 // [Timezone] +1 hour (Amsterdam) (-13 .. 14 = hours from UTC, 99 = use TIME_DST/TIME_STD)

/*********************************************************************************************\
 * END OF SECTION 1
 *
 * SECTION 2
 * - Enable a feature by removing both // in front of it
 * - Disable a feature by preceding it with //
\*********************************************************************************************/

// -- OTA -----------------------------------------
#undef USE_ARDUINO_OTA
#define USE_ARDUINO_OTA                          // Add optional support for Arduino OTA (+13k code)

#endif  // _USER_CONFIG_OVERRIDE_H_
```

Jason2866, 2019-04-02:
Please try release version from http://thehackbox.org/tasmota/release/020402/sonoff.bin
You can change via OTA upgrade. Your settings will stay.
If error is still there please erase flash with esptool.py and flash Tasmota again.
I cant reproduce this error, my POW and other power metering devices are showing correct values

yuuya45, 2019-04-02:
Thank you for the response.

Now I'm on the go, so I'll go back and try it.

It has been found in addition.
Devices not connected to MQTT broker and Home Assistant and not executing "SetOption19 1" were normal.

yuuya45, 2019-04-02:
I understand the cause.
It seems that it was a strange date because it can not connect to the NTP server because each device can not go to the Internet.
for that reason

tm.valid = (time_input> 1451602800); // 2016-01-01

Becomes false, and it seems strange because the following logic is not executed when the date is changed.

if (RtcTime.valid) {
      if (LocalTime () == Midnight ()) {
        Settings.energy_kWhyesterday = energy_kWhtoday;
        Settings.energy_kWhtotal + = energy_kWhtoday;
        RtcSettings.energy_kWhtotal = Settings.energy_kWhtotal;
        energy_kWhtoday = 0;
        energy_kWhtoday_delta = 0;
        energy_period = energy_kWhtoday;
        EnergyUpdateToday ();
        energy_max_energy_state = 3;
      }
      if ((RtcTime.hour == Settings.energy_max_energy_start) && (3 == energy_max_energy_state)) {
        energy_max_energy_state = 0;
      }
    }

Is it not possible to set the clock manually?

Jason2866, 2019-04-02:
No, see https://github.com/arendst/Sonoff-Tasmota/issues/5279

You can setup your own NTP server in your network

yuuya45, 2019-04-02:
Thank you for teaching me the solution.
I will try it.

closed 2019-04-02 by yuuya45