arendst/Tasmota #5573: Add support for Robotdyn Dimmer Module

akhif4u, 2019-04-02:
Hi 

Could anyone please integrate the Robotdyn Dimmer module ( https://m.banggood.in/3Pcs-RobotDyn-AC-Light-Dimmer-Module-For-PWM-Control-1-Channel-3_3V5V-Logic-AC-50hz-60hz-220V-110V-p-1255783.html?rmmds=myorder&cur_warehouse=CN ) which uses zero cross detection. The module works with wemos d1 mini using this code https://github.com/nassir-malik/IOT-AC-Light-Dimmer-With-Alexa

Jason2866, 2019-04-02:
Please read https://github.com/arendst/Sonoff-Tasmota/blob/master/CONTRIBUTING.md
You are welcome to contribute the needed driver.

ascillato, 2019-04-04:
@akhif4u

Hi,

Sorry, this issue is duplicated. It has been long discussed this module.

There is an issue that makes very difficult to support it on Tasmota. It requires the Esp8266 to be real time (or very close to it) so as to correctly manage the ZCD (zero crossing detector. Tasmota is not real time due it is supporting a lot of sensors and protocols. So the manage in real time of ZCD is not going to be efficient and the dimming will be really bad. 

So, I'm sorry to say but that type of dimmer module is not supported and not going to be supported.

A working solution for use that dimmer (as other commercial products also do) is to have a separate MCU to manage the ZCD and the dimmimg function. So you can do the dimmer with another arduino or esp8266 with the code you provided and control that device from serial from Tasmota. Using in that way, Tasmota will work using the actual codebase and the rules feature.

Hope this helps.

closed 2019-04-04 by ascillato2