arendst/Tasmota #5574: local schedule active when MQTT (Domoticz) not available

eschuermann, 2019-04-02:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
No, I consider this an enhancement,

**Describe the solution you'd like**
I am using Domoticz icw Sonoff devices. I am using Domoticz to centrally switch the lights on in the evening and off in the morning. Everything works fine except when domoticz is not reachable, for example:
- Domoticz has an issue
- There is a connectivity issue with wifi
Basically it comes down to the MQTT gateway not being reachable.

As Tasmota also allows to have a local schedule, my suggestion would be to use this when Domoticz is not reachable for whatever reason. This allows for having the same functionality (maybe in a limited form) when for instance something breaks during the holidays when you are not home.

So in other words:
- Domoticz avaible: switching is managed centrally by MQTT
- Domoticz not available: local schedule (if activated) of Tasmota manages the switch 

**Describe alternatives you've considered**
none

**Additional context**
none
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-04-02:
Hi,

This feature is already implemented in Tasmota. 

You can make a rule that when there is no mqtt it can turn on the internal timers (global timer control) and when mqtt is reconnected, it can turn off the global timer control.

ascillato2, 2019-04-02:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-02 by ascillato2

eschuermann, 2019-04-02:
Thx, will dive into the Rules options.

Regards,
Erwin

ascillato, 2019-04-02:
If you need help on that, please, just ask here or in the Tasmota Support Chat.