arendst/Tasmota #5575: WIFI_MANAGER dont  pop up in version 6.5.0.4

peiperakos, 2019-04-02:
### BUG DESCRIPTION
in version 6.5.0.1 the wifi manager auto pop up the configuration menu of the wifi!

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) :NodeMCU
- [x ] Tasmota binary firmware version number used :6.5.0.4/ (pre-compiled or self-compiled ?)
- [x ] Development IDE - Compiler / Upload tools used : Arduino / IDE
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
17:50:57 RSL: Group 0, Index 1, Command STATUS, Data 0
17:50:57 MQT: Home/r00t/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["r00t Home","r00t Home2","r00t Home3","r00t Home4"],"Topic":"r00t","ButtonTopic":"r00t","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"r00t","SwitchMode":[0,0,0,0,2,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:50:57 MQT: Home/r00t/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:56","StartupUTC":"2019-04-02T14:49:01","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":11,"SaveAddress":"3F9000"}}
17:50:57 MQT: Home/r00t/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0.4(sonoff)","BuildDateTime":"2019-04-02T17:42:06","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
17:50:57 MQT: Home/r00t/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["r00t",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00028009","280500000100000000000000000000000000","00000200"]}}
17:50:57 MQT: Home/r00t/stat/STATUS4 = {"StatusMEM":{"ProgramSize":666,"Free":3408,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":2,"Features":["00000809","5FFAE39C","000787A0","23B617CC","0001BBC0"]}}
17:50:57 MQT: Home/r00t/stat/STATUS5 = {"StatusNET":{"Hostname":"r00t-0000","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:66:E6:85","Webserver":2,"WifiConfig":1}}
17:50:57 MQT: Home/r00t/stat/STATUS6 = {"StatusMQT":{"MqttHost":"m15.cloudmqtt.com","MqttPort":12271,"MqttClientMask":"r00t-0000","MqttClient":"r00t-0000","MqttUser":"uftnwmep","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:50:57 MQT: Home/r00t/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 02 14:50:57 2019","Local":"Tue Apr 02 17:50:57 2019","StartDST":"Sun Mar 31 03:00:00 2019","EndDST":"Sun Oct 27 04:00:00 2019","Timezone":99,"Sunrise":"07:03","Sunset":"19:47"}}
17:50:57 MQT: Home/r00t/stat/STATUS10 = {"StatusSNS":{"Time":"2019-04-02T17:50:57","Switch5":"OFF","ANALOG":{"A0":0},"AM2301":{"Temperature":24.2,"Humidity":43.6},"BMP180":{"Temperature":22.8,"Pressure":1014.7},"PressureUnit":"hPa","TempUnit":"C"}}
17:50:58 MQT: Home/r00t/stat/STATUS11 = {"StatusSTS":{"Time":"2019-04-02T17:50:57","Uptime":"0T00:01:56","SleepMode":"Dynamic","Sleep":50,"LoadAvg":66,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","Wifi":{"AP":1,"SSId":"r00t","BSSId":"78:96:82:32:BE:D6","Channel":11,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:08"}}}
17:51:01 MQT: Home/r00t/tele/STATE = {"Time":"2019-04-02T17:51:01","Uptime":"0T00:02:00","SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","Wifi":{"AP":1,"SSId":"r00t","BSSId":"78:96:82:32:BE:D6","Channel":11,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:08"}}
17:51:01 MQT: Home/r00t/tele/SENSOR = {"Time":"2019-04-02T17:51:01","Switch5":"OFF","ANALOG":{"A0":0},"AM2301":{"Temperature":24.4,"Humidity":43.7},"BMP180":{"Temperature":22.8,"Pressure":1014.8},"PressureUnit":"hPa","TempUnit":"C"}
17:51:01 DBG: [Temperature  24.4�C  ]

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
i will flash with the arduino ide tasmota v6.5.0.4 and i havent setup wifi settings in arduino ide in the first power up after programming start wifi manager was is expected but in older version auto pop up wifi web page after a succesfull conection in access point

### EXPECTED BEHAVIOUR
i expected to pop up auto wifi manager web page like other versions

ascillato, 2019-04-02:
Confirmated Bug. Thanks for reporting.

peiperakos, 2019-04-02:
@ascillato you welcome ! thanks you !
 if you see this problem maybe you can see why the first time after the compiled and i run the wifi manager because in arduino ide the wifi setting is not set and i want to set it with wifi manager  the tosmota dosent reboot normal but its freeze in wach dog?
and i push the reset button but the setting is stored and after restart the tasmota boot normal in wifi network after the first time the wifi manager work perfect about this option!
this problem i can see it from many older version from 6.4.x.x

i have done this step every time i want to make a change in my tasmota 

1.compile with arduino ide with out wifi setting
2 in first boot tasmota run in wifi manager as i expected 
3 i will write the wifi setting in web page of wifi setting as i expected
4. on click save the tasmota save the setting but itsi freeze in wach dog
5 after press of reset the tasmota works perfect in wifi network
6 if i was run the wifi manager after the first time it works perfect 

thank you for your support!

arendst, 2019-04-02:
I know what I changed. After restore change it works. Investigating why earlier change does not work...

peiperakos, 2019-04-02:
@arendst thanks for your job !

arendst, 2019-04-02:
After fix works again as expected.

peiperakos, 2019-04-02:
its works thenx @arendst for your good job!

can you tell me about my second question ? 

if you see this problem maybe you can see why the first time after the compiled and i run the wifi manager because in arduino ide the wifi setting is not set and i want to set it with wifi manager the tosmota dosent reboot normal but its freeze in wach dog?
and i push the reset button but the setting is stored and after restart the tasmota boot normal in wifi network after the first time the wifi manager work perfect about this option!
this problem i can see it from many older version from 6.4.x.x

i have done this step every time i want to make a change in my tasmota

1.compile with arduino ide with out wifi setting
2 in first boot tasmota run in wifi manager as i expected
3 i will write the wifi setting in web page of wifi setting as i expected
4. on click save the tasmota save the setting but itsi freeze in wach dog
5 after press of reset the tasmota works perfect in wifi network
6 if i was run the wifi manager after the first time it works perfect

thank you for your support!

arendst, 2019-04-02:
Will have a quick look...

arendst, 2019-04-02:
Just tried:
```
00:00:17 CMD: reset 3
00:00:17 RSL: stat/sonoff/RESULT = {"Reset":"Erase, Reset and Restarting"}
00:00:20 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

00:00:00 CFG: Loaded from flash at F8, Count 4
00:00:00 Project sonoff Sonoff Version 6.5.0.4(basic)-2_4_2
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-6928 with IP address 192.168.4.1
00:01:18 CMD: seriallog 3
00:01:18 RSL: stat/sonoff/RESULT = {"SerialLog":"3 (Active 3)"}
00:01:19 CFG: Saved to flash at F6, Count 6, Bytes 3584
00:01:24 HTP: Redirected to captive portal
00:01:27 HTP: Configure WiFi
00:01:27 WIF: Hostname wemos4, SSId1 indebuurt1, SSId2 indebuurt2
00:01:27 HTP: Restart
00:01:27 SRC: WebGui from 0.0.0.0
00:01:27 CFG: Saved to flash at F5, Count 7, Bytes 3584
00:01:28 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.5.0.4(basic)-2_4_2
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 indebuurt1 in mode 11N as wemos4...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on wemos4 with IP address 192.168.2.223
00:00:06 APP: (UTC) Tue Apr 02 19:04:05 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
20:04:06 CFG: Saved to flash at F4, Count 8, Bytes 3584
20:04:07 RSL: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.5.0.4(basic)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"sonoffs"}
20:04:07 RSL: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
20:04:07 RSL: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
20:04:07 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
20:04:07 RSL: stat/sonoff/POWER = OFF
20:04:09 APP: Boot Count 3
20:04:09 CFG: Saved to flash at FB, Count 9, Bytes 3584
20:04:15 RSL: tele/sonoff/STATE = {"Time":"2019-04-02T20:04:15","Uptime":"0T00:00:16","Vcc":2.972,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt1","BSSId":"24:D3:F2:97:C0:A1","Channel":1,"RSSI":38,"LinkCount":1,"Downtime":"0T00:00:06"}}
```
No issues here.

If you have serial connected you might want to enable more logging with command ``seriallog 3`` and see if strange things happen.

peiperakos, 2019-04-02:
look mine 
`00:00:00 HTP: Web server active on sonoff-4738 with IP address 192.168.4.1
00:00:27 CMD: serialog 3
00:00:27 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
00:00:39 CMD: seriallog 3
00:00:39 RSL: stat/sonoff/RESULT = {"SerialLog":"3 (Active 3)"}
00:00:40 CFG: Saved to flash at F8, Count 4, Bytes 3584
00:00:58 HTP: Redirected to captive portal
00:00:59 HTP: Configure WiFi
00:01:00 HTP: Redirected to captive portal
00:01:00 HTP: Configure WiFi
00:01:01 HTP: Redirected to captive portal
00:01:02 HTP: Redirected to captive portal
00:01:04 HTP: Configure WiFi
00:01:06 WIF: Scan done
00:01:06 WIF: SSId r00t, BSSId 78:96:82:32:BE:D6, Channel 11, RSSI -58
00:01:06 WIF: SSId WindWiFi_885C, BSSId C4:71:54:49:88:5C, Channel 2, RSSI -71
00:01:06 WIF: SSId Maria, BSSId 24:D3:F2:9D:E8:B9, Channel 6, RSSI -79
00:01:06 WIF: SSId Forthnet-BE51D9, BSSId 9C:97:26:BE:51:D9, Channel 11, RSSI -82
00:01:06 WIF: SSId OTE23A304, BSSId 94:A7:B7:23:A3:04, Channel 1, RSSI -86
00:01:06 WIF: SSId TP-LINK_AA16, BSSId 84:16:F9:5F:AA:16, Channel 10, RSSI -87
00:01:06 WIF: SSId 1e550a22, BSSId F4:91:1E:55:0A:22, Channel 1, RSSI -89
00:01:08 HTP: Redirected to captive portal
00:01:09 HTP: Redirected to captive portal
00:01:09 HTP: Configure WiFi
00:01:09 HTP: Redirected to captive portal
00:01:10 HTP: Redirected to captive portal
00:01:20 HTP: Configure WiFi
00:01:20 WIF: Hostname %s-%04d, SSId1 r00t, SSId2 
00:01:20 HTP: Restart
00:01:20 SRC: WebGui from (IP unset)
00:01:21 CFG: Saved to flash at F7, Count 5, Bytes 3584
00:01:21 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
`
and freeze here until reset manually
this is only for the first time
after that works

arendst, 2019-04-02:
What happens if you also select an AP for SSID2?

arendst, 2019-04-02:
Ah. That's a very old issue and is related to the ESP8266 hardware. There are some issues here that explain it too but in short it has to do with your serial firmware upload which puts the device in a wrong reboot state.

The only solution is every time you serial upload firmware you need to do a hard reset, or power cycle first before starting to use the device.

Search the net about this issue which has nothing to doo with Tasmota.

See this too https://github.com/esp8266/Arduino/tree/master/doc/faq#i-have-observed-a-case-when-esprestart-doesnt-work-what-is-the-reason-for-that

peiperakos, 2019-04-02:
i have only one AP but in some try i have file both SSID1 And the SSID2 with the same AP of my router and password same and it have the same action!
i will compile in nodemcu and in one esp01 and i have the same thing!

peiperakos, 2019-04-02:
OMG thenx @arendst how many things do you have in your mind :)
i have many many try but not this to power off and on the tasmota first then run in wifi manager for the firs time thats it thenx again from Greece !!
and have a nice time

ascillato2, 2019-04-02:
Closing this issue as has been fixed. Thanks :+1:

closed 2019-04-02 by ascillato2