arendst/Tasmota #5576: Ht12e support

spanout, 2019-04-02:
Has anyone tried to use an encoded rf signal from the ht12e to sonoff rf bridge? I can send codes from my 433mhz Transmitter connected to an arduino and see the results on the tasmota console but i do not see anything if i connect the ht12e to the Transmitter.

Has anyone tried to do this?

ascillato, 2019-04-02:
You need to use portish firmware for the RF chip of the sonoff bridge. Please, check the wiki on how to flash that other chip.

ascillato2, 2019-04-02:
Closing as the question has been answered.

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-02 by ascillato2