arendst/Tasmota #5580: Sonoff RF and two 433 switch

techmike78, 2019-04-03:
There is a sonoff rf v1, in which for programming it is necessary to close the jumper R21, Tasmota is successfully filled in it.
It is necessary to configure work from two different switches in the trigger mode (one of the two switches can turn one sonoff on and off).
With a triple click of a button on the sonoff itself, I clear all the switch bindings. After 10 seconds, I double-click the switch 1, 5 seconds again after double-clicking the button on the sonoff, I remember the switch 2.
After 10 seconds, I try off / off with the switch1 - the LED on the sonoff lights up red, but the relay does not work. I try to switch off - it works.
Repeat cleaning and binding in reverse order - only the last associated switch controls the relay.

I tried to tie for v2 (long press of a button) - does not work. I read here https://github.com/are...Tasmota/wiki/Sonoff-RF, but did not find the answer.
How to make the sonoff relay work from both switches?

Jason2866, 2019-04-03:
The device supports only one RF Remote. It is hardware related and cant be changed from software side
See https://www.itead.cc/wiki/Sonoff_RF

ascillato2, 2019-04-03:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-03 by ascillato2

techmike78, 2019-04-03:
This link does not indicate about one RF Remote. How did this become known?
Are you sure?

Jason2866, 2019-04-03:
Yes

MikeVRT, 2019-04-12:
> supports only one RF Remote

One Sonoff RF successfully works from two remotes.

Jason2866, 2019-04-12:
Lucky 😀
Official supported is one remote

MikeVRT, 2019-04-13:
> Official supported is one remote
Inside standard 433Mhz module (receiver). It official supported more than two remotes :)

techmike78, 2019-04-13:
Receiver 433 gives commands and bindes with a microcontroller that pulls the esp pin. Most likely the problem is in the microcontroller program. How did you get two remotes tied?

MikeVRT, 2019-04-13:
> and bindes with a microcontroller
As I know, all more simple. Receiver shorts GPIO0 to GND as button, not more. (you can understand it from description of flash RF version - it has some specific to flash, yes? :) )

techmike78, 2019-04-13:
That's right, the microcontroller controls GPIO0 and no matter how many 433 MHz buttons are connected. But for me, as I wrote above, GPIO0 changes only to the last 433 connected button.

MikeVRT, 2019-04-13:
techmike78, if it's help - I uses EV1527 transmitters.

techmike78, 2019-04-13:
I to:
[chip](https://ru.aliexpress.com/item/Qiachip-5-433/32674247988.html?spm=a2g0s.9042311.0.0.274233edOIUV1P)
[button](https://ru.aliexpress.com/item/1-Sticky-RF-Smart/32836188044.html?spm=a2g0s.9042311.0.0.274233edOIUV1P)

MikeVRT, 2019-04-13:
Same 2 chips on 1 Sonoff RF works ok.

techmike78, 2019-04-13:
I don't understand why two chips for one sonoff don't work for me
What am I doing wrong?

MikeVRT, 2019-04-14:
> I don't understand why two chips for one sonoff don't work for me
> What am I doing wrong?

What about my preview p.s.?

techmike78, 2019-04-28:
Switch1 - switch with aliexpress with mark CreaJian-X21
Switch2 - TX-118S-4 V2 qiachip
Sonoff RF TH_V1.1 2017-5-5
Tried with p.s. also.
1. I quickly pressed the sonoff button three times, the red LED on it turned on for 3 seconds and went out. Sonoff was reacting to a previously attached button.
2. I quickly pressed the sonoff button twice, and for two seconds a red LED blinked on it. He pressed the button on the switch1, Sonoff After that, he blinked a red LED and turned on the light.
3. Turn off the light with the button on the sonoff.
4. quickly pressed the button on the sonoff twice, after 2 seconds it blinked with a red LED. I pressed the button on the sonoff switch2 then also blinked with a red LED and turned on the light.

Total: all associated switches are reset and two new switches are newly connected.

I press the button1, the sonoff lights up at the time of pressing the red LED, but the light does not turn on.
I press the button 2, the sonoff lights up with a red LED and turns on the light.

In total: both bound switches do not work.
What am I doing wrong?

[photo](https://yadi.sk/d/BhRZZqemqw_KGg)