arendst/Tasmota #5581: Exception when setting GPIO to WS2812

brainthee, 2019-04-03:
I previously raised this as #4645, however I've been experiencing this again today. The previous issue was mistakenly raised as boot loop reverting settings. To be clear - that is not the issue here. 

After upgrading to 6.5.0.4 today, my D1 mini goes into a boot loop. Once it's finally cleared the GPIO settings, I try and reset the D4 pin to use WS2812 and it causes an exception triggering the boot loop once again.

Things I've tried today:
- [x] Tried all three major core platforms (2.3, 2.4.2 and 2.5)
- [x] Tried changing the WS2812 to a different pin
- [x] Tried powering the D1 mini via the usual 20A 5V power supply (providing it to it's 5V in) AND tried powering JUST the board with no lights via USB to rule out power issues
- [x] Tried latest master checkout as well as 6.5.0 checkout. I was running 6.4.0 previously. The only reason I compile from source is due to loading my own user_config_override.h. I make no code changes other than that.
- [x] Tried clearing the configuration (via changing CFG_HOLDER) 
- [x] _Development/Compiler/Upload tools used_ : **Visual Studio Code & Platform IO**
- [x] Output of ''status 0'':
```
18:03:55 MQT: home/DVES_B3FA70/STATUS = {"Status":{"Module":18,"FriendlyName":["Office"],"Topic":"DVES_B3FA70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:03:55 MQT: home/DVES_B3FA70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.7:80/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:12:56","StartupUTC":"2019-04-03T16:50:59","Sleep":50,"CfgHolder":4374,"BootCount":8311,"SaveCount":60212,"SaveAddress":"FA000"}}
18:03:55 MQT: home/DVES_B3FA70/STATUS2 = {"StatusFWR":{"Version":"6.5.0.4(sonoff)","BuildDateTime":"2019-04-03T17:48:17","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
18:03:55 MQT: home/DVES_B3FA70/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"192.168.XX.8","LogPort":5140,"SSId":["ThXXe","XXX"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
18:03:55 MQT: home/DVES_B3FA70/STATUS4 = {"StatusMEM":{"ProgramSize":540,"Free":460,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F9AE394","040783A1","23B617CE","00003BC0"]}}
18:03:55 MQT: home/DVES_B3FA70/STATUS5 = {"StatusNET":{"Hostname":"DVES_B3FA70-6768","IPAddress":"192.168.XX.161","Gateway":"192.168.XX.1","Subnetmask":"255.255.255.0","DNSServer":"194.168.4.100","Mac":"84:F3:EB:B3:FA:70","Webserver":2,"WifiConfig":4}}
18:03:55 MQT: home/DVES_B3FA70/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.XX.7","MqttPort":1883,"MqttClientMask":"DVES_B3FA70","MqttClient":"DVES_B3FA70","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:03:55 MQT: home/DVES_B3FA70/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 03 17:03:55 2019","Local":"Wed Apr 03 18:03:55 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:38","Sunset":"19:43"}}
18:03:55 MQT: home/DVES_B3FA70/STATUS10 = {"StatusSNS":{"Time":"2019-04-03T18:03:55"}}
18:03:55 MQT: home/DVES_B3FA70/STATUS11 = {"StatusSTS":{"Time":"2019-04-03T18:03:55","Uptime":"0T00:12:56","Vcc":2.993,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"TheXXXse","BSSId":"C0:3E:0F:9D:9B:65","Channel":11,"RSSI":82,"LinkCount":2,"Downtime":"0T00:00:07"}}}
```

I can also provide the stack traces here:
```
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
;l␀d��|␀�$�|␂␄␄␌�␌d�␌#|��␃�␓�r�#�␄c��g'�lno���␌#␜x�${lsdp�g�␐␃␄␄�␄l␌��␄␄␌#␌g�|␃d�␌␄�c��'o�␀d��d`␃�␓␛g'␄l`␃␇␃n;���g␄␌c␄�␏l␏s��g␄␌c␄�␇l�␓$�p␂���␛�`␃��o�␃
00:00:00 CFG: Loaded from flash at F5, Count 60201

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

00:00:00 CFG: Loaded from flash at FB, Count 60203

Exception (0):
epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb00 end: 3fffffc0 offset: 01a0
3ffffca0:  3fff37ec 00000264 00000264 4023d16c
3ffffcb0:  3ffffcd0 3ffffcc0 3fff00e8 40101c80
3ffffcc0:  00000000 00000000 3ffffd2c 00000011
3ffffcd0:  00000032 00000024 3ffe8b10 00000001
3ffffce0:  00000004 0000000c 00000014 4021a30f
3ffffcf0:  3ffffd60 3ffffd50 00000004 4022021a
3ffffd00:  353a3731 36333a30 00000014 4020a6e1
3ffffd10:  3fff2220 00000000 00000001 40213afb
3ffffd20:  3ffffd90 40279d23 00000015 00000001
3ffffd30:  3fff22ac 00000000 00000001 4021bb30
3ffffd40:  00000001 00000000 3fff21bc 402051b2
3ffffd50:  3ffffe1a 40278af4 00000012 40278af4
3ffffd60:  00000001 00000000 3ffffd90 3ffffd90
3ffffd70:  00000001 3fff21c0 3ffffd90 4021c1fc
3ffffd80:  3fffc718 3fff22ac 000000fa 0000eb2c
3ffffd90:  00004e4f 3fff22ac 3fff06d4 3fff22ac
3ffffda0:  646e6d63 4556442f 33425f53 30374146
3ffffdb0:  2f62665f 3fff2200 00000063 401011ad
3ffffdc0:  40104ba9 3fff2055 3fff2198 00000001
3ffffdd0:  3fff22ac 3fff2172 3fff2198 402268e0
[... 231 lines cut ...]
3fffff30:  00000000 00000004 00000004 3fff35f4
3fffff40:  00000004 00000004 00000004 402221dc
3fffff50:  00000000 3fff36d0 3ffe8688 40226cfc
3fffff60:  001ee829 3fff35f4 40239708 3fffefa0
3fffff70:  40201dd6 00000027 00000004 4020a6e1
3fffff80:  3fffdad0 00000000 00001aba 40226a30
3fffff90:  00000000 00000000 00000001 3fff35f4
3fffffa0:  3fffdad0 00000000 3fff35c4 402397b8
3fffffb0:  feefeffe feefeffe 3ffe8688 40101cc5
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

00:00:00 CFG: Loaded from flash at F4, Count 60210

```

Any clues?

Jason2866, 2019-04-03:
Sdk 3.0.0 you used is known to be buggy. We provide no support for.
Please try a prebuild image with core 2.4.2 or core 2.3.0 if this issue exists here too.
If yes please provide `status 0 `outputs

brainthee, 2019-04-03:
Ahh fair enough. Will try the prebuilt images shortly. 

I wasn't aware I was using the 3.0 version of the SDK. Apologies for the naive question but is that a VS Code or Platform IO component?

brainthee, 2019-04-03:
Very interesting thing happened. So I uploaded the prebuild image of sonoff (the recommended 2.3.0 binary). Reconfigured wifi & GPIO and it worked perfectly. 

20 minutes later, went to configure my MQTT settings (server and topic) and it crashed with the same exception details as above.

```
00:00:00 Project sonoff Office Desk Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 XXX in mode 11N as DVES_B3FA70-6768...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on DVES_B3FA70-6768 with IP address 192.168.XX.161
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: home/DVES_B3FA70/LWT = Online (retained)
00:00:05 MQT: home/DVES_B3FA70/cmnd/POWER =
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100]}
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"Scheme":1}
00:00:05 MQT: home/DVES_B3FA70/INFO1 = {"Module":"Generic","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/DVES_B3FA70_fb/","GroupTopic":"sonoffs"}
00:00:05 MQT: home/DVES_B3FA70/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_B3FA70-6768","IPAddress":"192.168.XX.161"}
00:00:05 MQT: home/DVES_B3FA70/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40106fc5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"POWER":"ON"}
00:00:05 MQT: home/DVES_B3FA70/POWER = ON
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"000000","HSBColor":"0,0,0","Channel":[0,0,0]}
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"000000","HSBColor":"0,0,0","Channel":[0,0,0]}
00:00:05 MQT: home/DVES_B3FA70/RESULT = {"Command":"Unknown"}
00:00:06 MQT: home/DVES_B3FA70/RESULT = {"Speed":1}
00:00:06 MQT: home/DVES_B3FA70/RESULT = {"Width":1}

Exception (0):
epc1=0x40106fc5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff51d0 end: 3fff56a0 offset: 01a0

>>>stack>>>
3fff5370:  40250326 3fff1fa0 3fff87ea 3fff5470
3fff5380:  00000001 00000005 00000003 40235cc0
3fff5390:  3fff53b0 3fff53a0 3fff08dc 4010020c
3fff53a0:  00000018 00000000 00000000 3fff5430
3fff53b0:  3fff571c 000001d1 000001d1 00000001
3fff53c0:  00000004 0000000c 00000014 4021ace7
3fff53d0:  00000000 00000000 3fff299c 4022103a
3fff53e0:  303a3030 00000004 00000014 4020a895
3fff53f0:  3fff299c 00000000 00000001 40213797
3fff5400:  3fff5470 00000017 3fff5430 00000001
3fff5410:  3fff2a28 00000000 00000001 4021c71e
3fff5420:  00000001 00000000 0000007c 40204e65
3fff5430:  3fff54fa 00000017 3fff2938 3fff5471
3fff5440:  3fff87d0 0000002b 3fff5470 3fff5470
3fff5450:  00000001 3fff293c 3fff5470 4021ce34
3fff5460:  00000009 00000020 00000009 4022120c
3fff5470:  00004e4f 00000024 00000009 4021fa3e
3fff5480:  646e6d63 4556442f 33425f53 30374146
3fff5490:  2f62665f 3fff2c00 3fff2a28 4021f5b5
3fff54a0:  ffffffff 3fffc6fc 0000005c 40107859
3fff54b0:  3fff55a2 3fff27d5 3fff2914 40203d37
3fff54c0:  3fff2a28 3fff27d5 3fff2914 40225344
3fff54d0:  000000c0 00028300 0000a8c0 4025d204
3fff54e0:  40235a3e 656d6f68 4556442f 33425f53
3fff54f0:  30374146 6e6d632f 4f502f64 00524557
3fff5500:  3fff87a4 3ffe8a06 00000001 4010068c
3fff5510:  40250bf9 00000000 00000001 40107ca8
3fff5520:  40258ab2 00000006 3fff8ae4 40258ac9
3fff5530:  3fff6e94 00000000 00000000 00000000
3fff5540:  00001972 00000006 65776f50 00670072
3fff5550:  00001ac0 3fff6e94 3fff877c 40229b9a
3fff5560:  3fff893c 3fff55a2 0000004e 40229bd6
3fff5570:  3fff893c 00000001 3fff094c 4022fa60
3fff5580:  3fff5470 00000003 3fff55a0 0000001c
3fff5590:  3fff5470 3fff54fa 00000000 3fff5470
3fff55a0:  3fff5548 00000002 00000000 00000000
3fff55b0:  3fff55f0 00000001 3fff0d4c 0000001b
3fff55c0:  000003e5 3fffdad0 3fff7a64 0000001b
3fff55d0:  3fff0d4c 00001ac0 3fff094c 40235d8c
3fff55e0:  00000004 00001ac0 3fff094c 40230198
3fff55f0:  00000001 00000000 3fff7a64 4022b639
3fff5600:  00000001 00000004 3fff094f 00000021
3fff5610:  00000000 00000000 00019594 3fff466c
3fff5620:  00000004 00000004 00000000 40206e20
3fff5630:  3ffefaa0 3fff56f8 3fff4680 40223169
3fff5640:  3fffdad0 3fff466c 40235a44 3fff4680
3fff5650:  40202332 00000004 00000004 4020a895
3fff5660:  3fffdad0 00000000 00001ac0 402256b8
3fff5670:  00000000 00000000 00000001 3fff466c
3fff5680:  3fffdad0 00000000 3fff4665 40235a90
3fff5690:  feefeffe feefeffe 3fff4680 40100718
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld
```

Jason2866, 2019-04-03:
Looks like a power supply issue of the ESP8266.
Probably you have a wemos mini with a faked 3.3 Vreg on it.

ascillato2, 2019-04-04:
Exception 0 is a not enough power issue.

Please, try another board.

ascillato2, 2019-04-04:
Closing this issue as it is hardware related and nothing can be done from Tasmota software side. 

Please, if you need further assistant, you can address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md)).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-04 by ascillato2

brainthee, 2019-04-04:
Hey @ascillato2, thanks for replying.

Unfortunately while the exception 0 explanation may be correct most of the time, I think it may be incorrect here. My reasoning is by clearing retained MQTT messages and trying the prebuild (or using core in my own compiled version), the exact same device with the exact same power supply works perfectly fine past several power cycles and 100% brightness calls (from 0%). If it was due to a power supply issue or flash failure (or other hardware failure) then I would expect the problem to persist?

I think I'm getting slightly closer to the issue. As I mentioned above, one of the key actions I did last night was to clear the MQTT retained messages. That seemed to solve the issue. When I looked to see what messages had been retained, an openHAB rule I had made had tried to call HSBColor with 255,0,0 (I now know thats an invalid HSB code!) among other calls (dimmer & scheme) within quick succession.

I'm going to be trying to reproduce it today to hopefully have some steps for others to replicate what I'm seeing.

ascillato, 2019-04-04:
I have been searching and another cause of exception (0) is a bad flashing due to not using DOUT. So, can you confirm that you have erased all flash with esptool.py as explained in the wiki and flashed the precompiled bin with esptool.py with DOUT as explained in the wiki?

brainthee, 2019-04-04:
Not explicitly. Annoyingly I haven't been able to reproduce the problem today either.

I'll keep playing but if I come up against the problem again and can offer more information I'll update further. Until then, thank you for the suggestions (especially the core bits)!

ascillato, 2019-04-04:
So, don't worry. Sometimes this weird issues are just a bad flashing. Erasing and flashing again (with esptool.py) solves these problems