arendst/Tasmota #5582: Can't get DHT22 working with the latest Tasmota version

marclefevre, 2019-04-03:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I have an old Sonoff Basic (rev1 I'm sure) that was working great with a 1year old version of Tasmota.  It had a DHT22 attached and worked great until one day it threw some exceptions and could not be recovered with power cycling.  I flashed it with the latest Tasmota today (6.5.0.4) and it behaved very erratically in the web interface.  Any configuration change took 10s of seconds and the green LED flashed a lot.  Sometimes, the web interface would not come back and power cycling was required to get back a web interface.
I decided that the SPI flash was a likely culprit so I soldered on a 4Gb version and reflashed with my previous build.
That build has only a few customizations in my_user_config.h
- WIFI SSID and password
- MQTT hostIP, username, and password
- timezone for STD and DST

Once flashed, the web interface came up stable and I changed the module to Sonoff Basic and set GPIO14 to SI7021 which is how the previous, working instance was configured.  After the reset, the temp and humidity did not show on the main web page.  I double checked everything and found no issues.  So I started googling the issue and found that the DHT22 should be configured as AM2301.  I don't know when/if that changed, but I changed the GPIO configuration for GPIO14 to AM2301 expecting that to fix the problem.  It didn't.  In another post similar to mine, Theo said that the DHT22s can be flaky and not work sometimes.  I had another one so I soldered it in.  That didn't help either.
I put an oscilloscope on GPIO14 to see if I could see any activity and there is none.
I don't know where to go next in troubleshooting this.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [* ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [* ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [* ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [* ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [* ] Device used (i.e. Sonoff Basic) : Sonoff Basic rev 1
- [* ] Tasmota binary firmware version number used : 6.5.0.4 built with platformio
- [* ] Development IDE - Compiler / Upload tools used : Platformio
- [* ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
17:11:48 CMD: status 0
17:11:48 MQT: stat/sonoff11/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:11:48 MQT: stat/sonoff11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:52","StartupUTC":"2019-04-03T21:55:56","Sleep":50,"CfgHolder":4617,"BootCount":26,"SaveCount":58,"SaveAddress":"FA000"}}
17:11:48 MQT: stat/sonoff11/STATUS2 = {"StatusFWR":{"Version":"6.5.0.4(basic)","BuildDateTime":"2019-04-03T12:48:31","Boot":16,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
17:11:48 MQT: stat/sonoff11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"10.248.0.30","LogPort":514,"SSId":["darius",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:11:48 MQT: stat/sonoff11/STATUS4 = {"StatusMEM":{"ProgramSize":442,"Free":560,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1620C2","FlashMode":3,"Features":["00000809","0F082380","240483A0","21021706","000001C0"]}}
17:11:48 MQT: stat/sonoff11/STATUS5 = {"StatusNET":{"Hostname":"sonoff11","IPAddress":"10.248.0.151","Gateway":"10.248.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.248.0.1","Mac":"5C:CF:7F:96:3D:45","Webserver":2,"WifiConfig":4}}
17:11:48 MQT: stat/sonoff11/STATUS6 = {"StatusMQT":{"MqttHost":"10.248.0.30","MqttPort":1883,"MqttClientMask":"sonoff11","MqttClient":"sonoff11","MqttUser":"ha_mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:11:48 MQT: stat/sonoff11/STATUS7 = {"StatusTIM":{"UTC":"Wed Apr 03 22:11:48 2019","Local":"Wed Apr 03 17:11:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"00:24","Sunset":"13:23"}}
17:11:48 MQT: stat/sonoff11/STATUS10 = {"StatusSNS":{"Time":"2019-04-03T17:11:48"}}
17:11:48 MQT: stat/sonoff11/STATUS11 = {"StatusSTS":{"Time":"2019-04-03T17:11:48","Uptime":"0T00:15:52","Vcc":3.520,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"darius","BSSId":"1C:B7:2C:D9:54:38","Channel":4,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [* ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
No tele/sonoff/SENSOR messages are ever sent

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-04-04:
Your issue is duplicated. Sorry. You have uploaded the sonoff-basic.bin that does not have any sensor support. Please, upload the full version called sonoff.bin. Thanks

ascillato2, 2019-04-04:
Closing this issue as it is **duplicated**. Sorry.

Please, upload the **sonoff.bin** version for basic sensor support. See the release page. There it is a detailed table with all the features included in every firmware type. 

If you need further assistant with uploading, please find us in the Tasmota support chat. Thanks. 

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-04-04 by ascillato2